Koenkk/zigbee2mqtt #20582: Sengled E21-N1EA OTA fails with error "No image available for imageType '32972'"

rtpguy, 2024-01-05:
### What happened?

When I attempt to check for an OTA for the [Sengled E21-N1EA](https://www.zigbee2mqtt.io/devices/E21-N1EA.html#sengled-e21-n1ea) I see an error pop up in the top right that says "No image available for imageType '32972'". 
This was happening on dev and is now confirmed to be happening on release 1.35.0.

Note prior bug which was closed (possibly not fixed, refer to comments in the linked bug) https://github.com/Koenkk/zigbee2mqtt/issues/16618 

### What did you expect to happen?

Either the Sengled is not shown as available to check for update, or an update is retrieved.

### How to reproduce it (minimal and precise)

Join a Sengled E21-N1EA to the network.
Go to OTA
Query all or query the Sengled only for OTA.
Get failed message about image unavailable for imateType '32972'

Logs:
 2024-01-04 18:56:14Received MQTT message on 'zigbee2mqtt/bridge/request/device/ota_update/check' with data '{"id":"Office Sengled","transaction":"9dpq2-1"}'
Info 2024-01-04 18:56:14Checking if update available for 'Office Sengled'
Debug 2024-01-04 18:56:14Check if update available for '0xb0ce1814001b2da8' (E21-N1EA)
Debug 2024-01-04 18:56:14Using endpoint '1'
Debug 2024-01-04 18:56:16Received Zigbee message from 'Office Sengled', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":38,"imageType":32972,"manufacturerCode":4448}' from endpoint 1 with groupID 0
Debug 2024-01-04 18:56:16Got OTA request '{"fieldControl":0,"manufacturerCode":4448,"imageType":32972,"fileVersion":38}'
Debug 2024-01-04 18:56:16ZigbeeOTA: downloaded main index

### Zigbee2MQTT version

1.35.0

### Adapter firmware version

20230507

### Adapter

ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/13836502/log.txt)

Nerivec, 2024-01-05:
> Either the Sengled is not shown as available to check for update, or an update is retrieved.

If the product supports OTA, it makes sense that you can "check" for updates. But that doesn't mean an update is always going to be available. If an update is available, then the button "Check for new updates" will change accordingly (after a check).
That's the behavior I'd expect, the message lets you know there aren't any update for it yet ([docs](https://www.zigbee2mqtt.io/guide/usage/ota_updates.html#ota-updates)).

That particular model indeed doesn't have any update available via Z2M. [see here for the list](https://github.com/Koenkk/zigbee-OTA)
If Sengled is actually distributing an update for it, then I assume no one has yet shared the update `*.ota`/URL to be added in Z2M....

rtpguy, 2024-01-05:
Technically I should have said for expected - no red error messages and the system simply responds with 'no update available' vs ERROR level messages.  Which would make it consistent with the other devices - they simply don't have an update but aren't throwing error messages and red UI toasts about it.

Nerivec, 2024-01-05:
Well, actually I've encountered this recently with a couple of my devices that can check for firmware updates also (the "others" you mention). The problem is, if your `log_level` is above `info` as mine was to `warn`, then you won't see any toast and think the button didn't work. Whereas if it throws an `error`, you will always see it no matter the `log_level`.

I think something like removing the button after an update was just checked, and outputting the `no avail` message in its place would be an option. Decoupling the `log_level` from that specific `info` toast would probably be a pain (haven't checked the code on that one though). But until then, `error` will prevent people with `log_level` > `info` from spamming the button thinking they didn't click properly... It just needs a bit of cleanup for consistency, I'll see if I can PR something.

closed 2024-08-03 by github-actions[bot]