Koenkk/zigbee2mqtt #20584: Disabled device still working

Eistee82, 2024-01-05:
### What happened?

I disabled a device. Then my automation set a control command in MQTT and the device reacts.

### What did you expect to happen?

Z2M should not communicate with the disabled device (neither send nor receive) and it should not be possible to control the device.

### How to reproduce it (minimal and precise)

- Disable a device
- Control the device over MQTT

### Zigbee2MQTT version

1.35.0 commit: 3d8f0fc3

### Adapter firmware version

20230507

### Adapter

zStack3x0

SVH-Powel, 2024-01-05:
This is the documentation for disabled devices:

> disabled
> Disables the device. This does not prevent the device from communicating in the Zigbee network. E.g. when messages are received from this device, Zigbee2MQTT will still publish them. This feature is useful for devices which will be disconnected from the Zigbee network for a longer time (like Christmas lights). Disabling a device does the following:
> 
> The availability feature will not try to ping it
> The device is always marked as unavailable in Home Assistant
> It will not be configured on Zigbee2MQTT startup (required for some devices in order to start working)
> It will be excluded from network scans (network map)
> It will be excluded from optimistic group state updates

cgallred, 2024-02-07:
Bumping this thread. The documentation for disabled devices says the device can still communicate on the Zigbee network. That's fine, but it seems that if a device is disabled, Z2m should not try to _initiate_ communications. I get log messages like this for devices marked "disabled" that are definitely not trying to communicate on the network, seeing as they're unplugged and sitting in a box:
```
2024-02-06 22:25:27 Publish 'get' 'power_on_behavior' to 'Christmas Snowflakes' failed:
'Error: Read 0x90fd9ffffe652640/1 genOnOff(["startUpOnOff"], 
{
  "timeout":10000,
  "disableResponse":false,
  "disableRecovery":false,
  "disableDefaultResponse":true,
  "direction":0,"srcEndpoint":null,
  "reservedBits":0,
  "manufacturerCode":null,
  "transactionSequenceNumber":null,
  "writeUndiv":false
}) failed (no response received (139))'
```
Is this expected behavior?

Koenkk, 2024-02-07:
@cgallred `Publish 'get' 'power_on_behavior' to 'Christmas Snowflakes' failed:` sounds like you send an MQTT command to read something, did you do this? In the debug log you should see `Received MQTT message`

See [this](https://www.zigbee2mqtt.io/guide/usage/debug.html) on how to enable debug logging.

cgallred, 2024-02-15:
> @cgallred `Publish 'get' 'power_on_behavior' to 'Christmas Snowflakes' failed:` sounds like you send an MQTT command to read something, did you do this? In the debug log you should see `Received MQTT message`

If I did send an MQTT message it wasn't on purpose. This happened when I opened the Z2M dashboard. Up until early last week it would happen pretty consistently. Now, of course, it seems to have stopped. 🙄

closed 2024-09-13 by github-actions[bot]