Koenkk/zigbee2mqtt #20586: ZGP / greenPower messages do not seem to be forwarded by HUE routers

floyd1111, 2024-01-05:
### What happened?

Hi all, after a lot of reading and tinkering around with my zigbee network, I'm looking for some help. I hope I'm posting my problem in the right place. I'm running Zigbee2MQTT as Addon (1.35.0) in Home Assistant (2023.12.4). Home Assistant on a VM via Proxmox on a small pc. Recently I switched from a Conbee II + ZHA to a Sonoff ZBDongle-E (7.3.1.0 build 176) + Zigbee2MQTT. In general great improvement and so far my zigbee network is looking quite stable. It consists of about 100 devices. Mainly Philips Hue & IKEA Tradfri bulbs.

The problematic devices are zigbee green power Friends of Hue switches by Senic (EnOcean PTM 215z). 
I was able to pair them and they work very well around the house with a LQI of about 200. But in a room quite far away from the coordinator one of the switches shows a LQI of 8. Barely usable. Several Tradfri and Hue devices in the same room have a LQI around 150 and work very well if controlled via HA app. 

The switch was paired using "permit join (All)". I tried to pair it directly with a Hue bulb "permit join (Device)" in the same room with no success. If I take the switch for a walk towards the coordinator LQI increases and it works perfectly.
As I understood the green power messages from the switch should be forwarded by the Hue devices. Instead it seems like the only access point for the switches is the main coordinator.

As a workaround I decided to take a Sonoff ZBDongle-P I had lying around, flash it with the Z-Stack router firmware (CC1352P2_CC2652P_launchpad_router_20221102) and place it half way between switch and coordinator. I was able to pair the switch with the router but LQI did not improve although they a only a few meters apart.

Since a few parts of the house have not joined the network yet I would really like to have the switches running reliably. Be it via a zigbee router stick or a hue routing device.

During my research I found this issue: https://github.com/Koenkk/zigbee2mqtt/issues/10782
Sounds in general like my problem but I can't see how to solve it. Thanks in advance for your help!

### What did you expect to happen?

I expect hue routing devices to pick up and relay ZGP messages.
Alternatively I would expect the zigbee router stick to improve the LQI and forward the ZGP messages.

### How to reproduce it (minimal and precise)

1. Create a Zigbee network with z2m and a Sonoff ZBDongle-E coordinator on channel 11
2. Configure several ZGP switches for channel 11
3. Add Huerouters to the network
4. Position the ZGP switches far from the coordinator but in close range to HUE routers

### Zigbee2MQTT version

1.35.0

### Adapter firmware version

7.3.1.0 build 176

### Adapter

Sonoff ZBDongle-E

### Debug log

Not sure which part of which log would help here but I'll do my best to add the information you might need.

floyd1111, 2024-01-05:
Thanks for the good hint @marazmarci ! I retried pairing to a hue light with very short distance. Unfortunately without any success.

Since I read that not all hue devices can be used as green power proxy, here is a screenshot of the "cluster" tab of the specific bulb in z2m I am trying to pair the switch with. Does anybody know, whether the entries at the bottom mark this as a usable green power proxy? 
[image]

chris-1243, 2024-01-17:
Normally with Endpoint 242 - Input cluster - greenPower / Output cluster - greenPower as shown in your picture, it should work. 

I use 7 of them in my network and all have been paired via the "Permit join (Device)" flawlessly.

I just had a try on a test zigbee network (no hue device at all!) and I was able to pair a ptm215 (zigbee greenPower device) with both a Sunricher module and a Nodon module. Both had the exactly same input/output cluster frame on endpoint 242. When I tried to pair the same ZGP directly to the coordinator (Sonoff Dongle-P) and without any other device, I was unable to pair it.

chris-1243, 2024-01-19:
Do you have routers with the same clusters as shown in your picture from your ZGP device and the coordinator? This ensure a smooth transmission of the zigbee commands.
Some of my ZGP devices show LQI below 30 without any issues. I paired them directly to a router able to forward ZGP messages and my network is made with a bunch of routers able to read ZGP commands all around the house (some Ikea bulbs, Hue lights, Ubisys modules...). The endpoint 242 input/output cluster is the most important part.

I hope it may help you

HjalteHansen, 2024-02-28:
I am facing the same issue using a skyconnect adapter. 

Tried to pair it directly to a hue smart plug and light without being able to. need to be close to the coordinator for it to work.

extreme4u, 2024-05-19:
Same problem here. In the past, it worked perfectly. I just found that I have only the output cluster on my routers, not the input cluster. Is there already a solution for this problem?

[image]

extreme4u, 2024-05-21:
> I am facing the same issue using a skyconnect adapter.
> 
> Tried to pair it directly to a hue smart plug and light without being able to. need to be close to the coordinator for it to work.

Did you already solve the problem? Because I updated to ember and the newest firmware for the Skyconnect dongle. But the  ZGP Switches still just connecting with the coordinator. :(

HjalteHansen, 2024-05-22:
> > I am facing the same issue using a skyconnect adapter.
> > Tried to pair it directly to a hue smart plug and light without being able to. need to be close to the coordinator for it to work.
> 
> Did you already solve the problem? Because I updated to ember and the newest firmware for the Skyconnect dongle. But the ZGP Switches still just connecting with the coordinator. :( As I mentioned before, it was working without problems in the past.

Sadly no, i have placed the skyconnect a bit differently so the switches can connect to it. not a great solution, but works for me atm.

chris-1243, 2024-05-22:
@Nerivec

Sadly, it seems ZGP devices are force to be paired via the coordinator only. Alreday reported by myself. I know you are busy with the new `ember` driver. Any idea how to solve this ?

By the way, if some of you have logs to post, it would be really helpfull to debug the problem.

One thing which seems important in my opinion and it was well tested by myself in `zstack`. PTM216Z are hard to pair with Hue device. Not all of them are able to translate the ZGP frames (it seems to use ZGP7). On the other hand, PTM215Z are working fine with lot of devices from different brands (ZGP2, according to zigbee herdsmann converters...). It may be something to look at.

Nerivec, 2024-05-22:
As mentioned before, I don't have much knowledge on the GP spec, and I don't have any device to experiment with. This issue will likely require sniffing some traffic to identify where things are getting lost and possibly why...
@extreme4u When you say "it was working without problems in the past", when was that exactly, and was it with an EmberZNet adapter or another stack?

extreme4u, 2024-05-22:
@Nerivec I don't know exactly how many weeks ago (possibly 3-4 weeks), but I estimate it started with an update from 1.36.1-1 to 1.37.0-1. All my switches stopped working, and it was not possible to reconnect them. At that time, I was using the EZSP adapter stack. I was confused and temporarily connected the switches to deCONZ + Conbee for a few weeks. Then with version 1.37.1-1 I tried again to connect the PTM215Z and they successfully connected just to the coordinator.

Nerivec, 2024-07-01:
@chris-1243 (or someone else 😉) could you test this again with a v8 firmware with latest dev?
With one of these: https://github.com/Nerivec/silabs-firmware-builder/releases/tag/v8.0.0-pre
https://github.com/Koenkk/zigbee2mqtt/discussions/23224

chris-1243, 2024-07-01:
@Nerivec Until the end of July, I won't be able to do any tests.

If someone else is able to do it faster, it would be great otherwise you may have to be patient...😉

You may have a look there #22897, @burmistrzak is really helpfull to debug those zigbeegreen device.

By the way, any advice how to use a CC2652P as a sniffer like how simple it is as to do it with a Dongle-E ?

closed 2025-01-07 by github-actions[bot]

Nerivec, 2025-01-09:
@chris-1243 @burmistrzak you seem to have a few different GP devices on hand. Could one of you provide me with a [sniff](https://github.com/Nerivec/ember-zli/wiki/Sniff) (pcap) of a few different interactions (pairing, triggering buttons, basically anything you can do with GP) with different devices? I don't need anything specific, just samples of various messages.
I have a PTM216Z now, but I need vectors from different devices/interactions for what I'm cooking 😉

PS: you can find me on Discord if you'd prefer sharing it there.

chris-1243, 2025-11-26:
@burmistrzak There is still this improvement which needs to be tested

https://github.com/Koenkk/zigbee-herdsman/pull/1535

I honestly did not have the time to focus on it as I am more focused to improve and test ZoH (aka Zigbee on Host).

> We exclusively pair those EnOcean switches in unicast mode via ZGP-Proxies to prevent flooding the mesh with broadcast messages.

Exclusively, I would not say yes. Pairing via a specific device works perfectly great in `ember`. To pair directly via the coordinator is working (ember allows this) but it seems to be in broadcast unless the above PR change this statement which should be the case.

My main metwork is managed by a SLZB-07MG24 (same chip as your futur ZBT-2) and all my PTM21xZ/ZE work as expected (in unicast).

On point you should be aware of is `ember` network is not completely closed. This applies only to ZGP frames. It means, even if `permit join` is off, `ember` will react to a ZGP commissioning frame...

**Nerivec** might be more precise than me on this topic anyway.

burmistrzak, 2025-11-26:
@chris-1243 @Nerivec Thank you both for getting me up to speed! 🤝

So do I get this right? _Any_ GPZ device in range will be able to join a Ember-based network, without me having to explicitly allow it? This seems kinda problematic, IMHO.

chris-1243, 2025-11-27:
@burmistrzak 

> Any GPZ device in range will be able to join a Ember-based network, without me having to explicitly allow it?

If the device sends a commisioning frame, **yes**.

Like you, I am also concerned and would prefer to have the entire stack closed. As **Nerivec** explained well above, without this _hack_, some devices would not work at all. So Z2M dev team will have to decide wether to keep the situation as it is or not.

I finally tested on my main network 

Koenkk/zigbee-herdsman#1535

and it corrects a small issue when pairing directly to the coordinator. I would advise to wait until the next Z2M release (2.6.4, 1st of decemeber) is published and then move to your new coordinator.

I would also disable commisioning when your PTM215Z will be paired on your new network

https://www.zigbee2mqtt.io/devices/PTM_215Z.html#disabling-commissioning

I need to figure out how I may map the `unknown_102` command send by the device when commisionning is disabled..

Nerivec, 2025-11-28:
> Alternatively, this simply might be a bug in the Silabs stack.

It is. I think they're processing and handing over the GP broadcasts without checking for commissioning status. But as mentioned before, I'm wondering if it was left (done?) on purpose.
This is also what's making a fix rather "weird" on our end, because it wouldn't affect for e.g. zstack (and inversely, any "compatibility mode" would be impossible there).

burmistrzak, 2025-11-30:
> I've asked them if there was a better way to do this, but otherwise, I guess it will be up to the host to track GP commissioning state 😕

@Nerivec Thanks for asking! I'm really interested in their explanation.
Only thing I could find in the Z2M repo(s) was the [`GP_SINK_COMMISSION`](https://github.com/Koenkk/zigbee-herdsman/blob/af015d6be4bf7fbb821dc6650fb130ced969d8c7/src/adapter/ember/ezsp/enums.ts#L415) frame identifier.
So I guess we either implement support for this EZSP frame in Z2M or patch the NCP firmware with a custom stub to handle it internally on incoming ZCL frames?

> For regular permit join, Silabs has an API to trigger it for the NCP. But for GP, they do not (gpSinkCommission mentioned by the AI isn't available in NCP builds).

Do you happen to have a link to source code for said API?

Nerivec, 2025-11-30:
> Have you tried sending that EZSP frame over the wire? Any helpful error codes?

Yes, just in case, I tried it... but confirmed as `NOT_AVAILABLE`.

> Have you tried that before?

Those docs are outdated, some of the stuff they reference only applies to >2yo versions.
The GP stub is automatically removed when adding the GP component nowadays (or at least it should be)...

> the entire EZSP is already implemented

Ember ZLI makes use of quite a few more specific commands that Z2M does not need 😉

burmistrzak, 2025-12-01:
@Nerivec Okay, so I did some more digging and a bit of testing... 😅

Can you try building a fresh set of firmware(s) but add

```
- id: zigbee_green_power_adapter
- id: zigbee_green_power_client
- id: zigbee_green_power_server
```

right below `zigbee_gp` to `component` in the `zigbee_ncp.slcp` file?

This _should_, at least in theory, enable the missing APIs. 🤞

chris-1243, 2025-12-02:
> Jokes aside, please be careful to not burn out your adapter when running it without an antenna. I've cooked RF equipment myself before. 🙈

Thanks for the reminder.👍 It usually lasts a couple of minutes maximum. I should be on the safe side and I do not often test such feature.
This is also why I do not use my main network for this kind of try.
I have not cooked yet a coordinator but killed one by flashing wrong software...

burmistrzak, 2025-12-02:
@chris-1243 Oh absolutely! That NFC stuff was/is super interesting. I really need to get a FoH module with NFC... 😅
There's a bunch of cool Zigbee/Hue stuff I want to explore sometime and maybe implement later on. Among them Hue Entertainment (super low-latency effects) and Motion Aware (occupancy detection).

Nerivec, 2025-12-02:
> to behave like zstack

Well no, because then you are forced to use proxies (like zstack). The big advantage of Ember GP, is the built-in support on the NCP (albeit wonky on the permit subject..). _This is required by spec for Z3.0 by the way._

chris-1243, 2025-12-02:
@burmistrzak 

> Oh absolutely! That NFC stuff was/is super interesting

Below, a small guide to change the DeviceID via NFC.

[PTM216Z_DeviceID_via_NFC.pdf](https://github.com/user-attachments/files/23885002/PTM216Z_DeviceID_via_NFC.pdf)

burmistrzak, 2025-12-11:
@Nerivec Any updates from the Silabs _expert_ yet? 🫣

burmistrzak, 2025-12-11:
@Nerivec Ah, I see. Thanks for clarifying.
There really isn't much processing happening on device, huh? 😅

So, how do we proceed?

Keep track of GP commissioning via Z2M and drop unsolicited frames at the host **or** build a customized firmware and instruct the NCP via EZSP to drop those frames on its end.

IMHO, because we build fresh firmware images anyways, why not go that route and eliminate unnecessary UART traffic from GP frames?

Nerivec, 2025-12-11:
This is way too much to implement on the NCP, the whole state machine is missing.... I also don't like customizing firmware in ways that could require heavy maintenance and per-version support.... Would have to duplicate the logic that Z2M already does anyway.
Traffic is 1 frame every time a GP device is put in pairing mode, so, not an issue.

We'll have to do it in Z2M. Have to look at the impact on those devices that talk only in commissioning frames though, because this would completely drop support for them.... 😰

burmistrzak, 2025-12-11:
> This is way too much to implement on the NCP, the whole state machine is missing....

@Nerivec I guess the Green Power Server component would probably do most of the heavy lifting, but we still would have to implement the appropriate AF APIs calls via custom EZSP commands... So yeah. 😒

> We'll have to do it in Z2M. Have to look at the impact on those devices that talk only in commissioning frames though, because this would completely drop support for them.... 😰

How many devices are we talking roughly?
Some sort of _ZGP compatibility mode_ might be a good compromise.

Nerivec, 2025-12-11:
Green Power server cluster requires ZAP... non-starter for NCP, it just makes a huge mess (not intended for it).

Quite a few, even just the Moes ZEUS, they seem to have done some kind of promotion a while back, lots of users bought it without even realizing it was GP (& crappy)...

burmistrzak, 2025-12-11:
@Nerivec Did I get that right, that you _have to use_ an Ember adapter to make these crappy devices work? No zstack support at all?

If that's already the case, some sort of Ember-only _compatibility mode_ (i.e. restoring the current behavior), but disabled by default, seems like a reasonable solution..?

Alternatively, if we cannot come up with something cleaner, we might be better served by investing that time into ZoH and relegate these NCPs to RCP status. 😉

burmistrzak, 2025-12-12:
> It works on other stack only if `permit join` is always active... So right you are, only `ember` is working.

@chris-1243 Hmm, I guess there's an argument to be made that we should align Ember with the other adapters and drop GP frames outside of commissioning as well. Not pretty, but at least users would know what to expect..?

> Feel free to help😁 if time permits

Our main network still runs on the Sonoff zStack adapter, so I might flash my new ZBT-2 with RCP firmware sometime.
Do we have compatible firmware already built?

chris-1243, 2025-12-12:
> Hmm, I guess there's an argument to be made that we should align Ember with the other adapters and drop GP frames outside of commissioning as well. Not pretty, but at least users would know what to expect..?

I do agree. Now, the dev team (mainly Nerivec and Koenkk) will have to choose to still support thoses Tuya ZGP devices via Ember only or to align Ember to other stacks. Nerviec knows already my point of view on that topic which is like yours...

> Do we have compatible firmware already built?

https://github.com/Nerivec/silabs-firmware-builder/releases/download/v2025.6.2-pre2/nabucasa_zbt-2_openthread_rcp_2.7.2.0_GitHub-fb0446f53_460800_hw_flow.gbl

It would be nice if Silabs chip might be more tested. All my ZoH tries endep up by a crash of the dongle (SLZB-07MG24).

burmistrzak, 2025-12-12:
@chris-1243 Our Zigbee network runs on a dedicated, PoE-powered RPi4 with Docker containers for Z2M, Mosquitto, and Node-RED.
MQTT consumers like NRCHKB and HA run on a completely different, much more powerful machine to keep everything super snappy. I mainly went with this architecture so that everything Zigbee-related runs on a single node. This keeps latency low and improves overall performance.
In the short term, I'd like to replace Mosquitto with something that supports multithreading and also rework the Node-RED flows to be even more efficient.

Regarding the Faraday cage, it _should_ be possible to build a simple tabletop cage out of metal mesh and cardboard. A microwave's window should give us an idea of what kind of mesh would be needed.