Koenkk/zigbee2mqtt #20588: [New device support]: Ezviz T10C Waterleak Sensor

RaunoVV, 2024-01-05:
### Link

https://www.ezviz.com/product/T10C/42454

### Database entry

{"id":18,"type":"EndDevice","ieeeAddr":"0xa4c1384dbcbbfd0d","nwkAddr":40530,"manufId":4451,"manufName":"EZVIZ","powerSource":"Battery","modelId":"CS-T10C-A0-BG\u0000\u0000\u0000\u0000","epList":[1,0],"endpoints":{"0":{"epId":0,"inClusterList":[],"outClusterList":[],"clusters":{"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200}}},"binds":[],"configuredReportings":[],"meta":{}},"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,3,1280,65024],"outClusterList":[25,65024],"clusters":{"65024":{"attributes":{"1":{"type":"Buffer","data":[81,50,56,48,55,52,51,52,56,77,82,89,87,87,67,0]}}},"genBasic":{"attributes":{"modelId":"CS-T10C-A0-BG\u0000\u0000\u0000\u0000","manufacturerName":"EZVIZ","powerSource":3,"zclVersion":3,"appVersion":0}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"zclVersion":3,"interviewCompleted":false,"meta":{},"lastSeen":1704452438239,"defaultSendRequestWhen":"immediate"}

### Comments

Stick: CC2652 - Zigstar Lan Gateway

Just got a Ezviz Water Leaksensor and it's not able to complete the interview.

Some logs of pairing/interviewing:
```
Received Zigbee message from '0xa4c1384dbcbbfd0d', type 'attributeReport', cluster '65024', data '{"1":{"data":[81,50,56,48,55,52,51,52,56,77,82,89,87,87,67,0],"type":"Buffer"}}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2024-01-02 21:05:46: MQTT publish: topic 'zigbee2mqtt/0xa4c1384dbcbbfd0d', payload '{"last_seen":"2024-01-02T19:05:46.968Z","linkquality":163}'
Zigbee2MQTT:debug 2024-01-02 21:05:46: Device '0xa4c1384dbcbbfd0d' announced itself
Zigbee2MQTT:info  2024-01-02 21:05:47: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1384dbcbbfd0d","ieee_address":"0xa4c1384dbcbbfd0d"},"type":"device_announce"}'
Zigbee2MQTT:debug 2024-01-02 21:05:48: Received Zigbee message from '0xa4c1384dbcbbfd0d', type 'attributeReport', cluster '65024', data '{"1":{"data":[81,50,56,48,55,52,51,52,56,77,82,89,87,87,67,0],"type":"Buffer"}}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2024-01-02 21:05:48: Skipping message, still interviewing
Zigbee2MQTT:info  2024-01-02 21:05:48: MQTT publish: topic 'zigbee2mqtt/0xa4c1384dbcbbfd0d', payload '{"last_seen":"2024-01-02T19:05:47.987Z","linkquality":196}'
Zigbee2MQTT:debug 2024-01-02 21:05:48: Received Zigbee message from '0xa4c1384dbcbbfd0d', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":200}' from endpoint 0 with groupID 0
Zigbee2MQTT:debug 2024-01-02 21:05:48: Skipping message, still interviewing
Zigbee2MQTT:info  2024-01-02 21:05:48: MQTT publish: topic 'zigbee2mqtt/0xa4c1384dbcbbfd0d', payload '{"last_seen":"2024-01-02T19:05:48.029Z","linkquality":193}'
Zigbee2MQTT:debug 2024-01-02 21:06:03: Received Zigbee message from '0xa4c1384dbcbbfd0d', type 'readResponse', cluster 'genBasic', data '{"modelId":"CS-T10C-A0-BG\u0000\u0000\u0000\u0000"}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2024-01-02 21:06:03: Skipping message, still interviewing
Zigbee2MQTT:info  2024-01-02 21:06:03: MQTT publish: topic 'zigbee2mqtt/0xa4c1384dbcbbfd0d', payload '{"last_seen":"2024-01-02T19:06:03.375Z","linkquality":171}'
Zigbee2MQTT:debug 2024-01-02 21:06:06: Received Zigbee message from '0xa4c1384dbcbbfd0d', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"EZVIZ"}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2024-01-02 21:06:06: Skipping message, still interviewing
Zigbee2MQTT:info  2024-01-02 21:06:06: MQTT publish: topic 'zigbee2mqtt/0xa4c1384dbcbbfd0d', payload '{"last_seen":"2024-01-02T19:06:06.371Z","linkquality":167}'
Zigbee2MQTT:debug 2024-01-02 21:06:09: Received Zigbee message from '0xa4c1384dbcbbfd0d', type 'readResponse', cluster 'genBasic', data '{"powerSource":3}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2024-01-02 21:06:09: Skipping message, still interviewing
Zigbee2MQTT:info  2024-01-02 21:06:09: MQTT publish: topic 'zigbee2mqtt/0xa4c1384dbcbbfd0d', payload '{"last_seen":"2024-01-02T19:06:09.372Z","linkquality":171}'
Zigbee2MQTT:debug 2024-01-02 21:06:12: Received Zigbee message from '0xa4c1384dbcbbfd0d', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2024-01-02 21:06:12: Skipping message, still interviewing
Zigbee2MQTT:info  2024-01-02 21:06:12: MQTT publish: topic 'zigbee2mqtt/0xa4c1384dbcbbfd0d', payload '{"last_seen":"2024-01-02T19:06:12.371Z","linkquality":171}'
Zigbee2MQTT:debug 2024-01-02 21:06:15: Received Zigbee message from '0xa4c1384dbcbbfd0d', type 'readResponse', cluster 'genBasic', data '{"appVersion":0}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2024-01-02 21:06:15: Skipping message, still interviewing
Zigbee2MQTT:info  2024-01-02 21:06:15: MQTT publish: topic 'zigbee2mqtt/0xa4c1384dbcbbfd0d', payload '{"last_seen":"2024-01-02T19:06:15.373Z","linkquality":171}'
Zigbee2MQTT:debug 2024-01-02 21:06:18: Received Zigbee message from '0xa4c1384dbcbbfd0d', type 'readResponse', cluster 'genBasic', data '{"stackVersion":2}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2024-01-02 21:06:18: Skipping message, still interviewing
Zigbee2MQTT:info  2024-01-02 21:06:18: MQTT publish: topic 'zigbee2mqtt/0xa4c1384dbcbbfd0d', payload '{"last_seen":"2024-01-02T19:06:18.372Z","linkquality":196}'
```

Herdsman debug:
``` 
2024-01-02T18:14:53.505Z zigbee-herdsman:controller:endpoint Read 0xa4c1384dbcbbfd0d/1 genBasic(["swBuildId"], {"sendWhen":"immediate","timeout":10000,"disableR
esponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSeque
nceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
...
2024-01-02T18:14:53.507Z zigbee-herdsman:controller:device:log Interview - IAS - enrolling '0xa4c1384dbcbbfd0d' endpoint '1'
2024-01-02T18:14:53.508Z zigbee-herdsman:controller:endpoint Read 0xa4c1384dbcbbfd0d/1 ssIasZone(["iasCieAddr","zoneState"], {"sendWhen":"immediate","timeout":1
0000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"tr
ansactionSequenceNumber":null,"writeUndiv":false})
2024-01-02T18:14:53.509Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0xa4c1384dbcbbfd0d:47730/1 (0,0,1)
2024-01-02T18:14:53.509Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":47730,"destendpoint":1,"srcendpoint":1,"clusterid":1280,"tran
sid":11,"options":0,"radius":30,"len":7,"data":{"type":"Buffer","data":[16,9,0,16,0,0,0]}}
2024-01-02T18:14:53.510Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,17,36,1,114,186,1,1,0,5,11,0,30,7,16,9,0,16,0,0,0,226]
2024-01-02T18:14:53.520Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
2024-01-02T18:14:53.520Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
2024-01-02T18:14:53.520Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
2024-01-02T18:14:53.520Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
2024-01-02T18:14:53.521Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2024-01-02T18:14:56.480Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,11,205]
2024-01-02T18:14:56.480Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,11,205]
2024-01-02T18:14:56.480Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,11] - 205
2024-01-02T18:14:56.480Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":11}
2024-01-02T18:14:56.481Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
[... 81 lines cut ...]
    at processTicksAndRejections (node:internal/process/task_queues:95:5)
    at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)'
2024-01-02T18:15:53.338Z zigbee-herdsman:controller:database:log Writing database to '/app/data/database.db'
2024-01-02T18:15:53.347Z zigbee-herdsman:controller:error Interview failed for '0xa4c1384dbcbbfd0d with error 'Error: Read 0xa4c1384dbcbbfd0d/1 ssIasZone(["iasC
ieAddr","zoneState"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEn
[... cut ...]
tion expired' (240))'
Zigbee2MQTT:error 2024-01-02 20:15:53: Failed to interview '0xa4c1384dbcbbfd0d', device has not successfully been paired
``` 

### External defintion

```shell
const {batteryPercentage} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['CS-T10C-A0-BG    '],
    model: 'CS-T10C-A0-BG    ',
    vendor: 'EZVIZ',
    description: 'Automatically generated definition',
    extend: [batteryPercentage()],
};

module.exports = definition;
```

rcstanciu, 2024-02-23:
Interested to see this added

tudor934, 2024-02-26:
I have the exact same problem. Herdsman logs below:

2024-02-26T07:20:30.961Z zigbee-herdsman:adapter:ezsp:debg Device join request received: 17259 a4c138eeaec18e66
2024-02-26T07:20:30.961Z zigbee-herdsman:controller:log Device '0xa4c138eeaec18e66' joined
2024-02-26T07:20:30.962Z zigbee-herdsman:controller:log Device '0xa4c138eeaec18e66' accepted by handler
2024-02-26T07:20:30.962Z zigbee-herdsman:controller:log Delete device '0xa4c138eeaec18e66' joined, undeleting
2024-02-26T07:20:30.964Z zigbee-herdsman:controller:database:log Writing database to '/data/database.db'
2024-02-26T07:20:30.987Z zigbee-herdsman:controller:log Interview '0xa4c138eeaec18e66' start
2024-02-26T07:20:30.987Z zigbee-herdsman:controller:device:log Interview - start device '0xa4c138eeaec18e66'
2024-02-26T07:20:30.989Z zigbee-herdsman:controller:device:log Interview - skip node descriptor request for '0xa4c138eeaec18e66', already got it
2024-02-26T07:20:30.989Z zigbee-herdsman:adapter:ezsp:debg Requesting 'Active endpoints' for '17259'
2024-02-26T07:20:30.996Z zigbee-herdsman:adapter:ezsp:driv ZDO Active_EP_req params: {"dstaddr":17259}
Zigbee2MQTT:info  2024-02-26 09:20:30: Device '0xa4c138eeaec18e66' joined
Zigbee2MQTT:info  2024-02-26 09:20:31: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c138eeaec18e66","ieee_address":"0xa4c138eeaec18e66"},"type":"device_joined"}'
Zigbee2MQTT:info  2024-02-26 09:20:31: Starting interview of '0xa4c138eeaec18e66'
Zigbee2MQTT:info  2024-02-26 09:20:31: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c138eeaec18e66","ieee_address":"0xa4c138eeaec18e66","status":"started"},"type":"device_interview"}'
2024-02-26T07:20:31.037Z zigbee-herdsman:adapter:ezsp:ezsp ==> sendUnicast: {"type":0,"indexOrDestination":17259,"apsFrame":{"profileId":0,"sequence":7,"clusterId":5,"sourceEndpoint":0,"destinationEndpoint":0,"groupId":0,"options":256},"messageTag":8,"message":{"type":"Buffer","data":[7,107,67]}}
2024-02-26T07:20:31.037Z zigbee-herdsman:adapter:ezsp:ezsp ==> {"_cls_":"sendUnicast","_id_":52,"_isRequest_":true,"type":0,"indexOrDestination":17259,"apsFrame":{"profileId":0,"sequence":7,"clusterId":5,"sourceEndpoint":0,"destinationEndpoint":0,"groupId":0,"options":256},"messageTag":8,"message":{"type":"Buffer","data":[7,107,67]}}
2024-02-26T07:20:31.038Z zigbee-herdsman:adapter:ezsp:uart --> DATA (3,7,0): 3b00013400006b4300000500000000010000070803076b43
2024-02-26T07:20:31.038Z zigbee-herdsman:adapter:ezsp:uart --> [377921a9602a15d97d3a944a20aa5592499d4e27ace5cd60e0be95ef7e]
2024-02-26T07:20:31.040Z zigbee-herdsman:adapter:ezsp:uart -?- waiting (4)
2024-02-26T07:20:31.050Z zigbee-herdsman:adapter:ezsp:uart <-- [7479a1a9602a1550b9c47e]
2024-02-26T07:20:31.050Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (7,4,0): 7479a1a9602a1550b9c47e
2024-02-26T07:20:31.051Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (0)
2024-02-26T07:20:31.051Z zigbee-herdsman:adapter:ezsp:uart --> [8070787e]
2024-02-26T07:20:31.051Z zigbee-herdsman:adapter:ezsp:uart <-- ACK (4): 7479a1a9602a1550b9c47e
2024-02-26T07:20:31.051Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 3b8001340000e2
2024-02-26T07:20:31.051Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x34: {"_cls_":"sendUnicast","_id_":52,"_isRequest_":false,"status":0,"sequence":226}
2024-02-26T07:20:31.051Z zigbee-herdsman:adapter:ezsp:uart -+- waiting (4) success
2024-02-26T07:20:31.550Z zigbee-herdsman:adapter:ezsp:uart <-- [0479b1a97d312a7d31b259874a25aa5592499cded843868d9874f1d608ca9af0fe0905f51fcb0f64677e]
2024-02-26T07:20:31.550Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (0,4,0): 0479b1a9112a11b259874a25aa5592499cded843868d9874f1d608ca9af0fe0905f51fcb0f64677e
2024-02-26T07:20:31.550Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (1)
2024-02-26T07:20:31.551Z zigbee-herdsman:adapter:ezsp:uart --> [8160597e]
2024-02-26T07:20:31.551Z zigbee-herdsman:adapter:ezsp:uart <-- ACK (4): 0479b1a9112a11b259874a25aa5592499cded843868d9874f1d608ca9af0fe0905f51fcb0f64677e
2024-02-26T07:20:31.551Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 3b90014500040000130000000000000090ffe86b43ffff0c106b43668ec1aeee38c1a480
2024-02-26T07:20:31.551Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x45: {"_cls_":"incomingMessageHandler","_id_":69,"_isRequest_":false,"type":4,"apsFrame":{"profileId":0,"sequence":144,"clusterId":19,"sourceEndpoint":0,"destinationEndpoint":0,"groupId":0,"options":0},"lastHopLqi":255,"lastHopRssi":-24,"sender":17259,"bindingIndex":255,"addressIndex":255,"message":{"type":"Buffer","data":[16,107,67,102,142,193,174,238,56,193,164,128]}}
2024-02-26T07:20:31.552Z zigbee-herdsman:adapter:ezsp:debg processMessage: {"messageType":4,"apsFrame":{"profileId":0,"sequence":144,"clusterId":19,"sourceEndpoint":0,"destinationEndpoint":0,"groupId":0,"options":0},"lqi":255,"rssi":-24,"sender":17259,"bindingIndex":255,"addressIndex":255,"message":{"type":"Buffer","data":[16,107,67,102,142,193,174,238,56,193,164,128]}}
2024-02-26T07:20:31.552Z zigbee-herdsman:adapter:ezsp:debg ZDO Device announce: 17259, a4c138eeaec18e66
2024-02-26T07:20:31.552Z zigbee-herdsman:adapter:ezsp:debg Device join request received: 17259 a4c138eeaec18e66
2024-02-26T07:20:31.553Z zigbee-herdsman:controller:log Device '0xa4c138eeaec18e66' joined
2024-02-26T07:20:31.553Z zigbee-herdsman:controller:log Device '0xa4c138eeaec18e66' accepted by handler
[... cut ...] "immediate"}) failed (Timeout - 17259 - 1 - 9 - 1280 - 1 after 10000ms)'
Zigbee2MQTT:error 2024-02-26 09:21:22: Failed to interview '0xa4c138eeaec18e66', device has not successfully been paired
Zigbee2MQTT:info  2024-02-26 09:21:22: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c138eeaec18e66","ieee_address":"0xa4c138eeaec18e66","status":"failed"},"type":"device_interview"}'
2024-02-26T07:21:23.429Z zigbee-herdsman:adapter:ezsp:ezsp Time to watchdog ... 0

poughkeepsee, 2024-07-26:
Also interested in support for EZVIZ devices. I just bought a T4C smoke sensor and it's supported in ZHA but not Z2M.

Koenkk, 2024-07-27:
Could you try with the [generated external converter](https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_2-creating-the-external-definition) instead?

closed 2025-01-07 by github-actions[bot]