Koenkk/zigbee2mqtt #20592: IKEA LED2201G8 Become Unsupported Device

ncwhale, 2024-01-05:
### What happened?

After pairing IKEA LED2201G8 (Just brough from IKEA JP) to Zigbee2MQTT successful, the device info said this device is unsupported.

Zigbee2MQTT: 1.35.0 (Just pulled from master branch)

### What did you expect to happen?

This device should be supported as [document](https://www.zigbee2mqtt.io/devices/LED2201G8.html) said. 

BTW, I saw something strange in device model name:

OLD Blubs(which works fine) name: `TRADFRIbulbE26WSglobeopal1160lm`
NEW Blubs(which got issue) name: `TRADFRI bulb E26 WS globe 1160lm` 

 (Yes, there is SPACE between new blubs' name, both of them are LED2201G8, but brought not the same year)

### How to reproduce it (minimal and precise)

1. Enable Zigbee2MQTT pair for all devices.
2. Turn ON/OFF blub 5 times.
3. Blub join zigbee network successful
4. Zigbee2MQTT said it's unsupported.

### Zigbee2MQTT version

 1.35.0 commit: 1652128

### Adapter firmware version

20211116

### Adapter

Zigbee CC2531 USB Dongle

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/13843610/log.txt)

ncwhale, 2024-01-05:
Append: ALL the Exposes works fine, and the  external definition :
```JavaScript
const {light} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['TRADFRI bulb E26 WS globe 1160lm'],
    model: 'TRADFRI bulb E26 WS globe 1160lm',
    vendor: 'IKEA of Sweden',
    description: 'Automatically generated definition',
    extend: [light({"colorTemp":{"range":[250,454]}})],
};

module.exports = definition;
```

Koenkk, 2024-01-06:
Fixed!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-01-06 by Koenkk