Koenkk/zigbee2mqtt #20597: Can't reach mqtt server (mosquito) in 1.35.0

GroGerome, 2024-01-05:
### What happened?

Hello Guys,
Today I updated my running docker to 1.35.0 after that Z2M couldn't connect to my mosquitto docker.
I've restarted the NAS where they are both installed, it connected for a few minute and disconnected and never reconnect.

Downgrading to 1.34.0 solved the issue.
Unfortunately I forgot to save any log.

Has anything changed on the way it connect that would require some new settings for the mqtt server? 
I found an other user on reddit that had the same issue/solution.

### Zigbee2MQTT version

1.35.0

### Adapter firmware version

20221226

### Adapter

 SONOFF TI CC2652P

wifirefly, 2024-01-05:
I'm having the same problem. Below is my log:
```
[22:38:24] INFO: Preparing to start...
[22:38:24] INFO: Socat not enabled
[22:38:25] INFO: Starting Zigbee2MQTT...
Zigbee2MQTT:info  2024-01-05 22:38:27: Logging to console and directory: '/config/zigbee2mqtt/log/2024-01-05.22-38-27' filename: log.txt
Zigbee2MQTT:info  2024-01-05 22:38:27: Starting Zigbee2MQTT version 1.35.0 (commit #unknown)
Zigbee2MQTT:info  2024-01-05 22:38:27: Starting zigbee-herdsman (0.30.0)
Zigbee2MQTT:info  2024-01-05 22:38:27: zigbee-herdsman started (resumed)
Zigbee2MQTT:info  2024-01-05 22:38:27: Coordinator firmware version: '{"meta":{"maintrel":0,"majorrel":38,"minorrel":114,"product":0,"revision":"0x26720700","transportrev":0},"type":"ConBee2/RaspBee2"}'
Zigbee2MQTT:info  2024-01-05 22:38:27: Currently 0 devices are joined:
Zigbee2MQTT:info  2024-01-05 22:38:27: Zigbee: disabling joining new devices.
Zigbee2MQTT:info  2024-01-05 22:38:28: Connecting to MQTT server at //core-mosquitto
Zigbee2MQTT:error 2024-01-05 22:38:28: MQTT failed to connect, exiting...
Zigbee2MQTT:info  2024-01-05 22:38:28: Stopping zigbee-herdsman...
Zigbee2MQTT:info  2024-01-05 22:38:30: Stopped zigbee-herdsman
```

GuitarBingo, 2024-01-06:
I have a similar problem, I have no longer access to the zigbee2mqtt UI. The backend is running, all zigbee devices are working fine. When I try to start the addon it says:

`24-01-06 09:25:21 WARNING (MainThread) [supervisor.addons.addon] 45df7312_zigbee2mqtt is already running!
24-01-06 09:27:21 WARNING (MainThread) [supervisor.addons.addon] Timeout while waiting for addon Zigbee2MQTT to start, took more than 120 seconds `

Organizer21, 2024-01-06:
No idea if related, all my stuff has be down since the 1.3.5 update and my LOG page only says;

[16:16:26] INFO: Preparing to start...
[16:16:27] INFO: Socat not enabled
[16:16:29] INFO: Starting Zigbee2MQTT...

That's all, nothing more... and nothing working.

closed 2024-08-05 by github-actions[bot]