Koenkk/zigbee2mqtt #20599: Aqara E1 QBKG38LM single action only runs once

gbcatrinoiu, 2024-01-05:
### What happened?

I use Aqara E1 no neutral, model QBKG38LM, in decoupled mode. I have an automation that uses single_action trigger to toggle a light. The automation only runs when the light is on and I press the switch. If the light is off, the automation does not run.

To debug, I added a notification every time the automation runs. If the switch is on relay_mode, the automation does run every time the switch is pressed.

Maybe not related, but when I try to change the switch from decoupled_mode to relay_mode I get an error, and it never changes. The last error from he log is related to this.

### What did you expect to happen?

The automation runs every time, while the switch is set to decoupled mode.

### Zigbee2MQTT version

1.35.0 commit: unknown

### Adapter firmware version

7.3.1.0 build 176

### Adapter

Skyconnect

### Debug log

Info 2024-01-05 22:28:17MQTT publish: topic 'zigbee2mqtt/Switch cave', payload '{"action":null,"device_temperature":30,"flip_indicator_light":"ON","linkquality":212,"mode_switch":"quick_mode","operation_mode":"decoupled","power_outage_count":78,"power_outage_memory":true,"state":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
Info 2024-01-05 22:28:24MQTT publish: topic 'zigbee2mqtt/Switch cave', payload '{"action":null,"device_temperature":17,"flip_indicator_light":"ON","linkquality":192,"mode_switch":"quick_mode","operation_mode":"decoupled","power_outage_count":78,"power_outage_memory":true,"state":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
Info 2024-01-05 22:28:37MQTT publish: topic 'zigbee2mqtt/Switch cave', payload '{"action":"single","device_temperature":17,"flip_indicator_light":"ON","linkquality":236,"mode_switch":"quick_mode","operation_mode":"decoupled","power_outage_count":78,"power_outage_memory":true,"state":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
Info 2024-01-05 22:28:37MQTT publish: topic 'zigbee2mqtt/Switch cave', payload '{"action":"","device_temperature":17,"flip_indicator_light":"ON","linkquality":236,"mode_switch":"quick_mode","operation_mode":"decoupled","power_outage_count":78,"power_outage_memory":true,"state":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
Info 2024-01-05 22:28:37MQTT publish: topic 'zigbee2mqtt/Switch cave/action', payload 'single'

Error 2024-01-05 22:36:06Publish 'set' 'operation_mode' to 'Switch cave' failed: 'Error: Write 0x54ef4410009fa453/1 aqaraOpple({"512":{"value":1,"type":32}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 27430 - 1 - 190 - 64704 - 4 after 10000ms)'

nork1n, 2024-01-05:
I have a similar situation with Aqara E1 no neutral, model QBKG39LM (((

perfect-deform, 2024-01-18:
same problem. Sometimes tt switches lights once again, but then throws error again

gbcatrinoiu, 2024-02-09:
I noticed that when turning the lights off, if I turn them back on immediately the action works.

did anyone find a solution for this?

nork1n, 2024-02-09:
After updating z2m to version 1.35.2-1, my keys work perfectly, as before.

Gr4ph1xZ, 2024-02-15:
I am on 1.35.3-1 but I have the same problem :/

chiva, 2024-08-10:
I've been trying to use one QBKG41LM, which is on the same family than the QBKG38LM, but the model with neutral and 2 rockers.

I'm seeing a similar issue on the state reporting, can't get to receive any `action` when using it in decoupled mode.
`action` is supossed to report the state of the rockers, specially relevant when using it on decoupled mode.
https://github.com/Koenkk/zigbee-herdsman-converters/blob/d28d1994b84aad59c044e3445215fd2d2c77b680/src/devices/lumi.ts#L3426
```
            e.action(['single_left', 'double_left', 'single_right', 'double_right', 'single_both', 'double_both']),
```

When switching the state of the relays from Z2M, I can see `state_left: ON` and `state_left: OFF`, but `action` is always `null`.
I know `action` [when used in HomeAssistant mode](https://www.zigbee2mqtt.io/guide/faq/#why-is-the-action-property-always-empty) it's triggered to the appropiate value and then restored back to `null`, but I never see it having a value.

HomeAssistant doesn't show the value ever changing, the `action` entity doesn't have any history, and checking the debug info there's not much there reagrding this value.

I've tried both with the relays on and off, as @gbcatrinoiu suggested, but I'm not getting any `action` triggers back.
Also, I already set in `Settings Specific` the parameter `State action: True`.
[image]
And restarted everything, Z2M, HA, the device... even tried repairing the device, but still nothing.

This is the view of the device State:
```
{
    "flip_indicator_light": "ON",
    "linkquality": 84,
    "operation_mode_left": "decoupled",
    "operation_mode_right": "decoupled",
    "power_outage_memory": true,
    "state_left": "ON",
    "state_right": "OFF",
    "update": {
        "installed_version": 25,
        "latest_version": 25,
        "state": "idle"
    },
    "action": null,
    "device_temperature": null,
    "update_available": null
}
```

Any ideas on how to resolve this issue?

@gbcatrinoiu not trying to hijack your issue, guess it might be due to the same root cause?
Where does `single_action` appear? Is it an `action` of the device?
If it's a different issue I'll open a new issue.

chiva, 2024-08-11:
Everything got fixed when updating the controller firmware, bought recently but had an ancient firmware version running

Elphman, 2024-12-13:
I run into similar issue and i find the solution by using a capacitor from another smart switch `Moes ZS-EUB_1gang` and installed it into my ceiling rose.(no noice and not getting hot) And it's working fine in decoupled mode. Decoupled mode without capacitor running but only when the smart bulb is on. So if you turned smart bulb off , switch will not turn it back on and will give in zigbee2mqtt errors. Like i understood it's all about a smart bulb low wattage in some forums someone mentioned about it my run without capacitor if you control several smart bulbs from same `no neutral` switch. Hope it`s helps someone. I would like to know if someone able to manage running it wothout capacitor and would be thankful if you`ll explain how.Thanks!

closed 2025-02-20 by github-actions[bot]