Koenkk/zigbee2mqtt #20601: OTA - Failed to check if update available for Ikea and Philips Hue devices (Cannot read properties of undefined (reading 'data'))

phbasler, 2024-01-05:
### What happened?

I have two instances running the docker container of zigbee2MQTT in my home. One with debian 11 and the other one on a raspberry pi 4 64 bit.
On my raspberry pi one I can check for OTA Updates but on the debian one I get the error message 
`Failed to check if update available for device (Cannot read properties of undefined (reading 'data'))`

I have no problems connecting to the web from my debian server but I am confused why  it does not work on this instance but on the other.

### What did you expect to happen?

Not throwing an error message.

### How to reproduce it (minimal and precise)

Click on `Check for Updates` for a device.

### Zigbee2MQTT version

1.35.0

### Adapter firmware version

6.10.3.0 build 297

### Adapter

Sonoff Zigbee Dongle E

### Debug log

info 2024-01-06 00:25:00MQTT publish: topic 'zigbee2mqtt/bridge/response/options', payload '{"data":{"restart_required":false},"status":"ok","transaction":"8v75c-5"}'
debug 2024-01-06 00:25:03Received MQTT message on 'zigbee2mqtt/bridge/request/device/ota_update/check' with data '{"id":"Büro 2","transaction":"11oy0-5"}'
info 2024-01-06 00:25:03Checking if update available for 'Büro 2'
debug 2024-01-06 00:25:03Check if update available for '0x804b50fffeef6957' (TRADFRI bulb GU10 WS 400lm)
debug 2024-01-06 00:25:03Using endpoint '1'
debug 2024-01-06 00:25:03Received Zigbee message from 'Büro 2', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":587798065,"imageType":16900,"manufacturerCode":4476}' from endpoint 1 with groupID 0
debug 2024-01-06 00:25:03Got OTA request '{"fieldControl":1,"manufacturerCode":4476,"imageType":16900,"fileVersion":587798065}'
debug 2024-01-06 00:25:05Saving state to file /app/data/state.json
debug 2024-01-06 00:25:06Received Zigbee message from 'Büro 2', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":587798065,"imageType":16900,"manufacturerCode":4476}' from endpoint 1 with groupID 0
info 2024-01-06 00:25:08MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{"id":"Büro 2"},"error":"Failed to check if update available for 'Büro 2' (Cannot read properties of undefined (reading 'data'))","status":"error","transaction":"11oy0-5"}'
error 2024-01-06 00:25:08Failed to check if update available for 'Büro 2' (Cannot read properties of undefined (reading 'data'))
debug 2024-01-06 00:25:08TypeError: Cannot read properties of undefined (reading 'data') at getImageMeta (/app/node_modules/zigbee-herdsman-converters/src/lib/ota/tradfri.ts:15:43) at processTicksAndRejections (node:internal/process/task_queues:95:5) at isNewImageAvailable (/app/node_modules/zigbee-herdsman-converters/src/lib/ota/common.ts:347:18) at Object.isUpdateAvailable (/app/node_modules/zigbee-herdsman-converters/src/lib/ota/common.ts:338:29) at OTAUpdate.onMQTTMessage (/app/lib/extension/otaUpdate.ts:205:45)
debug 2024-01-06 00:25:09Received Zigbee message from 'Büro 2', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":587798065,"imageType":16900,"manufacturerCode":4476}' from endpoint 1 with groupID 0
debug 2024-01-06 00:25:09Device 'Büro 2' requested OTA
debug 2024-01-06 00:25:12Received Zigbee message from 'Büro 2', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":587798065,"imageType":16900,"manufacturerCode":4476}' from endpoint 1 with groupID 0
debug 2024-01-06 00:25:12Device 'Büro 2' requested OTA
debug 2024-01-06 00:25:15Received Zigbee message from 'Büro 2', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":587798065,"imageType":16900,"manufacturerCode":4476}' from endpoint 1 with groupID 0
debug 2024-01-06 00:25:15Device 'Büro 2' requested OTA

jfidot, 2024-01-25:
Hi,

any news on this?

I have similar issue with the OTA Updates. Currently I get following error:

```Failed to check if update available for 'Büro Steckdose' (Cannot destructure property 'data' of '(intermediate value)' as it is undefined.)```

Previously I got the same error message as you.

I have HA, Zigbee2MQTT and mosquitto (newest versions) installed as a docker containers on Raspberry PI 4.
I'm using different version of the zigbee adapter (Sonoff Zigbee Dongle P).

iverona, 2024-05-11:
I've the same (or similar) issue. It happens with both IKEA and Philipps devices (I don't have other brands on my setup). I've tried with the test server and fails both when checking for new firmware and when trying to update. 

Maybe I'm missing  config variable? I'm running with docker compose and currently on version `1.37.1 commit ea39d86`

Thanks! 

```
zigbee2mqtt  | [2024-05-11 22:57:17] error: 	z2m: Failed to check if update available for 'ilc16_tira_dormitorio' (Cannot destructure property 'data' of '(intermediate value)' as it is undefined.)
zigbee2mqtt  | [2024-05-11 22:57:17] debug: 	z2m: TypeError: Cannot destructure property 'data' of '(intermediate value)' as it is undefined.
zigbee2mqtt  |     at getIndex (/app/node_modules/zigbee-herdsman-converters/src/lib/ota/zigbeeOTA.ts:42:18)
zigbee2mqtt  |     at processTicksAndRejections (node:internal/process/task_queues:95:5)
zigbee2mqtt  |     at getImageMeta (/app/node_modules/zigbee-herdsman-converters/src/lib/ota/zigbeeOTA.ts:61:20)
zigbee2mqtt  |     at Object.isNewImageAvailable (/app/node_modules/zigbee-herdsman-converters/src/lib/ota/common.ts:406:18)
zigbee2mqtt  |     at Object.isUpdateAvailable (/app/node_modules/zigbee-herdsman-converters/src/lib/ota/common.ts:395:29)
zigbee2mqtt  |     at OTAUpdate.onMQTTMessage (/app/lib/extension/otaUpdate.ts:209:45)
zigbee2mqtt  |     at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17)
```

B3WiN, 2024-10-06:
With Ikea Parasoll Door sensor first I have to wake up the sensor with an opening, and then I can run an update search without any error message.

lolorc, 2024-10-23:
same issue here with all my hue bulbs, running dev from today, bulbs are Philips model 9290022169, running firmware with this build date: 20240329

closed 2025-01-07 by github-actions[bot]