Koenkk/zigbee2mqtt #20605: Neo NAS-AB06B2 always no battery

kwetnico, 2024-01-06:
### What happened?

Hello since version 1.33 [#19077](https://github.com/Koenkk/zigbee2mqtt/issues/19077) external siren is supported all the features work well but the battery percentage was displayed since 1.34 apparently this no longer works.
Could you help because an external siren on battery without knowing its charge is very problematic.
Thanks
[image]

### What did you expect to happen?

That the battery information is reported

### Zigbee2MQTT version

1.35

### Adapter firmware version

 6.10.3.0 build 297

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus-E

emilianogetino, 2024-01-12:
I have the same problem, any solution? thank you.

emilianogetino, 2024-01-12:
the same to me

emilianogetino, 2024-02-07:
Version 1.35.3 todo igual.

kwetnico, 2024-02-07:
I still confirm no information on the battery level. :(
@Koenkk  could you fix this problem, the siren is an important element of the alarm system and if not we do not know the battery level this is a real problem. Thanks

constapel, 2024-02-11:
Please raise importancy, urgency en priority!

HA-FAMILLEPHC, 2024-03-05:
Hi,
I've been starting HA for a few months now.
I encounter the same problem, the battery percentage which does not go up in Z2MQTT.
and I have another question regarding charging by the solar panel, does it appear in Z2MQTT when charging is done via the solar panel.
Thanks for your help

kwetnico, 2024-03-05:
Not appear in z2m the charging with solar panel

Koenkk, 2024-06-08:
Was the battery % working in z2m 1.33? Can someone downgrade to that version and confirm that it works there.

kwetnico, 2024-12-06:
No problem still current and nothing is done to correct it. Several topics open. The battery percentage information is still current

Sabjul33, 2025-02-01:
News for me with the 2.1.0 update !!!
https://github.com/Koenkk/zigbee2mqtt/issues/19827#issuecomment-2629130678

Miguelin96, 2025-03-09:
I have the same problem, is there a solution?

Sabjul33, 2025-03-09:
AS explained, it's work with the new version of Z2M.

Miguelin96, 2025-03-09:
I have this version of zigbee2mqtt

[image]

kwetnico, 2025-03-09:
I have the latest version of z2m: 2.1.3 and it still doesn't work !

[image]

Sabjul33, 2025-03-09:
I have this version with an other coordinator

[image]

SirSheik, 2025-03-09:
> I have this version with an other coordinator
> 
> [image]

Your version is Model NAS-AB06B2?
Because that version for which this topic was opened still does not report the battery.
Maybe you have another model or version or there are several models, although physically they are the same

Sabjul33, 2025-03-09:
My model
Fabriquant
Neo
Modèle
NAS-AB06B2
Modèle Zigbee
TS0601
Fabricant Zigbee
_TZE200_nlrfgpny
Description
Outdoor solar alarm

Miguelin96, 2025-03-09:
My alarm model

[image]

Miguelin96, 2025-03-13:
Some news about the battery report, for me it is very important that you report that value

Rushmed, 2025-07-18:
Im also interested.

brusini, 2025-09-19:
Same issue in 2.6.1-1. Any updates\fixes?

auanasgheps, 2026-01-18:
> Wow. I can't believe it but the battery percentage HAS FINALLY SHOWN UP! My manufaturer is the same as before i.e. `_TZE284_nlrfgpny` which was unreported as working as the sirens from [@gregoriusus](https://github.com/gregoriusus) and others, which were of different manufacturers.

> I can confirm that their strategy is the way to go, i.e., keep the siren out of direct sunlight, and very importantly as well, keep constantly activating the lights to drain the battery as much as possible. I ended up doing this last step manually, but as said, it'd be useful to be able to do this with a script in an automated way. Hope this helps others.

If my theory is correct, the battery is reported ONLY when below 100%.
If you put the siren out to the sun and recharge it, it will stay stuck at 98% but in reality will be fully charged.

gregoriusus, 2026-01-18:
I agree with you theory.

Just for info. How long have you been playing with lights until report?