Koenkk/zigbee2mqtt #20610: [Feature request]: get a list of groups via mqtt, get a list of devices in the group via mqtt

grafalex82, 2024-01-06:
### Is your feature request related to a problem? Please describe

Current MQTT API has a way to add and remove a group, but there is no way to get a list of groups.
Same for group membership - you can add and remove a device to a group, but you cannot get a list of devices on the group.

These getters could be useful for various scripting purposes to set up groups and membership in an automatic way.

Particularly I am developing a [set of automated tests](https://github.com/grafalex82/hellozigbee/blob/main/doc/part21_test_automation.md) for my device. A subset of tests suppose to check groups functionality of my device. In order to do this a test group shall be created, and device endpoints shall be added to this group. In order to ensure group and group membership is correct it would be nice to get current state of the group, and skip setting up the group if everything is set already.

### Describe the solution you'd like

Add the following mqtt topics:
- `zigbee2mqtt/bridge/request/group/list` that would return a list of registered groups
- `zigbee2mqtt/bridge/request/group/members/list` that would return a list of devices or device endpoints on the group

### Describe alternatives you've considered

Currently I have to use one of the 2 approaches:
- Create a group regardless of whether it exists. Skip error in case group exists. Same for group membership - add group membership regardless whether the device is already a member of the group
- Blindly delete and then re-create the group. Ignore deletion errors, if any. Same for group membership - remove device from the group and then re-add it again, ignore errors if any.

### Additional context

I could probably help with implementing this feature, if someone could point me to the right files and data structures to use.

Koenkk, 2024-01-06:
See https://www.zigbee2mqtt.io/guide/usage/mqtt_topics_and_messages.html#zigbee2mqtt-bridge-groups

grafalex82, 2024-01-06:
Thanks for pointing me to this. That is what I need.

Perhaps this link shall be mentioned at https://www.zigbee2mqtt.io/guide/usage/groups.html as well. I was studying that page and did not get any knowledge how to get groups list.

closed 2024-01-06 by grafalex82