Koenkk/zigbee2mqtt #20614: TS0601 - heating attribute not working

esben, 2024-01-06:
### What happened?

I have a couple of TS0601 (_TZE200_c88teujp) radiator thermostats.
The heating attribute which should show "Device valve is open or closed (heating or not)" always report "null".

I am totally new to zigbee2mqtt (and not a js/ts developer), but the relevant lines: https://github.com/Koenkk/zigbee-herdsman-converters/blob/b881ab3ece047f96ca55ad15521772d56112abed/src/lib/legacy.ts#L4116 and https://github.com/Koenkk/zigbee-herdsman-converters/blob/b881ab3ece047f96ca55ad15521772d56112abed/src/lib/legacy.ts#L4116 looks a bit strange. In all other similar lines in legacy.ts, the keyword ("heating" here) is not put in quote. Is this simply a typo?

### What did you expect to happen?

The heating attribute exposed should show "ON" or "OFF"

### How to reproduce it (minimal and precise)

Add a _TZE200_c88teujp device and look at the heating value. Either in webinterface or directly on mqtt broker. It is always "null".

### Zigbee2MQTT version

1.35.0

### Adapter firmware version

0x26580700

### Adapter

ConBee2/RaspBee2

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/13849679/log.txt)

ASNNetworks, 2024-01-06:
Could be related to this:

esben, 2024-01-06:
It doesn't sound like the same issue. Everything seems to work fine. It is just the "heating" attribute that doesn't work.

I am now testing with 1.35.0-dev commit: [14ac9a8](https://github.com/Koenkk/zigbee2mqtt/commit/14ac9a8), and the issue is still there.

esben, 2024-01-06:
The same problem was also mentioned in https://github.com/Koenkk/zigbee2mqtt/discussions/14295 (last line in the issue description).

sc-ampersand-p, 2024-03-04:
@esben I see the same thing ("heating" is null), though it never bothered me until now. I also have the larger problem in #14295, but I think that all goes back to the fact that z2m does not allow any kind of interaction with the "auto" mode (which the official TuYa hub probably allows, I wouldn't know). 

My _guess_ is that the "heating" attribute becomes especially important when the "auto" mode is properly programmed. I opened an [issue](https://github.com/Koenkk/zigbee-herdsman-converters/issues/3307) about the weekly schedule/auto mode a long time ago, but nothing ever came of it. 

Interacting with this feature was probably never implemented, and considering Homeassistant (which is where I use these TRVs) can do all the scheduling, I don't mind it _that_ much. But honestly, I'd personally love to see this properly working, too.

closed 2024-10-02 by github-actions[bot]