Koenkk/zigbee2mqtt #20615: [Feature request]: Ability to queue changes and updates (especially for battery powered devices)

FrederikNS, 2024-01-06:
### Is your feature request related to a problem? Please describe

Currently when you want to change a setting, establish a binding or trigger an OTA update for a device that is battery powered, you need to also trigger the battery powered device somehow to wake it up, such that it actually receives the request.

This process is a bit annoying, when you want to make a binding for a battery powered switch such as the IKEA Trådfri ones. Here you need to actually press the switch close to when you want to issue the binding. The same thing goes for OTA updates and other setting changes.

But it's much worse with stuff like motion sensors, where you risk triggering the sensor, and it going to sleep, and you now need to wait for it to time out before you can wake it up again.

And downright infuriating for other devices such as a temperature and humidity sensor like the Sonoff SNZB-02, where there isn't really a good way to wake up the sensor... you will just have to keep sending the request, and hope that the sensor happens to wake up while you are sending the request.

### Describe the solution you'd like

I would like to be able to enqueue requests, such as bindings, OTA updates, and settings changes. For example, if I wanted to issue a binding for a switch, then I should just be able to go in the UI and issue the binding. This request for a binding should then go into a queue, and sit there waiting. Then when the switch wakes up (either on it's own or because it was pressed) zigbee2mqtt should detect that the switch woke up, and check it's queue for whether that device had anything pending, and immediately issue those pending requests.

This would allow you to sit and look at something like your OTA update list, and immediately request all the updates to be run. Then when each device wakes up in the list zigbee2mqtt will trigger the update for each device.

### Describe alternatives you've considered

The only real alternative I can see would be to be able to somehow wake up a device remotely, but I don't think that is actually possible.

### Additional context

[image]

Asmod4n, 2024-01-06:
The Ikea tradfri gateway App does it that way, when it wants to update a device it just waits till the device gets waken up and does the update in the background.

It would also be good to define bindings in advance, issung the bindings while adopting the device in z2m automatically.
Something like user defineable templates for default settings of new adopted devices

svenwanzenried, 2024-01-15:
This is exactly the problem I'm facing, and also exactly the solution I'd wish to be implemented. 👍
I am not sure, how Z2M is handling the configurations change currently. Does it try to retransmit the package until the timeout occurs?. If so, maybe as a temporary workaround it could be feasible to increase that timeout to something like multiple hours and the retransmit frequency to something like once every second? It would be very much spamming on the network but at least it would transmit eventually.

closed 2024-08-13 by github-actions[bot]

nowherenearaninja, 2024-12-02:
Same for me, wouöld really like to see this. Have bought 12 cheap Tuya temperature sensors that now have an update. Have not found a way to wake them up, when I press the only button, they unpair.
Not only that, would make all updates of battery devices much more smooth. It is very rare that the update needs to happen exactly when you push the update button.