Koenkk/zigbee2mqtt #20617: Regression on pairing some Legrand devices with v1.35.0 (was somewhat working on 1.34.0)

Xavinooo, 2024-01-06:
### What happened?

Since v1.35.0, the pairing of [Legrand 067776A](https://www.zigbee2mqtt.io/devices/067776A.html) is not working anymore.
It was somewhat working on v1.34.0 with [this method](https://github.com/Koenkk/zigbee2mqtt/issues/16948#issuecomment-1680592070).

### What did you expect to happen?

The pairing of the device, in one step, or at least, like it was in v1.34.0

### How to reproduce it (minimal and precise)

(with both version, I started with a fresh z2m, running in docker)
With version 1.34.0, I had to do : 
- reseting the device (fixed red led)
- start z2m container
- permit join
- push the pairing button on the Legrand device (starts blinking green)
- once z2m successufully configured the device, it's still blinking green (like [here ](https://github.com/Koenkk/zigbee2mqtt/issues/19108) and [here](https://github.com/Koenkk/zigbee2mqtt/issues/16948))
- stop z2m container
- resetting the device (fixed red led)
- start z2m container
- push the pairing button (green blinking led)
- permit join

then, after a few seconds, it was fully paired, and the led was solid green.

The same steps with v1.35.0 are not working anymore. On the second pair, it leaves the network and rejoins, but keeps blinking red.

I've attatched the 2 logs of each versions.

### Zigbee2MQTT version

1.35.0

### Adapter firmware version

 6.10.3.0 build 297

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus

### Debug log

[1.34_second_pair.txt](https://github.com/Koenkk/zigbee2mqtt/files/13850050/1.34_second_pair.txt)
[1.34_first_pair.txt](https://github.com/Koenkk/zigbee2mqtt/files/13850052/1.34_first_pair.txt)
[1.35_first_pair.txt](https://github.com/Koenkk/zigbee2mqtt/files/13850051/1.35_first_pair.txt)
[1.35_second_pair.txt](https://github.com/Koenkk/zigbee2mqtt/files/13850053/1.35_second_pair.txt)

Eoras, 2024-01-07:
Hello, I have same probleme here :/

I'm with HA with Z2M 1.35.1

When i want to pair LEGRAND teleruptor 412170, It works very well, it is recognized and configured, and after a few seconds, the remote control resets and leaves the network.

The LED never turns green on this teleruptor (light toggle switch)

```
Zigbee2MQTT:info  2024-01-07 18:36:29: Device '0x0004740001241a63' joined
Zigbee2MQTT:info  2024-01-07 18:36:29: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0004740001241a63","ieee_address":"0x0004740001241a63"},"type":"device_joined"}'
Zigbee2MQTT:info  2024-01-07 18:36:29: Starting interview of '0x0004740001241a63'
Zigbee2MQTT:info  2024-01-07 18:36:29: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0004740001241a63","ieee_address":"0x0004740001241a63","status":"started"},"type":"device_interview"}'
Zigbee2MQTT:info  2024-01-07 18:36:29: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0004740001241a63","ieee_address":"0x0004740001241a63"},"type":"device_announce"}'
Zigbee2MQTT:info  2024-01-07 18:36:30: Successfully interviewed '0x0004740001241a63', device has successfully been paired
Zigbee2MQTT:info  2024-01-07 18:36:30: Device '0x0004740001241a63' is supported, identified as: Legrand DIN smart relay for light control (412170)
Zigbee2MQTT:info  2024-01-07 18:36:30: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"DIN smart relay for light control","exposes":[{"features":[{"access":7,"description":"On/off state of the switch","label":"State","name":"state","property":"state","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"},{"access":7,"description":"On/off (works only if device is in \"switch\" mode)","label":"State","name":"state","property":"state","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"access":5,"description":"Instantaneous measured power","label":"Power","name":"power","property":"power","type":"numeric","unit":"W"},{"access":7,"description":"switch: allow on/off, auto will use wired action via C1/C2 on teleruptor with buttons","label":"Device mode","name":"device_mode","property":"device_mode","type":"enum","values":["switch","auto"]},{"access":1,"description":"Link quality (signal strength)","label":"Linkquality","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"412170","options":[{"access":2,"description":"Calibrates the power value (percentual offset), takes into effect on next report of device.","label":"Power calibration","name":"power_calibration","property":"power_calibration","type":"numeric"},{"access":2,"description":"Number of digits after decimal point for power, takes into effect on next report of device. This option can only decrease the precision, not increase it.","label":"Power precision","name":"power_precision","property":"power_precision","type":"numeric","value_max":3,"value_min":0},{"access":2,"description":"State actions will also be published as 'action' when true (default false).","label":"State action","name":"state_action","property":"state_action","type":"binary","value_off":false,"value_on":true}],"supports_ota":true,"vendor":"Legrand"},"friendly_name":"0x0004740001241a63","ieee_address":"0x0004740001241a63","status":"successful","supported":true},"type":"device_interview"}'
Zigbee2MQTT:info  2024-01-07 18:36:30: Configuring '0x0004740001241a63'
Zigbee2MQTT:info  2024-01-07 18:36:30: Successfully configured '0x0004740001241a63'
Zigbee2MQTT:warn  2024-01-07 18:36:40: Device '0x0004740001241a63' left the network
```

Xavinooo, 2024-01-07:
I've updated to v1.35.1, and I have again the same behaviour than v1.34.0.
So, maybe it got "fixed" ?

Xavinooo, 2024-01-07:
@Eoras Did you try these steps ? : 

- reseting the device (fixed red led)
- start z2m container
- permit join
- push the pairing button on the Legrand device (starts blinking green)
- once z2m successufully configured the device, it's still blinking green (like https://github.com/Koenkk/zigbee2mqtt/issues/19108and https://github.com/Koenkk/zigbee2mqtt/issues/16948)
- stop z2m container
- resetting the device (fixed red led)
- start z2m container
- push the pairing button (green blinking led)
- permit join

Eoras, 2024-01-07:
> I've updated to v1.35.1, and I have again the same behaviour than v1.34.0. So, maybe it got "fixed" ?
No, 1.35.1 don't solve nothing :( i restart all system and do same.

I try to do this but i think my problem is not same as you ^^

-> Restore the device (fixed red led)
-> permit join on Z2M
-> dive is paired, all configuration is ok but led stay RED on light switch toggle (LEGRAND 412170)
-> after 5/6 second light on and off alone, and device left Z2M.
-> The toggle switch resets and leaves as if it did not validate the pairing

this repeat until i'm on permit joint on Z2M.

CptAdhoc, 2024-01-07:
Hello,
same behavior here with Z2M 1.35 and 1.35.1 and blind command LEGRAND 600086A ( identified by Z2M as 067776A, probably the same device behind the faceplate ).

If paired with red led, the led will stay red.
If paired after pressing the button, the led keeps blinking green.

After power loss, the device will leave the network.
It also seems that the calibration process can't be started before the device is properly paired.

Trying @Xavinooo procedure, at the end, the device will leave the network and immediatly join with red led.

nicozocco, 2024-01-13:
Hi! I've spent a whole week trying to solve this.

I have the same issue ZHA was able to pair the devices properly, Z2M won't pair them.

I have the SONOFF Zigbee 3.0 P as well and several legrand/bticino switches, I've tried with a wireless switch and an shutter and both had the same issue.

I managed to keep the devices paired by following the steps mentioned above but only until the first reset:

reseting the device (fixed red led)
start z2m container
permit join
push the pairing button on the Legrand device (starts blinking green)
once z2m successufully configured the device, it's still blinking green (like 
stop z2m container

After this the devices stay connected, but i think they are probably not fully paired as after some time without using them, when i first push the button it does nothing at all, i need to press it a couple of times for it to react.

CptAdhoc, 2024-01-14:
I have installed 1.34.0 on a seperate VM and imported the configuration from the Home Assistant Z2M addon.
Under this version, I have followed @Xavinooo pairing process with succes ( making the calibration possible ).

I have re-imported the configuration of Z2M on the VM to the Home Assistant addon under 1.35.1.
The switch seems to behave correctly for now.

Xavinooo, 2024-01-14:
With 1.35.1, the process is again working ;)

CptAdhoc, 2024-01-14:
> With 1.35.1, the process is again working like with 1.34.0 ;)

Unfortunately, on my side, it wasn't successful under 1.35.1, that why I ended going back to 1.34.0.

CptAdhoc, 2024-01-21:
Hello,

New try today with a Dooxie Light switch without neutral 199232, reported by Z2M as 199182, version 0012 / 20221114.
The behaviour is a little bit different on 1.35.1: the device will pair without change on the led behaviour, exactly like the blind sensor, but will leave the network few seconds later.

I paired it successfully under 1.34.0, again with @Xavinooo process, and import the configuration under 1.35.1 with no problem.

sjorge, 2024-03-16:
This is probably related to https://github.com/Koenkk/zigbee-herdsman/issues/979, until this is fixed you can theoretically manually comment the readResponse coming from inside zh. I've tried this and then the device pairs again.

nicozocco, 2024-03-16:
@sjorge  That's interesting I did some testing and decided to roll back to 1.34, then paired all the legrand devices.

Funny enough this version seems to work pretty well with these devices and it also looks like the devices believe they're connecting to the Legrand Gateway, as they flash green when completely paired vs magenta when paired in a higher version (I understsand from their [documentation](https://developer.legrand.com/local-interoperability/) magenta flash is when the Legrand devices successfully pairs in a Zigbee 3.0 network).
A benefit of this is that he switches keep on working with the Toggle state instead of On/Off, this means you can press anywhere in the switch instead of top or bottom. This is also the standard behaviour when connected to the Legrand gateway.

sjorge, 2024-03-16:
With the double readResponse gone (by eliminating the call in zh) mine shows up as green, same as when it used to when I originally paired it a year or so ago.

closed 2024-10-14 by github-actions[bot]