Koenkk/zigbee2mqtt #20618: Unable to Configure/Reconfigure Sonoff Plugs (S31ZB) in 1.3.5

sirjmann92, 2024-01-06:
### What happened?

Since the update to 1.3.5 I am unable to configure or reconfigure Sonoff S31ZB plugs. I have four of them and all of them exhibit the exact same behavior. I've tried to repair them, same symptom. I've tried forcing them off the network and then pairing again, same symptom. In 1.3.4 I was able to configure and reconfigure these plugs without error.  See logs and details below.

Zigbee2MQTT via Docker
- Zigbee2MQTT version 1.35.0 (tried both 1.35.0 commit: b5a90e9 and 1.35.0-dev commit: 14ac9a8)
- Coordinator type zStack3x0
- Coordinator revision 20230507
- Frontend version 0.6.150
- Zigbee-herdsman-converters version 18.3.0
- Zigbee-herdsman version 0.30.0
Home Assistant version:
- Core 2024.1.0
- Frontend 20240103.3
Mosquitto broker via Docker (latest)
UZG-01 Coordinator
- Model: | UZG-01
- ESP32 Firmware ver: | 0.1.5 (2023-09-06 12:01:02.723464)
- ESP32 version: | ESP32-D0WD
- ESP32 temperature: | 33.67 °C
- ESP32 frequency: | 2 cores @ 160 MHz
- ESP32 flash size: | 4 Mb, external
- ESP32 Free heap: | 272 / 337 KiB
- CC2652P version: | CC2652PP1FRGZR
- CC2652P flash size: | 352 Kb

### What did you expect to happen?

Expected behavior: No errors on initial configure or reconfigure of Sonoff S31ZB smart plugs.

### How to reproduce it (minimal and precise)

1. Plug in a Sonoff S31ZB plug
2. Press "Permit Join (All)" to allow new device joining in zigbee2mqtt
3. Device is discovered and interviewed
4. Observe error during configure stage

*OR* (for a device already on the network)
1. Locate the Sonoff S31ZB plug in zigbee2mqtt
2. Press the "Reconfigure" button
3. Observe error during reconfigure

### Zigbee2MQTT version

1.35.0 commit: b5a90e9 and 1.35.0-dev commit: 14ac9a8

### Adapter firmware version

20230507

### Adapter

UZG-01 (zStack3x0)

### Debug log

On reconfigure:
`2024-01-06 08:25:30Failed to configure 'Sonoff Plug 03', attempt 6 (Error: ConfigureReporting 0x00124b002a63371e/1 genOnOff([{"minimumReportInterval":0,"maximumReportInterval":65000,"reportableChange":1,"attribute":"onOff"}], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 8111 - 1 - 129 - 6 - 7 after 10000ms) at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35) at listOnTimeout (node:internal/timers:569:17) at processTimers (node:internal/timers:512:7))`

On first join:
`2024-01-06 08:13:34 Failed to configure 'Sonoff Plug 03', attempt 4 (Error: Bind 0x00124b002a63371e/1 genOnOff from '0x00124b002a7b524f/1' failed (AREQ - ZDO - bindRsp after 10000ms) at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35) at listOnTimeout (node:internal/timers:569:17) at processTimers (node:internal/timers:512:7))`

On reboot/restart:
`Error 2024-01-06 09:00:15Failed to configure 'Sonoff Plug 03', attempt 2 (Error: ConfigureReporting 0x00124b002a63371e/1 genOnOff([{"minimumReportInterval":0,"maximumReportInterval":65000,"reportableChange":1,"attribute":"onOff"}], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 8111 - 1 - 38 - 6 - 7 after 10000ms) at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35) at listOnTimeout (node:internal/timers:569:17) at processTimers (node:internal/timers:512:7))
Info 2024-01-06 09:00:15MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure', payload '{"data":{"id":"Sonoff Plug 03"},"error":"Failed to configure (ConfigureReporting 0x00124b002a63371e/1 genOnOff([{\"minimumReportInterval\":0,\"maximumReportInterval\":65000,\"reportableChange\":1,\"attribute\":\"onOff\"}], {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Timeout - 8111 - 1 - 38 - 6 - 7 after 10000ms))","status":"error","transaction":"b6hat-2"}'
Info 2024-01-06 09:00:26Configuring 'Sonoff Plug 04'
Error 2024-01-06 09:00:49Failed to configure 'Sonoff Plug 04', attempt 3 (Error: ConfigureReporting 0x00124b002a638753/1 genOnOff([{"minimumReportInterval":0,"maximumReportInterval":65000,"reportableChange":1,"attribute":"onOff"}], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 19401 - 1 - 44 - 6 - 7 after 10000ms) at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35) at listOnTimeout (node:internal/timers:569:17) at processTimers (node:internal/timers:512:7))`

Koenkk, 2024-01-07:
When factory reseting the device/re-pairing it, does the state update when physically controlling the switch (even when reconfigure fails?)

sirjmann92, 2024-01-07:
I have a couple of the plugs that I haven't reset and a couple that I have. The state and LQI still update and the device still functions, both before and after a factory reset. Mapping also still works for these devices. The map shows their routes and end devices still appear as connected to the plugs (both before and after reset). Toggling state manually is still working as well as all would-be-affected automations.

I think it's just an annoying error/message that doesn't affect the behavior/performance of the device - but I'm not a developer (I work on the business side of software dev as a Product Manager) so I can't say if something else internal to the application might be affected by this defect. Other people may not go as far in testing and might just revert to 1.34 (as baudfather did) - or worse, pay for more/other devices and replace their S31ZB plugs not understanding the affect (or lack) of the issue.

I appreciate your prompt response and everything you do on this app, @Koenkk. It really makes our lives easier (and less expensive!)

ghoulsfrank, 2024-01-07:
Same issue here, going back to 1.34 resolve the issue for me

gorddel, 2024-01-07:
Same as above on both 1.35 and 1.35.1 on my S31ZB's. 
On restart:
`Error 2024-01-07 12:37:27Failed to configure 'ZB-SO-SP-001', attempt 1 (Error: ConfigureReporting 0x00124b00251b5c09/1 genOnOff([{"minimumReportInterval":0,"maximumReportInterval":65000,"reportableChange":1,"attribute":"onOff"}], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 23413 - 1 - 3 - 6 - 7 after 10000ms) at Timeout._onTimeout (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35) at listOnTimeout (node:internal/timers:569:17) at processTimers (node:internal/timers:512:7))`

The errors continue for several retries on each device. Other than the logged errors, the devices appear to be operating completely normally (although I only have 2 devices which are lightly used).
Thanks for all your effort and this great software!

star114, 2024-01-08:
similar issue I reported : #20666 (not timeout, SOFTWARE_FAILURE)

Koenkk, 2024-01-09:
Fixed!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-01-09 by Koenkk

sirjmann92, 2024-01-09:
Thanks @Koenkk - I'll give it a shot after work this evening and report back.