Koenkk/zigbee2mqtt #20619: Impossible change preset mode in Moes BHT-002-GCLZB device

jvcsw, 2024-01-06:
### What happened?

Impossible change preset mode from "program" to "hold".

A few weeks ago this worked.

I tried with HA and Z2M UI.

You can change the value from any UI but when refresh it the value return to "program" again.

Device: 
https://www.zigbee2mqtt.io/devices/BHT-002-GCLZB.html#moes-bht-002-gclzb

### What did you expect to happen?

I want to change preset mode from "program" to "hold".

### How to reproduce it (minimal and precise)

Configuring this device from Z2M UI.

### Zigbee2MQTT version

1.35.0-dev

### Adapter firmware version

20211210

### Adapter

zzh

### Debug log

no error in traces

luism2p, 2024-01-12:
.... same with BHT-006... Un saludo

jvcsw, 2024-01-16:
@Koenkk Could you help us when you have a moment? Thanks

Koenkk, 2024-01-17:
What's the last known working z2m version?

jvcsw, 2024-01-18:
I think 1.34, 1.33 sure

richsimpson, 2024-01-18:
Yes noticed the same, sorry can;t confirm when it last worked but recently I would say within last couple of weeks. I tend to have latest z2m updates always if that is any help for time line. 

Electric floor heating BHT-002-GCLZB 

Click Program and it ignores the command and stays in Hold. On Z2M the option turns grey but makes no change and leave the page and return shows Hold again.

Koenkk, 2024-01-19:
@jvcsw could you provide the debug log of setting this with both 1.35.1 and 1.33.0?

See [this](https://www.zigbee2mqtt.io/guide/usage/debug.html) on how to enable debug logging.

jvcsw, 2024-01-19:
I attach logs filtered by device while I tried to change preset mode.

[Log 1.33.0 .txt](https://github.com/Koenkk/zigbee2mqtt/files/13995555/Log.1.33.0.txt)
[Log 1.35.1 .txt](https://github.com/Koenkk/zigbee2mqtt/files/13995556/Log.1.35.1.txt)

Koenkk, 2024-01-20:
@a2nt this is due to https://github.com/Koenkk/zigbee-herdsman-converters/pull/6654, `dataPoints.tvMode` and `dataPoints.moesHold` are both 2, meaning that https://github.com/Koenkk/zigbee-herdsman-converters/blob/8ff72e77e2d09bb217f2b8bf16a95d99d17a92fb/src/lib/legacy.ts#L3899 is never executed anymore and thus `preset` is never set anymore, could you take a look at this?

jvcsw, 2024-02-01:
Hi again, 

Is there any news on this topic?

Thanks

a2nt, 2024-02-01:
@jvcsw Hey there, I was the developer of changing heat-fan-cold modes.
Can you tell me how do you change mode to hold manually at the device?

jvcsw, 2024-02-02:
Hi,

in my case, this button change from hold to program and viceversa.
The button works fine and HA show the changes, but it's impossible change it from HA/Z2M.

[image]

a2nt, 2024-02-04:
> Hi,
> 
> in my case, this button change from hold to program and viceversa. The button works fine and HA show the changes, but it's impossible change it from HA/Z2M.
> 
> ![imagen](https://private-user-images.githubusercontent.com/37216513/301910135-59ced4cd-734b-4779-a779-f9274d7c29ac.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDcwNjA4ODQsIm5iZiI6MTcwNzA2MDU4NCwicGF0aCI6Ii8zNzIxNjUxMy8zMDE5MTAxMzUtNTljZWQ0Y2QtNzM0Yi00Nzc5LWE3NzktZjkyNzRkN2MyOWFjLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAyMDQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMjA0VDE1Mjk0NFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWU4NjAzMTMwYzRhMTY2ZGE1MDEzMjQ4NTZlMjVhMWY4ZGUzNGU5ZmM2NjJlMmU4ZmM2NDZiZDc0Mzg0YzAxZjQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.27D_DiO_X63vBRId4IVUndH9pnctW5qAD8rnUk97v6Q)
> 
> I found this error but I don't know if can help you.
> 
> Logger: homeassistant.components.mqtt.climate Source: components/mqtt/climate.py:828 Integration: MQTT (documentation, issues) First occurred: 17:30:11 (14 occurrences) Last logged: 17:31:15 Invalid modes mode: cool

Hmm it works different for me.
I can see that there's Sun icon under the temperature it's heating mode
Can u switch this icon? When I click to the same button as you do it switch modes: cool, heat, fan-only

[image]

richsimpson, 2024-02-04:
Hi. My model is https://www.zigbee2mqtt.io/devices/BHT-002-GCLZB.html#moes-bht-002-gclzb

GCLZB. 16amp electric heating. 

My options. Are hold or preset. 
I have the option to heat or off

Same issue. Main button on device works. The off/heat commands work but not the hold/preset. 

Warning refers to cold and no applicable state. 

[image]

a2nt, 2024-02-04:
Ok made a fix, looks like BAC-002-ALZB only has cold and fan controls

richsimpson, 2024-02-04:
> Ok made a fix, looks like BAC-002-ALZB only has cold and fan controls

Hi, Does that fix mean the change will impact the GCLZB version? to avoid the constant reminders on a HASS restart 

Logger: homeassistant.helpers.entity
Source: helpers/entity.py:1129
First occurred: 13:21:57 (1 occurrences)
Last logged: 13:21:57

Failed to set state for sensor.underfloor_heating_program, fall back to unknown
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1129, in _async_write_ha_state
    hass.states.async_set(
  File "/usr/src/homeassistant/homeassistant/core.py", line 1807, in async_set
    state = State(
            ^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 1384, in __init__
    validate_state(state)
  File "/usr/src/homeassistant/homeassistant/core.py", line 204, in validate_state
    raise InvalidStateError(
homeassistant.exceptions.InvalidStateError: Invalid state with length 933. State max length is 255 characters.

jvcsw, 2024-02-05:
> Ok made a fix, looks like BAC-002-ALZB only has cold and fan controls

With this change it seems to work. I will check it in a few days.

Thanks.

richsimpson, 2024-02-05:
I have switched to the edge addon of z2m and once I have initially set to hold or preset from the z2m edge addon, (device, exposed tab) I am indeed able to control again from Home Assistant using the climate card options etc.

I still get some random errors. Fall back to UNKNOWN state, which then triggers an error on the homekit bridge that Unknown is and invalid state.

I think connected to the message length as it is sending over mqtt the full schedule list, which is long. So unconnected? Not sure
[image]

richsimpson, 2024-02-05:
Actually scrap that, on a hass restart, the change from hold to preset is ignored. 

Have to go back into z2m and change there for it to take effect. Then it seems to work sometimes. Not always.

closed 2024-09-04 by github-actions[bot]