Koenkk/zigbee2mqtt #20621: Notification Exception unknown

devildant, 2024-01-06:
### What happened?

hello, 
randomly I have exception notifications in zigbee2mqtt, but I don't know which device is affected

[image]

### Zigbee2MQTT version

 1.35.0

### Adapter firmware version

20210708

### Adapter

sonoff P

### Debug log

info  2024-01-06 20:22:15: MQTT publish: topic 'zigbee2mqtt/Prise ..cran 2', payload '{"child_lock":"UNLOCK","current":0.17,"energy":10.59,"indicator_mode":"off","linkquality":168,"power":0,"power_outage_memory":"restore","state":"OFF","
error 2024-01-06 20:22:15: Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')}
debug 2024-01-06 20:22:16: TypeError: Cannot set properties of undefined (setting 'power')
    at Object.convert (/app/node_modules/zigbee-herdsman-converters/src/devices/tuya.ts:477:29)
    at Receive.onDeviceMessage (/app/lib/extension/receive.ts:147:35)
debug 2024-01-06 20:22:16: Received Zigbee message from 'Prise ..cran 1', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsCurrent":0}' from endpoint 1 with groupID 0
debug 2024-01-06 20:22:16: Received Zigbee message from 'Prise ..cran 1', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0}' from endpoint 1 with groupID 0

devildant, 2024-01-06:
the device mentioned before and after in the log is a connected socket from the brand "nous" so nothing to do with tuya

[image]

devildant, 2024-01-06:
it seems to come from this device

[image]

Koenkk, 2024-01-07:
Will be fixed in the next release.

closed 2024-01-07 by Koenkk