Koenkk/zigbee2mqtt #20627: Availability show N/A in initial page load

majorsl, 2024-01-07:
### What happened?

Since updating to 1.35.0, on the initial page load all devices show "N/A".  Clicking on any menu item and then clicking back to Devices then shows their Availability state properly.  It isn't a show stopper, just cosmetic.

### What did you expect to happen?

Devices would show their available state.

### How to reproduce it (minimal and precise)

Just visit the page for the first time.

### Zigbee2MQTT version

1.35.0

### Adapter firmware version

20230507

### Adapter

zStack3x0

sirjmann92, 2024-01-07:
I've noticed the same behavior with 1.35.0

- When you first load Z2M, the Availability column is replaced by a "Last Seen" column label (i.e. no devices are "available").
- The Last Seen date/time for every device is N/A.
- If you click on any other tab (e.g. Dashboard) after the initial load and then back again (to Devices tab) the Last Seen column label is now replaced by the Availability column label and all devices display their appropriate availability (Online/Offline).

tomadan1968, 2024-01-07:
The same here...

Koenkk, 2024-01-07:
Should already be fixed in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2024-08-05 by github-actions[bot]