Koenkk/zigbee2mqtt #20631: [Wrong device]: [SPM02-D2TZ] power meter won't report negative values

elektrinis, 2024-01-07:
### Link

https://zigbee.blakadder.com/Zemismart_SPM02-D2TZ.html

### Model

_TZE200_ves1ycwx

### Description

3-phase power meter

### Vendor

TuYa

### Picture (link)

https://zigbee.blakadder.com/assets/images/devices/Zemismart_SPM02-D2TZ.webp

### Database entry

{}

### Notes

Device is working well, however it is reporting negative current/power as positive. Positive also as positive, so I'm sure there is no issue with wiring. Not sure if this is integration, or issue with device itself.
Newbie here, please explain in detail if I need to submit any extra info.

HobboRobin, 2024-02-08:
I have the same Problem with the Same Device. I have a 1Phase Solar inverter on phase 3. when i have a power load of 50W and then inject 600W it goes to 550W. So there is something wrong. I got the from the manufacture a converter File. But i dont get it working correctly. Or i dont even know it is correct.

HobboRobin, 2024-02-08:
/**********************************************************************
Version    01.00.02

Date       2023/10/26

update     支持读取总有功功率，三相无功功率，总无功功率，
           三相势在功率，总势在功率，系统频率
**********************************************************************/

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const { Numeric } = require('zigbee-herdsman-converters/lib/exposes');
const {
    precisionRound, mapNumberRange, isLegacyEnabled, toLocalISOString, numberWithinRange, hasAlreadyProcessedMessage,
    calibrateAndPrecisionRoundOptions, addActionGroup, postfixWithEndpointName, getKey,
    batteryVoltageToPercentage,
} = require('zigbee-herdsman-converters/lib/utils');
const utils = require('zigbee-herdsman-converters/lib/utils');

let preEnergy = 0;
let preProduced_energy = 0;

const converters = {
    seMetering: {

        cluster: 'seMetering',
        type: ['attributeReport', 'readResponse'],
        options: (definition) => {
            const result = [];
            if (definition.exposes.find((e) => e.name === 'power')) {
                result.push(exposes.options.precision('power'), exposes.options.calibration('power', 'percentual'));
            }
            if (definition.exposes.find((e) => e.name === 'energy')) {
                result.push(exposes.options.precision('energy'), exposes.options.calibration('energy', 'percentual'));
            }
            if (definition.exposes.find((e) => e.name === 'produced_energy')) {
                result.push(exposes.options.precision('produced_energy'), exposes.options.calibration('energy', 'percentual'));
            }
            return result;

        },

        convert: (model, msg, publish, options, meta) => {
            if (utils.hasAlreadyProcessedMessage(msg, model)) return;
            const payload = {};
            const multiplier = msg.endpoint.getClusterAttributeValue('seMetering', 'multiplier');
            const divisor = msg.endpoint.getClusterAttributeValue('seMetering', 'divisor');
            const factor = multiplier && divisor ? multiplier / divisor : null;

            if (factor != null && (msg.data.hasOwnProperty('currentSummDelivered') ||
                msg.data.hasOwnProperty('currentSummReceived'))) {
                let energy  = preEnergy;
                let produced_energy  = preProduced_energy;
                if (msg.data.hasOwnProperty('currentSummDelivered')) {
                    const data = msg.data['currentSummDelivered'];
                    const value = (parseInt(data[0]) << 32) + parseInt(data[1]);
                    energy = value * factor;
                    preEnergy = energy;
                    // produced_energy = preProduced_energy;
                }
                if (msg.data.hasOwnProperty('currentSummReceived'))  {
                    const data = msg.data['currentSummReceived'];
                    const value = (parseInt(data[0]) << 32) + parseInt(data[1]);
                    produced_energy = value * factor;
                    preProduced_energy = produced_energy;
[... 131 lines cut ...]
        await reporting.activePower(endpoint);// 有功功率
        await reporting.rmsCurrent(endpoint);// 电流
        await reporting.rmsVoltage(endpoint);// 电压
        await reporting.powerFactor(endpoint);
        await reporting.apparentPower(endpoint);
        await reporting.reactivePower(endpoint);
        await reporting.currentSummDelivered(endpoint);
        await reporting.currentSummReceived(endpoint);
        device.save();
    },
    exposes:  [e.ac_frequency(), e.voltage(), e.voltage_phase_b(), e.voltage_phase_c(),
        power_kW(), power_kW_phase_b(), power_kW_phase_c(), total_power_kW(),
        e.current(), e.current_phase_b(), e.current_phase_c(),
        e.power_factor(), power_factor_phase_b(), power_factor_phase_c(),
        power_reactive(), power_reactive_phase_b(), power_reactive_phase_c(), total_power_reactive(),
        power_apparent(), power_apparent_phase_b(), power_apparent_phase_c(), total_power_apparent(),
        // Change the description according to the specifications of the device
        e.energy().withDescription('Total forward active energy'),
        e.produced_energy().withDescription('Total reverse active energy'),
        hw_version(), locationDesc(),
        //e.paySwitch().withDescription('预付费开关'),
    ],
};

module.exports = definition;

Le-Syl21, 2024-03-14:
How to use this informations ?
Have the same device and show not supported for the moment.
Does somebody as a git page with code and a doc ???

Le-Syl21, 2024-03-18:
I've update to Z2M EDGE. It's work but I can't show energy and produced energy:
[image]
Can someone help me ?

sesame1215, 2024-03-22:
I have also purchased the same product from them. They have told me that there are two versions: V1 (older) and V2 (newer). The V2 versions can show the negative values and have more DP (more user friendly in my view). Too bad that V1 cannot OTA to V2.

maciey, 2024-04-03:
> I've update to Z2M EDGE

How did you manage to update this device? Did you resolve missign readings issues?

Le-Syl21, 2024-04-04:
I have found a solution:
use this file: [SPM02_Z3_Local_02.j](https://github.com/script0803/SPM0X-Z2M-Configuration/blob/main/SPM02_Z3_Local_02.js)
then replace line 15:
BEFORE = const extend = require('zigbee-herdsman-converters/lib/extend');
AFTER = const extend = require('zigbee-herdsman-converters/lib/modernExtend');

It's work but need rework like Hz. Look at this:
[image]
[image]
[image]
Bye

maciey, 2024-04-04:
> I have found a solution:

Thanks a lot.
Can you plz advise where should I put this file and how to configure HA to read it?

dfrepos, 2024-09-28:
> I have found a solution: use this file: [SPM02_Z3_Local_02.j](https://github.com/script0803/SPM0X-Z2M-Configuration/blob/main/SPM02_Z3_Local_02.js) then replace line 15: BEFORE = const extend = [...]

Can you please share where exacly do we have to implement this? Is it a OTA firmware update or some other config in HA?

maciey, 2024-10-02:
@Le-Syl21 - same question here... 🙏

script0803, 2024-10-08:
https://github.com/script0803/SPM0X-Z2M-Configuration/blob/main/SPM02_Z3_Local_02.js
This file is the external converter of zigbee2mqtt, it is suitable for the zigbee3.0 version of SPM02, it has the suffix of U01. 

So it is not the same product as the SPM02-D2TZ mentioned at the beginning. 

I think the U01 version of the device does not need this file now. I have added its configuration to the zigbee2mqtt support library.

akira215, 2024-10-24:
I'm not using your device but I'm acutally witing a converter for another device. But I note that 
`const value = (parseInt(data[0]) << 32) + parseInt(data[1]);` is failing from my side:

It fail sto left shift 32 bits due to js number limitation. Basically,  `(parseInt(data[0]) << 32)` is equal to `parseInt(data[0]) `

The workaround is to use BigInt : ` (BigInt(parseInt(data[0])) << BigInt(32) + BigInt(parseInt(data[1]));`, after what you will have to always add BigInt in your computations. 

Then I received `EventBus error 'Receive/deviceMessage': Do not know how to serialize a BigInt` . I will try to solve that, but does anyone experienced the same ? Obviously, it will not affect the result until you reach a value higher than 32bits length int.

closed 2025-01-07 by github-actions[bot]