Koenkk/zigbee2mqtt #20633: [New device support]:

dompagoj, 2024-01-07:
### Link

https://moeshouse.com/products/zigbee-star-ring-curtain-switch?variant=46106364117307

### Database entry

{"id":5,"type":"Router","ieeeAddr":"0xa4c138e21142d588","nwkAddr":29520,"manufId":4417,"manufName":"_TZE204_srmahpwl","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0601","manufacturerName":"_TZE204_srmahpwl","powerSource":1,"zclVersion":3,"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1704629444944,"defaultSendRequestWhen":"immediate"}

### Comments

I tried adding it myself but no luck

### External defintion

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE204_srmahpwl',
    description: 'Automatically generated definition',
    extend: [],
};
```

tdelsupexhe, 2024-01-19:
```js
// Add the lines below
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const {} = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE204_srmahpwl'}],
    model: 'ZS-SR-EUC',
    vendor: 'Moes',
    description: 'Star Ring - Smart Curtain Switch',
    options: [exposes.options.invert_cover()],
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    //onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    //configure: tuya.configureMagicPacket,
    exposes: [
        // Here you should put all functionality that your device exposes
        //e.cover_position(),
        e.cover_position().setAccess('position', ea.STATE_SET),
        e.enum('calibration', ea.STATE_SET, ['START', 'END']).withDescription('Calibration'),
        //e.binary('backlight_mode', ea.STATE_SET, 'ON', 'OFF').withDescription('Backlight'),
        e.enum('indicator_mode', ea.STATE_SET, ['off', 'on_off', 'off_on']).withDescription('Relay LED indicator mode'),
        e.enum('motor_steering', ea.STATE_SET, ['FORWARD', 'BACKWARD']).withDescription('Motor Steering')
    ],
    meta: {
        // All datapoints go in here
        tuyaDatapoints: [
            [1, 'state', tuya.valueConverterBasic.lookup({'CLOSE': tuya.enum(2), 'STOP': tuya.enum(1), 'OPEN': tuya.enum(0)})], //Curtain Switch 1 
            [2, 'position', tuya.valueConverter.coverPosition], // Percentage 1
            [3, 'calibration', tuya.valueConverterBasic.lookup({'START': tuya.enum(0), 'END': tuya.enum(1)})],, // Accurate Calibration
            [8, 'motor_steering', tuya.valueConverterBasic.lookup({'FORWARD': tuya.enum(0), 'BACKWARD': tuya.enum(1)})], // Motor steering
            // [14, 'backlight_mode', tuya.valueConverter.backlightModeOffNormalInverted], // Indicator Status
            [14, 'indicator_mode', tuya.valueConverterBasic.lookup({'off': tuya.enum(0), 'on_off': tuya.enum(1), 'off_on': tuya.enum(2)})],
        ],
    },
};
    
    module.exports = definition;
```

This works to open/close the curtain but the other fonctions not

dompagoj, 2024-01-21:
Thanks that works!

closed 2024-01-21 by dompagoj