Koenkk/zigbee2mqtt #20638: Last seen column is N\A

almirus, 2024-01-07:
### What happened?

When you first enter the interface, the LAST SEEN column shows N/A, but if you switch to any device and then back, the values are displayed correctly. 
[image]

At this time, spam messages 'Unknown last_seen type undefined' in the console.
[image]

### Zigbee2MQTT version

1.35.0

### Adapter firmware version

20230507

### Adapter

egony Stick V4

almirus, 2024-01-07:
This occurs after switching to a specific device and back.
[image]

Koenkk, 2024-01-07:
Should be fixed in z2m 1.35.1

closed 2024-01-07 by Koenkk