Koenkk/zigbee2mqtt #20640: Error update 1.35.1 don't start :  Error: Cannot find module './dist/util/settings

kwetnico, 2024-01-07:
### Is your feature request related to a problem? Please describe

I test this and don't work :  find : [#19318](https://github.com/Koenkk/zigbee2mqtt/issues/19318)
git pull
npm run build

### Describe the solution you'd like

update to 1.35.1

### Describe alternatives you've considered

/

### Additional context

`rrypi npm[2268596]: > node index.js
janv. 07 15:04:55 raspberrypi npm[2268642]: Error: Cannot find module './dist/util/settings'
janv. 07 15:04:55 raspberrypi npm[2268642]: Require stack:
janv. 07 15:04:55 raspberrypi npm[2268642]: - /opt/zigbee2mqtt/index.js
janv. 07 15:04:55 raspberrypi npm[2268642]:     at Function.Module._resolveFilename (node:internal/modules/cjs/loader:1077:15)
janv. 07 15:04:55 raspberrypi npm[2268642]:     at Function.Module._load (node:internal/modules/cjs/loader:922:27)
janv. 07 15:04:55 raspberrypi npm[2268642]:     at Module.require (node:internal/modules/cjs/loader:1143:19)
janv. 07 15:04:55 raspberrypi npm[2268642]:     at require (node:internal/modules/cjs/helpers:121:18)
janv. 07 15:04:55 raspberrypi npm[2268642]:     at start (/opt/zigbee2mqtt/index.js:90:22)
janv. 07 15:04:55 raspberrypi systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE
janv. 07 15:04:55 raspberrypi systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
janv. 07 15:04:55 raspberrypi systemd[1]: zigbee2mqtt.service: Consumed 1.751s CPU time.
`

Koenkk, 2024-01-07:
Try executing `rm -rf dist tsconfig.tsbuildinfo && npm run build`

kwetnico, 2024-01-07:
> rm -rf dist tsconfig.tsbuildinfo && npm run build

hi, it's better and ok 👍 
[image]
1.35.1 :
[image]

kwetnico, 2024-01-07:
Solution :
[image]

closed 2024-02-22 by kwetnico