Koenkk/zigbee2mqtt #20641: LEDVANCE 4058075729025: update firmware available but "Images currently unavailable"

somar05, 2024-01-07:
### What happened?

Control panel (OTA) show a firmware update available:

[image]

But the update fail showing this log message:
```
Info 2024-01-07 15:19:02MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"id":"luz_salita"},"error":"Update of 'luz_salita' failed (Images currently unavailable)","status":"error","transaction":"bi8ek-1"}'

Error 2024-01-07 15:19:02Update of 'luz_salita' failed (Images currently unavailable)
```

### What did you expect to happen?

Image available

### How to reproduce it (minimal and precise)

Just use the update button

### Zigbee2MQTT version

1.35.1

### Adapter firmware version

 20220219

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus 

### Debug log

> Info 2024-01-07 15:19:02MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"id":"luz_salita"},"error":"Update of 'luz_salita' failed (Images currently unavailable)","status":"error","transaction":"bi8ek-1"}'

>Error 2024-01-07 15:19:02Update of 'luz_salita' failed (Images currently unavailable)

andarotajo, 2024-01-07:
I'm experiencing the same problem but with a different LEDVANCE product: https://www.zigbee2mqtt.io/devices/4058075729384.html#ledvance-4058075729384

`OTA: Images currently unavailable for device 'FLEX RGBW T', hardwareVersion '0', manufacturerName LEDVANCE, {"fieldControl":0,"manufacturerCode":4489,"imageType":193,"fileVersion":52442740}'`

Gorouk, 2024-01-07:
Same with [Gledopto GL-C-006P](https://www.zigbee2mqtt.io/devices/GL-C-006P.html#gledopto-gl-c-006p)

mircolino, 2024-01-07:
Same issue here with a [Ledvance Compact Outdoor Plug EU](https://www.zigbee2mqtt.io/devices/4058075729322.html#ledvance-4058075729322)

Koterak, 2024-01-07:
Probably problem with more devices - I have update info for most of gledopto devices. Device report for example firmware version 11, newest -1
[image]

fv71, 2024-01-07:
Same with [Gledopto GL-B-007P](https://www.zigbee2mqtt.io/devices/GL-B-007P.html#gledopto-gl-b-007p)

jpkessler, 2024-01-08:
Same for these ledvance devices:

```
Jan  7 17:31:51 hassos ZigbeeOBBE OTA: Images currently unavailable for device 'FLEX RGBW T', hardwareVersion '0', manufacturerName LEDVANCE, {"fieldControl":0,"manufacturerCode":4489,"imageType":193,"fileVersion":52442740}'/n
Jan  7 17:31:59 hassos ZigbeeOBBE OTA: Images currently unavailable for device 'FLEX RGBW T', hardwareVersion '0', manufacturerName LEDVANCE, {"fieldControl":0,"manufacturerCode":4489,"imageType":193,"fileVersion":52442740}'/n
Jan  7 17:33:19 hassos ZigbeeOBBE OTA: Images currently unavailable for device 'A60 RGBW T', hardwareVersion '0', manufacturerName LEDVANCE, {"fieldControl":0,"manufacturerCode":4489,"imageType":182,"fileVersion":52573812}'/n
Jan  7 17:33:55 hassos ZigbeeOBBE OTA: Images currently unavailable for device 'OUTDOOR FLEX RGBW T', hardwareVersion '0', manufacturerName LEDVANCE, {"fieldControl":0,"manufacturerCode":4489,"imageType":197,"fileVersion":52442740}'/n
Jan  7 17:34:06 hassos ZigbeeOBBE OTA: Images currently unavailable for device 'FLEX RGBW T', hardwareVersion '0', manufacturerName LEDVANCE, {"fieldControl":0,"manufacturerCode":4489,"imageType":193,"fileVersion":52442740}'/n
Jan  7 17:35:19 hassos ZigbeeOBBE OTA: Images currently unavailable for device 'A60 RGBW T', hardwareVersion '0', manufacturerName LEDVANCE, {"fieldControl":0,"manufacturerCode":4489,"imageType":182,"fileVersion":52573812}'/n
Jan  7 17:36:28 hassos ZigbeeOBBE OTA: Images currently unavailable for device 'FLEX RGBW T', hardwareVersion '0', manufacturerName LEDVANCE, {"fieldControl":0,"manufacturerCode":4489,"imageType":193,"fileVersion":52442740}'/n
Jan  7 18:34:23 hassos ZigbeeOBBE OTA: Images currently unavailable for device 'FLEX RGBW T', hardwareVersion '0', manufacturerName LEDVANCE, {"fieldControl":0,"manufacturerCode":4489,"imageType":193,"fileVersion":52442740}'/n
Jan  8 12:43:14 hassos ZigbeeOBBE OTA: Images currently unavailable for device 'FLEX RGBW T', hardwareVersion '0', manufacturerName LEDVANCE, {"fieldControl":0,"manufacturerCode":4489,"imageType":193,"fileVersion":52442740}'/n

```

recklezz, 2024-01-08:
Same with [EcoDim-Zigbee 3.0](https://www.ecodim.nl/nl/eco-dim07-zigbee-pro.html)

JordiPF, 2024-01-08:
Same for me with Develco KEYZB-110 (https://www.zigbee2mqtt.io/devices/KEYZB-110.html).

Here is the report:

```
Debug 2024-01-08 16:00:10Received MQTT message on 'zigbee2mqtt/bridge/request/device/ota_update/update' with data '{"id":"Teclado alarma","transaction":"ytgm6-1"}'
Info 2024-01-08 16:00:10Updating 'Teclado alarma' to latest firmware
Info 2024-01-08 16:00:10MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Updating 'Teclado alarma' to latest firmware","meta":{"device":"Teclado alarma","status":"update_in_progress"},"type":"ota_update"}'
Debug 2024-01-08 16:00:14Received Zigbee message from 'Teclado alarma', type 'readResponse', cluster 'genBasic', data '{"dateCode":"2022-04-20 21:10"}' from endpoint 44 with groupID 0
Debug 2024-01-08 16:00:14Received Zigbee message from 'Teclado alarma', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":67082,"imageType":915,"manufacturerCode":4117}' from endpoint 44 with groupID 0
Debug 2024-01-08 16:00:14Update of 'Teclado alarma' failed (AssertionError [ERR_ASSERTION]: Images currently unavailable)
Info 2024-01-08 16:00:14MQTT publish: topic 'zigbee2mqtt/Teclado alarma', payload '{"action":null,"action_code":null,"action_transaction":null,"action_zone":null,"battery":100,"battery_low":false,"linkquality":222,"tamper":false,"update":{"installed_version":67082,"latest_version":-1,"state":"available"},"update_available":true,"voltage":6300}'
Info 2024-01-08 16:00:14MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Update of 'Teclado alarma' failed (Images currently unavailable)","meta":{"device":"Teclado alarma","status":"update_failed"},"type":"ota_update"}'
Info 2024-01-08 16:00:14MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"id":"Teclado alarma"},"error":"Update of 'Teclado alarma' failed (Images currently unavailable)","status":"error","transaction":"ytgm6-1"}'
Error 2024-01-08 16:00:14Update of 'Teclado alarma' failed (Images currently unavailable)
Debug 2024-01-08 16:00:17Received Zigbee message from 'Teclado alarma', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":67082,"imageType":915,"manufacturerCode":4117}' from endpoint 44 with groupID 0
Debug 2024-01-08 16:00:17Device 'Teclado alarma' requested OTA
Debug 2024-01-08 16:00:20Received Zigbee message from 'Teclado alarma', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":67082,"imageType":915,"manufacturerCode":4117}' from endpoint 44 with groupID 0
Debug 2024-01-08 16:00:20Device 'Teclado alarma' requested OTA
Debug 2024-01-08 16:00:23Received Zigbee message from 'Teclado alarma', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":67082,"imageType":915,"manufacturerCode":4117}' from endpoint 44 with groupID 0
Debug 2024-01-08 16:00:23Device 'Teclado alarma' requested OTA
Debug 2024-01-08 16:00:27Received Zigbee message from 'Teclado alarma', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":67082,"imageType":915,"manufacturerCode":4117}' from endpoint 44 with groupID 0
Debug 2024-01-08 16:00:27Device 'Teclado alarma' requested OTA
```

Thanks for help.

Scialla, 2024-01-08:
same here with ikea device (battery and acdc)

Koenkk, 2024-01-09:
Please check if this is fixed in the latest dev branch.

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

andarotajo, 2024-01-10:
Looks good to me. Had one light strip showing -1 in HA and with the latest dev it vanished after checking for updates. 

```json
{
    "brightness": 254,
    "color": {
        "x": 0.699,
        "y": 0.301
    },
    "color_mode": "xy",
    "color_temp": 165,
    "last_seen": "2024-01-10T22:35:08+01:00",
    "linkquality": 87,
    "state": "OFF",
    "update": {
        "installed_version": 52442740,
        "latest_version": 52442740,
        "state": "idle"
    },
    "update_available": null
}
```

closed 2024-01-11 by Koenkk

vient41, 2024-01-19:
When will this PR pushed on stable please ?

Koenkk, 2024-01-21:
Yes, next release is on 1 Feb

sirjmann92, 2024-01-24:
I'm seeing similar behavior from Sylvania LEDVANCE 74746 LED Panels. Zigbee2MQTT.io shows the device has OTA support. The devices show up in the OTA list, however there are no firmware versions or build dates in their respective columns in the OTA table. Clicking "Check for new updates" results in error mesage: 
`Error 2024-01-23 17:54:42Failed to check if update available for 'LED Panel 01' (OTA: Device didn't respond to OTA request)`
[image]

@Koenkk, If this is a different issue, or if you want a separate issue opened for this, just let me know and I'll collect the info and open it. 

I don't recall seeing these devices show up in the OTA list until 1.35

vient41, 2024-02-21:
@sirjmann92 , I am still facing the issue after 1st feb release and you mentionned opening a new issue, can you provide with new ticket number please ?

sirjmann92, 2024-02-22:
@vient41 - It's here: https://github.com/Koenkk/zigbee2mqtt/issues/20983