Koenkk/zigbee2mqtt #20643: Tamper binary sensor for IKEA E2013 disappeared after update 1.35.1

zaihan, 2024-01-07:
### What happened?

Tamper binary sensor for IKEA E2013 disappeared after updating to zigbee2mqtt last version 1.35.1.

binary_sensor.mydevice_tamper

### Zigbee2MQTT version

1.35.1

### Adapter firmware version

7.3.1.0 build 176

### Adapter

SONOFF ZigBee 3.0 USB Dongle Plus, EFR32MG21 Coordinator

gewoonivar, 2024-01-12:
Same here.

In addition the battery indicator has changed to a % value. However is since change _N/A %_.

LarsStegman, 2024-01-12:
It has no tamper sensor, so it makes sense it was removed. The battery is a problem though, I am running into that as well.

closed 2024-08-10 by github-actions[bot]