Koenkk/zigbee2mqtt #20646: after Update to 1.35.1-1 OTA update problem, Updates for products they have no Update

Oggymator, 2024-01-07:
### What happened?

Since the new version it show me updates for products that have no update.
New Firmware Version is -1, looks not correct.

In the Log I got also this message:
`EventBus error 'OTAUpdate/deviceMessage': CommandResponse 0xa4c13856492e18f0/11 genOta.queryNextImageResponse({"status":152}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC no ack' (233))
2024-01-07 17:21:34Failed to configure 'Fernsehtisch', attempt 1 (Error: Read 0x385b44fffe20b910/1 genBasic(["modelId"], `

### Zigbee2MQTT version

1.35.1-1 

### Adapter firmware version

-

### Adapter

ZigStar

VivantSenior, 2024-01-07:
I have similar issue. Home Assistants reports available updates for [Eurotronic SPZB0001](https://www.zigbee2mqtt.io/devices/SPZB0001.html#eurotronic-spzb0001):
```
Installed version: 4294967295
Latest version: -1
```

and 3 (out of ~22) [Xiaomi MCCGQ14LM](https://www.zigbee2mqtt.io/devices/MCCGQ14LM.html#xiaomi-mccgq14lm)
```
Installed version: 4
Latest version: -1
```

Z2M reports `Failed to check if update available for 'Living room window' (OTA: Device didn't respond to OTA request)` when checking available updates for MCCGQ14LM.

aletzi1, 2024-01-07:
Same problem here with Ikea LED2003G10 and Ikea LED2005R5

benstoked, 2024-01-07:
...
Home Assitant:
 Core 2024.1.2
Supervisor 2023.12.0
Operating System 11.3
Frontend 20240104.0`

zigbee2mqtt 1.35.1-1 

...

five of my eight Sengled-e21-n1ea are showing the same message
```
        "installed_version": 38,
        "latest_version": -1,
```

robitsrl, 2024-01-07:
Same problem with SONOFF | SNZB-01P

"installed_version": 8704
"latest_version": -1

Ronnnnny, 2024-01-08:
Same here 21 out of 110 devices.

Gadeo, 2024-01-09:
Same problem with LEDVANCE 4058075729322 since 1.35.1-1 version. Cheers.

Oggymator, 2024-01-09:
Nobody that can fix this issue?

I have now 54 devices that show me an update…

Koenkk, 2024-01-09:
Should be fixed in the dev branch.

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-01-09 by Koenkk

Bebebe1974, 2024-01-17:
Same problem with LYWSD03MMC(zigbee ver 1.0.6), Zigbee2mqtt version 1.35.1
`0xa4c13852f97db079/1 genOta.queryNextImageResponse({"status":152}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'Timeout' (9999))`

benstoked, 2024-02-02:
persists with update to 1.35.2-1 in home assistant os,

Oggymator, 2024-02-04:
yes, same problem with 1.35.2-1

ReneTode, 2024-04-04:
@Koenkk i see this problem also after i updated to 1.36.0 2 weeks ago.
even though i got Disable automatic update check set to true.