Koenkk/zigbee2mqtt #20651: EZSP Firmware 7.4.0.0 causes fail

Frostis12, 2024-01-07:
### What happened?

Updatet my dongle to EZSP firmware 7.4.0.0 and zigbee2mqtt wont open. throws error: "Assertion failed: Command (setConfigurationValue) returned unexpected state: [object Object]"

Downgrade to FW 7.3.2.0 fixes problem.

### Zigbee2MQTT version

1.35.1

### Adapter firmware version

7.4.0.0

### Adapter

zb-gw04 v1.2

### Debug log

Assertion failed: Command (setConfigurationValue) returned unexpected state: [object Object]

burntoc, 2024-01-07:
Same versions here, same issues.  HW is SkyConnect

Koenkk, 2024-01-08:
See https://github.com/Koenkk/zigbee-herdsman/issues/854, should be fixed in the dev branch.

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-01-08 by Koenkk

burntoc, 2024-01-09:
@Koenkk Maybe I just don't understand GH very will. I switched branch to Dev, but Iv still see 1.35.1. I think I'm looking for 1.35.2 with this change, right?

Koenkk, 2024-01-09:
The version should report `1.35.1-dev` in the frontend -> settings -> about section

burntoc, 2024-01-09:
Thanks. I'm probably looking in the wrong place as in HA that only says current version: edge and clicking changelog just confirms it tracks the dev branch.  But it works, so thanks!