Koenkk/zigbee2mqtt #20655: z2m crashes randomly

Mikkooooo, 2024-01-07:
### What happened?

Hi,
my z2m crashes randomly every 24 - 48 hours and the z2m server stops.
To fix this, I must unplug and replug the Sonoff P stick and restart z2m.
After some days the failure occurs again.

### Zigbee2MQTT version

1.33.1

### Adapter firmware version

20230507

### Adapter

Sonoff dongle P

adekloet, 2024-01-08:
Possible thing to test: Please downgrade the firmware of the dongle to a 2022 version, and retest firmware stability.

bartplessers, 2024-01-09:
I have same problem with 2022 firmware, see

https://github.com/Koenkk/zigbee2mqtt/issues/19371#issuecomment-1882777194

Grtz
Bartplessers

ottoketney, 2024-01-12:
I have the same behaviour

Zigbee2MQTT version
: 1.35.1
Coordinator type
: EZSP v9
Coordinator revision
: 7.1.1.0 build 273

darthrater78, 2024-01-16:
I have the same behavior

Zigbee2MQTT version
: 1.35.1
Coordinator type
: UZG-01
Coordinator revision
: CC2652P version: CC2652PP1FRGZR
:ESP32 Firmware ver: 0.1.5 (2023-09-06 12:01:02.723464)

bartplessers, 2024-01-21:
FYI: for some other reason I had to remove and add nginx proxy manager addon few days ago. The new version of nginx doesn't need mariadb anymore, so I also removed this addon...
Since then now problems anymore..... 
Probably not related, but just for your interest.

closed 2024-08-19 by github-actions[bot]