Koenkk/zigbee2mqtt #20658: GL-C-009P / Firmware '-1'

mainmind83, 2024-01-07:
### What happened?

Update to v1.35.1 and show firmware "-1"

{
    "brightness": 241,
    "linkquality": 149,
    "power_on_behavior": "on",
    "state": "OFF",
    "update": {
        "installed_version": 369176577,
        "latest_version": -1,
        "state": "idle"
    }
}

### What did you expect to happen?

Show new firmware?

### Zigbee2MQTT version

1.35.1

### Adapter firmware version

20230507

### Adapter

SONOFF USB ZBDongle-P

LarsStegman, 2024-01-07:
I get the same thing for the IKEA Parasoll sensors

blakefrain, 2024-01-07:
I am getting the same thing with 3 Sengled E21-N1EA
[image]

mainmind83, 2024-01-07:
It seems like it could be a more general error, not associated with a specific device... all of which there is no firmware available?

blakefrain, 2024-01-07:
I would say you are correct. The OTA does not work for the Sengled E21-N1EA but I see no errors coming up, just those listed and as an update in Home Assistant settings.

ScottG489, 2024-01-08:
I'm getting this as well for [Sengled E1F-N5E](https://www.zigbee2mqtt.io/devices/E1F-N5E.html#sengled-e1f-n5e) bulbs.

If I go into Z2M and the OTA tab and click on "Check for new updates" for one of those bulbs I get the following error pop up:

```
OTA: Images currently unavailable for device 'E1F-N5E', hardwareVersion '1', manufacturerName sengled, {"fieldControl":0,"manufacturerCode":4448,"imageType":32977,"fileVersion":22}'
```

nicolinuxfr, 2024-01-08:
Same here, for two Sonoff SNZB-02P devices.

eeeeb, 2024-01-08:
Go the same issue with some Sonoff devices.

corvy, 2024-01-08:
I got the same for 2 Namron devices: https://www.zigbee2mqtt.io/devices/4512700.html#namron-4512700

[image]

sdotter, 2024-01-08:
Same for Ikea parasoll sensors

https://www.zigbee2mqtt.io/devices/E2013.html#ikea-e2013

[image]

kuduacz, 2024-01-08:
same with hive trv's

Frankster-NL, 2024-01-08:
To add: same here with Gledopto GL-C-009P…

Veldkornet, 2024-01-09:
I found the same issue at https://github.com/Koenkk/zigbee2mqtt/issues/20678

LarsStegman, 2024-01-09:
I think https://github.com/Koenkk/zigbee-herdsman-converters/pull/6843 introduced this bug.

mainmind83, 2024-01-09:
> I think [Koenkk/zigbee-herdsman-converters#6843](https://github.com/Koenkk/zigbee-herdsman-converters/pull/6843) introduced this bug.

It should be marked as OTA update not available for this device...

Nerivec, 2024-01-09:
https://github.com/Koenkk/zigbee-herdsman-converters/pull/6864

Already fixed, problem with HA (doesn't recognized negative number as "don't try to update"...), but Z2M should work around it as soon as this is merged.

HA PR https://github.com/home-assistant/core/pull/107483

mainmind83, 2024-01-09:
version -1 in software is equivalent to the moment the idea comes out of the head of the developer or project manager 😄

Nerivec, 2024-01-09:
@LarsStegman HA core wouldn't behave properly ("OFF") on all of these too. [link](https://github.com/home-assistant/core/blob/29cac5b093443a86044224ed5365fbbb70760840/homeassistant/components/update/__init__.py#L396)
HA actually fails because it doesn't recognize "-1" as a version number (or any negative number)... and since it defaults to ON whenever it fails to compare... it reports "update available". _The fault ultimately lies in the `AwesomeVersion` package used by HA._
-1 should at least always be less than 0 (which hopefully any valid version number aspires to be greater to) :smile:

With the PR, Z2M will report current version as latest version in case no image is available and fix the issue. It just limits what the frontend could have done with the -1 special case; could have been used "as-is" in the future to handle "no image available" differently than "no update available" (what I had in mind @mainmind83 😁), but I guess if needed we can revisit the implementation at that time!

closed 2024-01-09 by Koenkk

88fingerslukee, 2024-01-12:
Why is it that there should be an update available but it's still showing as -1? I have an E1744 IKEA device that is on an old version. It's still showing as -1

```
Info 2024-01-12 11:52:02MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{"id":"0xec1bbdfffe396590"},"error":"Failed to check if update available for '0xec1bbdfffe396590' (Cannot destructure property 'data' of '(intermediate value)' as it is undefined.)","status":"error","transaction":"hh8mg-5"}'
Error 2024-01-12 11:52:02Failed to check if update available for '0xec1bbdfffe396590' (Cannot destructure property 'data' of '(intermediate value)' as it is undefined.)
Debug 2024-01-12 11:52:02TypeError: Cannot destructure property 'data' of '(intermediate value)' as it is undefined. at getImageMeta (/app/node_modules/zigbee-herdsman-converters/src/lib/ota/tradfri.ts:15:18) at processTicksAndRejections (node:internal/process/task_queues:95:5) at isNewImageAvailable (/app/node_modules/zigbee-herdsman-converters/src/lib/ota/common.ts:398:18) at Object.isUpdateAvailable (/app/node_modules/zigbee-herdsman-converters/src/lib/ota/common.ts:387:29) at OTAUpdate.onMQTTMessage (/app/lib/extension/otaUpdate.ts:205:45) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:167:17)
```

Nerivec, 2024-01-12:
@88fingerslukee  I just tried the update URL for IKEA `https://fw.ota.homesmart.ikea.net/feed/version_info.json`, it's currently not working. We'll need more checks to make that kind of error clearer.

LarsStegman, 2024-01-20:
@Koenkk  I know the next release is feb 1, because you mentioned it, but is there a certain schedule?

Koenkk, 2024-01-21:
Every 1st of the month!

SatAmnesiac, 2024-01-26:
FWIW I have a Gledopto GL-C-001P (working as as a 003P) giving the same error. Is there a fix coming in Feb? At the moment I've turned off Automatic OTA checking in Zigbee2MQTT and 'skipped' the install message in HA.

mainmind83, 2024-02-01:
https://github.com/Koenkk/zigbee-herdsman-converters/pull/6864

1 fixed on v1.35.2-1

Pferdebockwurst, 2024-02-01:
I've updated to 1.35.2 today and still get these weird -1 firmware update notifications in home assistant.

khangwei, 2024-02-02:
Same. just updated today.. still showing -1

khangwei, 2024-02-02:
thank you! that worked

fv71, 2024-02-03:
I confirm, after rescanning, the -1 errors disappeared.
Thank you!

Veldkornet, 2024-02-06:
> I have also updated to v1.35.2-1 and I believe it worked for 5 out of my 9 Eco-Dim.07 units, at least they are no longer listed in the "settings" - "updates" section.
> 
> 
> 
> For the remaining 3-4 units they are still in the list. I tried the suggested fix by @elderec without any luck. I have since also restarted HA a couple of times but that didn't have any effect.
> 
> 
> 
> **Zigbee2MQTT version**: Zigbee2MQTT version 1.35.2-1 commit: unknown
> 
> **Coordinator type:** EZSP v12
> 
> **Coordinator revision**: 7.3.1.0 build 0
> 
> **Adapter:** Sonoff ZBDongle-E USB Zigbee 3.0 USB Dongle Plus,EFR32MG21 + CH9102F Zigbee USB-Stick
> 
> **Zigbee-herdsman-converters version:** 18.25.1
> 
> **Zigbee-herdsman version:** 0.33.5
> 
> 
> 
> [image]
> 
> 

What is the imageType of those devices? I added the firmware for imageType 60 which is probably why some of them are fixed.

velodromen, 2024-02-06:
Thanks for looking into this! How do I check which imageType they have please?

I tried to update one of the units to the "-1" version and received the following log entry:

`
Info 2024-02-06 11:19:37Updating 'Hall | Belysning | Spotlights' to latest firmware
`
`
Debug 2024-02-06 11:20:38Update of 'Hall | Belysning | Spotlights' failed (Error: OTA: Device didn't respond to OTA request)
`
`
Info 2024-02-06 11:20:38MQTT publish: topic 'zigbee2mqtt/Hall | Belysning | Spotlights', payload '{"brightness":178,"linkquality":255,"power_on_behavior":null,"state":"ON","update":{"installed_version":16869200,"latest_version":-1,"state":"available"},"update_available":null}'
`
`
Info 2024-02-06 11:20:38MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"id":"Hall | Belysning | Spotlights"},"error":"Update of 'Hall | Belysning | Spotlights' failed (OTA: Device didn't respond to OTA request)","status":"error","transaction":"a4bx9-1"}'
`
`
Error 2024-02-06 11:20:38Update of 'Hall | Belysning | Spotlights' failed (OTA: Device didn't respond to OTA request)
`

Veldkornet, 2024-02-06:
> Thanks for looking into this! How do I check which imageType they have please?
>
>
>
> I tried to update one of the units to the "-1" version and received the following log entry:
>
>
>
> `
>
> Info 2024-02-06 [...]

It should be there in the debug logs somewhere I guess if you check for an update. 
Example:

```
2024-02-06 18:39:54Received Zigbee message from 'Eettafellamp', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":16869200,"imageType":60,"manufacturerCode":4714}' from endpoint 1 with groupID 0
```

tonynieman, 2024-05-19:
After updating to 1.37.1, I received this message again for seven Ikea devices.
OTA-Updates Option "IKEA TRADFRI OTA use test url" is not activated.