Koenkk/zigbee2mqtt #20659: Zigbee2MQTT stopped working "Configuration is not consistent with adapter state/backup!"

edsub, 2024-01-07:
### What happened?

I noticed my Zigbee network was not functioning anymore (because certain device events did not result in actions).
After checking the machine Zigbee2MQTT runs on I noticed it kept on restarting.
Examining the log revealed:
`debug 2024-01-05 23:36:00: Loaded state from file /opt/zigbee2mqtt/data/state.json
info  2024-01-05 23:36:00: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2024-01-05.23-36-00' filename: log.txt
debug 2024-01-05 23:36:00: Removing old log directory '/opt/zigbee2mqtt/data/log/2024-01-05.23-32-55'
info  2024-01-05 23:36:01: Starting Zigbee2MQTT version 1.34.0 (commit #56589dcc)
info  2024-01-05 23:36:01: Starting zigbee-herdsman (0.25.0)
debug 2024-01-05 23:36:01: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","databasePath":"/opt/zigbee2mqtt/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"/dev/ttyACM-ZIGBEE"}}'
error 2024-01-05 23:36:01: Configuration is not consistent with adapter state/backup!
error 2024-01-05 23:36:01: - PAN ID: configured=6754, adapter=65534
error 2024-01-05 23:36:01: - Extended PAN ID: configured=00124b0014d9529a, adapter=00124b0014d9529a
error 2024-01-05 23:36:01: - Network Key: configured=bc271738041ec3e12a0a1799d1f67f8d, adapter=bc271738041ec3e12a0a1799d1f67f8d
error 2024-01-05 23:36:01: - Channel List: configured=11, adapter=
error 2024-01-05 23:36:01: Please update configuration to prevent further issues.
error 2024-01-05 23:36:01: If you wish to re-commission your network, please remove coordinator backup at /opt/zigbee2mqtt/data/coordinator_backup.json.
error 2024-01-05 23:36:01: Re-commissioning your network will require re-pairing of all devices!
error 2024-01-05 23:36:01: Error while starting zigbee-herdsman
error 2024-01-05 23:36:01: Failed to start zigbee
error 2024-01-05 23:36:01: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
error 2024-01-05 23:36:01: Exiting...
error 2024-01-05 23:36:02: Error: startup failed - configuration-adapter mismatch - see logs above for more information
    at ZnpAdapterManager.determineStrategy (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:229:35)
    at ZnpAdapterManager.start (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:63:26)
    at Controller.start (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/opt/zigbee2mqtt/lib/zigbee.ts:60:27)
    at Controller.start (/opt/zigbee2mqtt/lib/controller.ts:101:27)
    at start (/opt/zigbee2mqtt/index.js:107:5)`

I noticed two inconsistencies here:
PAN ID: configured=6754, adapter=65534
Channel List: configured=11, adapter=

Question now is: How to fix this and in which files?

### What did you expect to happen?

Zigbee2mqtt keeps running

### Zigbee2MQTT version

1.35.1

### Adapter firmware version

20211115

### Adapter

CC2531 stick

sjorge, 2024-01-07:
I'm not 100% sure, but if you have a valid coordinator_backup file in your data/ directory, you should be able to restore it to your adapter after you reflash it.

edsub, 2024-01-08:
@sjorge OK, so you suggest to reflash the adapter?

edsub, 2024-01-08:
Fixed it. This is what I did:
1. Reflash my adapter
2. Restarted Zigbee2MQTT, but no luck. Noticed no new logfiles were created so figured startup crashed before anything could happen => probably an issue with node/npm
3. Uninstalled node/npm 
4. Reinstalled node/npm
5. Restarted Zigbee2MQTT, did not work as it complaine about existing advanced.network_key not being 'GENERATE' (I did not want that as I guess it would require me to repair all devices) and no array but a string
6. Edited configuration yaml and modified the network key string into a hex array
7. Restarted Zigbee2MQTT and ..... WORKING!!!
No clue why my config got messed up as nothing was going on when it broke down.
At least I have an updated Zigbee2MQTT and an updated node/npm

closed 2024-01-08 by edsub