Koenkk/zigbee2mqtt #20660: Wrong Device Identified

onechop, 2024-01-07:
### What happened?

Ikea LED2106R3 identified as LED2005R5

Photos of Box & Bulb with Model
[image]
[image]

### What did you expect to happen?

Identified as LED2106R3 which is on the Wiki 
https://www.zigbee2mqtt.io/devices/LED2106R3.html

### How to reproduce it (minimal and precise)

Join the device to Zigbee Network

### Zigbee2MQTT version

 1.35.0

### Adapter firmware version

 0x26780700

### Adapter

 ConBee2/RaspBee2

### Debug log

Entry from database.db
`{
    "id": 21,
    "type": "Router",
    "ieeeAddr": "0x5cc7c1fffe6a1451",
    "nwkAddr": 23164,
    "manufId": 4476,
    "manufName": "IKEA of Sweden",
    "powerSource": "Mains (single phase)",
    "modelId": "TRADFRI bulb GU10 WS 345lm",
    "epList": [
        1,
        242
    ],
    "endpoints": {
        "1": {
            "profId": 260,
            "epId": 1,
            "devId": 268,
            "inClusterList": [
                0,
                3,
                4,
                5,
                6,
                8,
                768,
                4096,
                64599,
                64636
            ],
            "outClusterList": [
                25
            ],
            "clusters": {
                "genBasic": {
                    "attributes": {
                        "modelId": "TRADFRI bulb GU10 WS 345lm",
                        "manufacturerName": "IKEA of Sweden",
                        "powerSource": 1,
                        "zclVersion": 8,
                        "appVersion": 17,
                        "stackVersion": 106,
                        "hwVersion": 1,
                        "dateCode": "20230425",
                        "swBuildId": "1.1.20"
                    }
                },
                "lightingColorCtrl": {
                    "attributes": {
                        "colorCapabilities": 16,
                        "colorTempPhysicalMin": 250,
                        "colorTempPhysicalMax": 454
                    }
                }
            },
            "binds": [],
            "configuredReportings": [],
            "meta": {}
        },
        "242": {
            "profId": 41440,
            "epId": 242,
            "devId": 97,
            "inClusterList": [
                33
            ],
            "outClusterList": [
                33
            ],
            "clusters": {},
            "binds": [],
            "configuredReportings": [],
            "meta": {}
        }
    },
    "appVersion": 17,
    "stackVersion": 106,
    "hwVersion": 1,
    "dateCode": "20230425",
    "swBuildId": "1.1.20",
    "zclVersion": 8,
    "interviewCompleted": true,
    "meta": {
        "configured": 1324213189
    },
    "lastSeen": 1704656758511,
    "defaultSendRequestWhen": "immediate"
}`

Koenkk, 2024-01-08:
Both devices are using the same modelID, I've renamed it to `LED2005R5/LED2106R3`

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-01-08 by Koenkk

mrmaximas, 2024-01-09:
@Koenkk please fix lost icons for this model
[image]

Koenkk, 2024-01-10:
Will be fixed with the next z2m release.

mrmaximas, 2024-02-01:
> Will be fixed with the next z2m release.

1.35.2-dev commit: [6621f6fd](https://github.com/Koenkk/zigbee2mqtt/commit/6621f6fd)
[image]