Koenkk/zigbee2mqtt #20661: _TYZB01_dl7cejts doesn't send any exposes when using zigbee2mqtt/bridge/devices

Burki24, 2024-01-07:
### What happened?

Since switch from 1.34.0 to 1.35.0 TuYa TY-ZPR06 (_TYZB01_dl7cejts) doesn't send any exposes when using zigbee2mqtt/bridge/devices. 

### What did you expect to happen?

To get an overview of exposes of this device (needed for including into IP Symcon), we use an user-extension, which sends the device-Infos to IP Symcon. There normaly the exposes will be extracted and finaly stored as variables:
[image]

The extension can be found here: https://github.com/Schnittcher/IPS-Zigbee2MQTT/blob/main/libs/IPSymconExtension.js

For all devices it runs perfectly except _TYZB01_dl7cejts. there are no exposes send:

### How to reproduce it (minimal and precise)

Start interviewing the device for getting the needed exposes. The definition comes without the part exposes.
[_TYZB01_dl7cejts.txt](https://github.com/Koenkk/zigbee2mqtt/files/13855056/_TYZB01_dl7cejts.txt)

### Zigbee2MQTT version

1.35.0-dev

### Adapter firmware version

zStack3.0 20210708

### Adapter

slaesh

Koenkk, 2024-01-08:
The extension is probably not compatible with 1.35.0 and has to be updated.

Burki24, 2024-01-08:
> The extension is probably not compatible with 1.35.0 and has to be updated.

Okay, I restructured the extension.js 
[Symcon_new.txt](https://github.com/Koenkk/zigbee2mqtt/files/13866303/Symcon_new.txt)

All devices and groups send ther exposes in the right way. Without this special device. In V. 1.35.1-dev it's well known, but it's not possible to find the exposes within the payload, called with zigbee2mqtt/bridge/devices. 
So for me, the Extension works perfectly, except this device. I have over 160 devices running perfect in zigbee2MQTT and Symcon, using this extension.

Have you any further idea, finding the issue? 

Sincerly
Burkhard

[image]
[image]

Koenkk, 2024-01-09:
You need to add `await` before some calls, this is due to https://github.com/Koenkk/zigbee2mqtt/commit/56f57625b0d5ce5b3bf6fa6a1a048f6ae8902311 (should also give some inspiration to update your user extension)

Schnittcher, 2024-01-25:
Hello @koenkk,
we need your support. According to the new exposes structure, we are trying to get them as a payload. So far we have been able to receive this as a payload through zigbee2mqtt/bridge/devices. However, after switching to 1.35.0 and the associated restructuring of the exposes, this no longer works.
Now how can we get the exposes using zigbee2mqtt/bridge/devices?
At the moment we simply cannot find the error in our extension.
Would it be possible for you to give us a code snippet to get us on the right track?
In your description it says that, among other things, the exposes are delivered with zigbee2mqtt/bridge/devices. But this no longer happens if the devices work with const.exps instead of exposes:.

Koenkk, 2024-01-26:
Nothing should have changed in the `exposes` of `zigbee2mqtt/bridge/devices`, what changed here? (can you give an example)

Burki24, 2024-01-26:
Hi @Koenkk,

thanks for your replay. I have it more detailed written in the new V1.35.1 discussion:
https://github.com/Koenkk/zigbee2mqtt/discussions/20637#discussioncomment-8252479

Also with samples what difference I mean.

Burki24, 2024-01-27:
@Koenkk 

This is what comes, when we try out 'zigbee2mqtt/bridge/devices':

There are no exposes shown in the payload.
```
{
    "zh":{
        "_events":{},
        "_eventsCount":0,
        "ID":52,
        "_applicationVersion":66,
        "_dateCode":"",
        "_endpoints":[{
            "_events":{},
            "_eventsCount":0,
            "deviceID":1026,
            "inputClusters":[1,3,1280,0],
            "outputClusters":[25,10],
            "profileID":260,
            "ID":1,
            "clusters":{
                "genBasic":{
                    "attributes":{
                        "modelId":"TS0207",
                        "manufacturerName":"_TZ3000_ocjlo4ea",
                        "powerSource":3,
                        "zclVersion":3,
                        "appVersion":66,
                        "stackVersion":0,
                        "hwVersion":1,
                        "dateCode":""
                    }
                },
                "ssIasZone":
                {
                    "attributes":
                    {
                        "iasCieAddr":"0x00124b00042ad5fa",
                        "zoneState":1
                    }
                },
                "genPowerCfg":
                {
                    "attributes":
                    {
                        "batteryPercentageRemaining":200
                    }
                }
            },
            "deviceIeeeAddress":"0xa4c1389a0e5a8872",
            "deviceNetworkAddress":2561,
            "_binds":
            [
                {
                    "cluster":1,
                    "type":"endpoint",
                    "deviceIeeeAddress":"0x00124b00042ad5fa",
                    "endpointID":1
                }
            ],
            "_configuredReportings":
            [
                {
                    "cluster":1,
                    "attrId":33,
                    "minRepIntval":3600,
                    "maxRepIntval":65000,
                    "repChange":0,
                    "manufacturerCode":null
[... 140 lines cut ...]
            "key":["scene_store"]
        },{
            "key":["scene_recall"]
        },{
            "key":["scene_add"]
        },{
            "key":["scene_remove"]
        },{
            "key":["scene_remove_all"]
        },{
            "key":["scene_rename"]
        },{
            "key":["read"]
        },{
            "key":["write"]
        },{
            "key":["command"]
        },{
            "key":["reset"]
        }
    ],
    "options":[]
},
"_definitionModelID":"TS0207"}
```

Koenkk, 2024-01-28:
Is this really published to `zigbee2mqtt/bridge/devices`? Can you provide the debug log when this is published?

See [this](https://www.zigbee2mqtt.io/guide/usage/debug.html) on how to enable debug logging.

closed 2024-02-02 by Burki24