Koenkk/zigbee2mqtt #20662: Firmware update fail for Aqara Curtain Motor E1 (ZNCLBL01LM) by Xiaomi

GitCharlieHub, 2024-01-07:
### What happened?

Zigbee2MQTT OTA Menu indicates firmware updates are available for the Aqara Curtain Motor E1 (ZNCLBL01LM). Also pops up in home assistant settings notifications.
Press "update device firmware". 
After a few minutes firmware update fails with message "Update of 'Living room Curtain' failed (OTA: Timeout, device did not request any image blocks)"

Current firmware according to Zigbee2MQTT
firmware date: May 31 2022	
firmware version: 0.0.0_0027

Can be updated to (seen on home assistant): 6168

Other devices (from Aqara like E1 thermostat) do update correctly

### What did you expect to happen?

.. firmware updated.

### How to reproduce it (minimal and precise)

press update firmware button
wait until error message appears

### Zigbee2MQTT version

1.35.1

### Adapter firmware version

6.10.3.0 build 297

### Adapter

Sonoff ZBdongle-E

### Debug log

Debug 2024-01-07 22:55:24Received MQTT message on 'zigbee2mqtt/bridge/request/device/ota_update/update' with data '{"id":"Living room Curtain","transaction":"7nfya-6"}'
Info 2024-01-07 22:55:24Updating 'Living room Curtain' to latest firmware
Debug 2024-01-07 22:55:24Received Zigbee message from 'Living room Curtain', type 'readResponse', cluster 'genBasic', data '{"dateCode":"May 31 2022","swBuildId":"0.0.0_0027"}' from endpoint 1 with groupID 0
Debug 2024-01-07 22:55:24OTA: Updating to latest '0x54ef441000560cfb' (lumi.curtain.agl001)
Debug 2024-01-07 22:55:24OTA: Using endpoint '1'
Debug 2024-01-07 22:55:25Received Zigbee message from 'Living room Curtain', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":27,"imageType":4105,"manufacturerCode":4447}' from endpoint 1 with groupID 0
Debug 2024-01-07 22:55:25OTA: Got request '{"fieldControl":0,"manufacturerCode":4447,"imageType":4105,"fileVersion":27}'
Debug 2024-01-07 22:55:25ZigbeeOTA: call getImageMeta for lumi.curtain.agl001
Debug 2024-01-07 22:55:25ZigbeeOTA: downloaded main index
Debug 2024-01-07 22:55:25OTA: Get new image for '0x54ef441000560cfb'
Debug 2024-01-07 22:55:25OTA: Get new image for '0x54ef441000560cfb', latest meta {"fileVersion":6168,"fileSize":288213,"url":"https://github.com/Koenkk/zigbee-OTA/raw/master/images/Xiaomi/20220506181329_lumi.curtain.agl001_Multi_JN5189_FMSH_0.0.0_2424_20220422_1aa302.ota","sha512":"0d1bae25759420898d381b98e97858d2e47f48cd3495abd8754661d0e1957eb5b570a2ae7435acc435d5a18665409e07148abae6f1f4a190c9af786d94ec725d"}
Debug 2024-01-07 22:55:25OTA: downloading firmware image from https://github.com/Koenkk/zigbee-OTA/raw/master/images/Xiaomi/20220506181329_lumi.curtain.agl001_Multi_JN5189_FMSH_0.0.0_2424_20220422_1aa302.ota using the zigbeeOTA custom CA certificates
Debug 2024-01-07 22:55:26OTA: Update checksum validation succeeded for '0x54ef441000560cfb'
Debug 2024-01-07 22:55:26OTA: Get new image for '0x54ef441000560cfb', image header {"otaUpgradeFileIdentifier":{"type":"Buffer","data":[30,241,238,11]},"otaHeaderVersion":256,"otaHeaderLength":56,"otaHeaderFieldControl":0,"manufacturerCode":4447,"imageType":4105,"fileVersion":6168,"zigbeeStackVersion":2,"otaHeaderString":"CURTAIN-OCC-JN5189---ENCRYPTED00","totalImageSize":288213}
Debug 2024-01-07 22:55:26OTA: Got new image for '0x54ef441000560cfb'
Debug 2024-01-07 22:55:26OTA: Starting upgrade
Debug 2024-01-07 22:55:26Received Zigbee message from 'Living room Curtain', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":27,"imageType":4105,"manufacturerCode":4447}' from endpoint 1 with groupID 0
Debug 2024-01-07 22:56:01Received Zigbee message from 'Coordinator', type 'commandNotification', cluster 'greenPower', data '{"data":[25,6,2,0],"type":"Buffer"}' from endpoint 242 with groupID null, ignoring since it is from coordinator
Debug 2024-01-07 22:57:56Update of 'Living room Curtain' failed (Error: OTA: Timeout, device did not request any image blocks)
Info 2024-01-07 22:57:56MQTT publish: topic 'zigbee2mqtt/Living room Curtain', payload '{"action":null,"battery":78,"charging":false,"device_temperature":23,"hand_open":true,"hooks_lock":"LOCK","hooks_state":"locked","illuminance_lux":0,"linkquality":192,"motor_state":"stopped","position":100,"power_outage_count":0,"power_source":"battery","running":false,"state":"OPEN","target_position":100,"update":{"installed_version":27,"latest_version":6168,"state":"available"},"update_available":null,"voltage":3000}'
Info 2024-01-07 22:57:56MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"id":"Living room Curtain"},"error":"Update of 'Living room Curtain' failed (OTA: Timeout, device did not request any image blocks)","status":"error","transaction":"7nfya-6"}'
Error 2024-01-07 22:57:56Update of 'Living room Curtain' failed (OTA: Timeout, device did not request any image blocks)
Debug 2024-01-07 22:57:56Error: OTA: Timeout, device did not request any image blocks at /app/node_modules/zigbee-herdsman-converters/src/lib/ota/common.ts:518:28 at runNextTicks (node:internal/process/task_queues:60:5) at listOnTimeout (node:internal/timers:538:9) at processTimers (node:internal/timers:512:7)

marcgarciamarti, 2024-01-14:
Hi,

I confirm this is happening. I have 4 of these curtain devices and OTA fail for mee too:

```
info  2024-01-13 15:39:03: MQTT publish: topic 'zigbee2mqtt/curtain2xiaomi', payload '{"action":null,"battery":53,"charging":true,"device_te>
info  2024-01-13 15:39:03: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Update of 'curtain2xiaomi' failed (OTA: Timeou>
info  2024-01-13 15:39:03: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"id":"curtain2xiaom>
error 2024-01-13 15:39:03: Update of 'curtain2xiaomi' failed (OTA: Timeout, device did not request any image blocks)
```

I'm on an updated coordinator firmware:
```
Zigbee2MQTT version
[1.35.1](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.35.1) commit: [unknown](https://github.com/Koenkk/zigbee2mqtt/commit/unknown)
Coordinator type
zStack3x0
Coordinator revision
20230507
Coordinator IEEE Address
0x00124b0012022476
Frontend version
0.6.151
Zigbee-herdsman-converters version
18.9.0
Zigbee-herdsman version
0.30.0
```

Happy to provide further logs/configurations or even test any beta version.

Thanks!

GitCharlieHub, 2024-01-26:
@marcgarciamarti did you manage to get your to update?

marcgarciamarti, 2024-01-27:
> @marcgarciamarti did you manage to get your to update?
> 
> Saw this post [#13801 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/13801#issuecomment-1238263065) where @Otnow mentioned a fix was coming.

Not fixed yet. See this: https://github.com/Koenkk/zigbee2mqtt/issues/15745#issuecomment-1892152805

closed 2024-01-31 by Koenkk

GitCharlieHub, 2024-01-31:
Thank you! @Koenkk

Otnow, 2024-01-31:
Note for [PR](https://github.com/Koenkk/zigbee-herdsman-converters/pull/6983): when pairing, device will immediately display correct firmware version, otherwise you need to wait until device sends next heartbeat message (~ 6-6,5 minutes) or perform its manual reconfiguration and then restart zigbee2mqtt (so that correct version is displayed in web interface).

noizex, 2024-04-15:
Hi, I think I still have similar error with updates being available but not updating in the end... 

Here's screenshots of all the version numbers:
[image]

Here's how it looks in OTA UI:
[image]

So the current version is 0.0.0_2727 and it seems to find a new one - this shows in generic "Update" UI for Home Assistant:
[image]

Here are the logs when I kick off the update (which fails after some time):
```
Debug 2024-04-16 00:51:57Received MQTT message on 'zigbee2mqtt/bridge/request/device/ota_update/update' with data '{"id":"Right Courtain Driver","transaction":"kquat-1"}'
Info 2024-04-16 00:51:57Updating 'Right Courtain Driver' to latest firmware
Debug 2024-04-16 00:51:57Received Zigbee message from 'Right Courtain Driver', type 'readResponse', cluster 'genBasic', data '{"dateCode":"May 31 2022","swBuildId":"0.0.0_0027"}' from endpoint 1 with groupID 0
Debug 2024-04-16 00:51:57OTA: Updating to latest '0x54ef4410005b6ad1' (lumi.curtain.agl001)
Debug 2024-04-16 00:51:57OTA: Using endpoint '1'
Debug 2024-04-16 00:51:58Received Zigbee message from 'Right Courtain Driver', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":27,"imageType":4105,"manufacturerCode":4447}' from endpoint 1 with groupID 0
Debug 2024-04-16 00:51:58OTA: Got request '{"fieldControl":0,"manufacturerCode":4447,"imageType":4105,"fileVersion":27}'
Debug 2024-04-16 00:51:58ZigbeeOTA: Getting image metadata for 'lumi.curtain.agl001'
Debug 2024-04-16 00:51:59ZigbeeOTA: Downloaded main index
Debug 2024-04-16 00:51:59OTA: Getting new image for '0x54ef4410005b6ad1' (lumi.curtain.agl001), latest meta {"fileVersion":6168,"fileSize":288213,"url":"https://github.com/Koenkk/zigbee-OTA/raw/master/images/Lumi/20220506181329_lumi.curtain.agl001_Multi_JN5189_FMSH_0.0.0_2424_20220422_1aa302.ota","sha512":"0d1bae25759420898d381b98e97858d2e47f48cd3495abd8754661d0e1957eb5b570a2ae7435acc435d5a18665409e07148abae6f1f4a190c9af786d94ec725d"}
Debug 2024-04-16 00:51:59OTA: Downloading firmware image from 'https://github.com/Koenkk/zigbee-OTA/raw/master/images/Lumi/20220506181329_lumi.curtain.agl001_Multi_JN5189_FMSH_0.0.0_2424_20220422_1aa302.ota' using the zigbeeOTA custom CA certificates
Debug 2024-04-16 00:51:59OTA: Update checksum validation succeeded for '0x54ef4410005b6ad1' (lumi.curtain.agl001)
Debug 2024-04-16 00:51:59OTA: Get new image for '0x54ef4410005b6ad1' (lumi.curtain.agl001), image header {"otaUpgradeFileIdentifier":{"type":"Buffer","data":[30,241,238,11]},"otaHeaderVersion":256,"otaHeaderLength":56,"otaHeaderFieldControl":0,"manufacturerCode":4447,"imageType":4105,"fileVersion":6168,"zigbeeStackVersion":2,"otaHeaderString":"CURTAIN-OCC-JN5189---ENCRYPTED00","totalImageSize":288213}
Debug 2024-04-16 00:51:59OTA: Got new image for '0x54ef4410005b6ad1' (lumi.curtain.agl001)
Debug 2024-04-16 00:51:59OTA: Starting upgrade
Debug 2024-04-16 00:52:00Received Zigbee message from 'Right Courtain Driver', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":27,"imageType":4105,"manufacturerCode":4447}' from endpoint 1 with groupID 0
Debug 2024-04-16 00:52:35Received Zigbee message from 'Living Room Main Light', type 'attributeReport', cluster 'genBasic', data '{"65506":31,"65508":0,"appVersion":68}' from endpoint 1 with groupID 0
Debug 2024-04-16 00:53:39Received Zigbee message from 'Right Courtain Driver', type 'attributeReport', cluster 'manuSpecificLumi', data '{"247":{"data":[1,33,184,11,3,40,26,5,33,1,0,10,33,54,121,12,32,1,13,35,27,27,0,0,100,32,2,101,32,132,102,32,131,103,32,6,104,33,27,1,105,32,1,106,32,0,107,32,100],"type":"Buffer"}}' from endpoint 1 with groupID 0
Debug 2024-04-16 00:53:39ZNCLBL01LM: Processed buffer into data {"1":3000,"3":26,"5":1,"10":31030,"12":1,"13":6939,"100":2,"101":132,"102":131,"103":6,"104":283,"105":1,"106":0,"107":100}
Debug 2024-04-16 00:53:40ZNCLBL01LM: unknown key 104 with value 283
Debug 2024-04-16 00:53:40ZNCLBL01LM: Processed data into payload {"voltage":3000,"device_temperature":26,"power_outage_count":0,"battery":66,"position":0,"state":"OPEN"}
Debug 2024-04-16 00:53:40ZNCLBL01LM: Processed data into payload {"voltage":3000,"device_temperature":26,"power_outage_count":0,"battery":66,"position":0,"state":"OPEN"}
Info 2024-04-16 00:53:40MQTT publish: topic 'zigbee2mqtt/Right Courtain Driver', payload '{"action":null,"battery":66,"charging":false,"device_temperature":26,"hand_open":true,"hooks_lock":"LOCK","hooks_state":"locked","illuminance_lux":0,"linkquality":255,"motor_state":"stopped","position":0,"power_outage_count":0,"power_source":"battery","running":false,"state":"OPEN","target_position":0,"update":{"installed_version":27,"latest_version":6168,"state":"available"},"update_available":null,"voltage":3000}'
Debug 2024-04-16 00:54:01Received Zigbee message from 'Left Courtain Driver', type 'read', cluster 'genTime', data '["time","timeZone"]' from endpoint 1 with groupID 0
Debug 2024-04-16 00:54:29Update of 'Right Courtain Driver' failed (Error: OTA: Timeout, device did not request any image blocks)
Info 2024-04-16 00:54:29MQTT publish: topic 'zigbee2mqtt/Right Courtain Driver', payload '{"action":null,"battery":66,"charging":false,"device_temperature":26,"hand_open":true,"hooks_lock":"LOCK","hooks_state":"locked","illuminance_lux":0,"linkquality":255,"motor_state":"stopped","position":0,"power_outage_count":0,"power_source":"battery","running":false,"state":"OPEN","target_position":0,"update":{"installed_version":27,"l [... cut ...] ,"status":"error","transaction":"kquat-1"}'
Error 2024-04-16 00:54:29Update of 'Right Courtain Driver' failed (OTA: Timeout, device did not request any image blocks)
```

I'm not really sure how from the detected version of 0.0.0_2727 it arrives at this update file:
`https://github.com/Koenkk/zigbee-OTA/raw/master/images/Lumi/20220506181329_lumi.curtain.agl001_Multi_JN5189_FMSH_0.0.0_2424_20220422_1aa302.ota`

Can someone advise? According to the my installed versions it should be fixed already?