Koenkk/zigbee2mqtt #20664: Pairing Legrand relay 412170 devices with v1.35.1 leave network after pairring

Eoras, 2024-01-07:
### What happened?

Hello
I'm using : LEGRAND DIN smart relay for light control ref: [412170](https://www.zigbee2mqtt.io/devices/412170.html#legrand-412170)

When i try to pair my relay, all work fine on Z2M but after configuration, device leave from the network (and my relay reset)

-> Reset my relay (fixed red led => waiting to join network)
-> Z2M : Permit join
-> Z2M: The relay is detected
-> Z2M : The relay is paired, all configuration is ok. At this moment, led had to become GREEN (success pairring) but he don't, he leave red fix.
-> Z2M : I can control my light but after 5/6 second the relay look like resets and leave network. At this moment, led red flash 1 time and fixe red (waiting to join network)
-> Z2M: The relay is detected
-> Z2M : The relay is paired ....
-> This repeat until i'm on permit joint on Z2M.

With ZHA he work fine, i don't know what happen with Z2M

### What did you expect to happen?

I expected that relay pair correctly without leaving automaticly after pairraing.

### How to reproduce it (minimal and precise)

Trying to pair 412170 with Z2M and with SONOFF ZigBee P CC2652P  (Same with SONOFF ZigBee E EFR32MG21 i tryed)

### Zigbee2MQTT version

1.35.1

### Adapter firmware version

20210708

### Adapter

 SONOFF ZigBee 3.0 USB Dongle Plus, TI CC2652P Coordinator

### Debug log

```
Info 2024-01-07 21:52:29MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0004740001241a63","ieee_address":"0x0004740001241a63"},"type":"device_leave"}'
Info 2024-01-07 21:52:31Device '0x0004740001241a63' joined
Info 2024-01-07 21:52:31MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0004740001241a63","ieee_address":"0x0004740001241a63"},"type":"device_joined"}'
Info 2024-01-07 21:52:31Starting interview of '0x0004740001241a63'
Info 2024-01-07 21:52:31MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0004740001241a63","ieee_address":"0x0004740001241a63","status":"started"},"type":"device_interview"}'
Debug 2024-01-07 21:52:31Device '0x0004740001241a63' announced itself
Info 2024-01-07 21:52:31MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0004740001241a63","ieee_address":"0x0004740001241a63"},"type":"device_announce"}'
Debug 2024-01-07 21:52:31Received Zigbee message from '0x0004740001241a63', type 'readResponse', cluster 'genBasic', data '{"appVersion":0}' from endpoint 1 with groupID 0
Debug 2024-01-07 21:52:31Skipping message, still interviewing
Info 2024-01-07 21:52:31Successfully interviewed '0x0004740001241a63', device has successfully been paired
Info 2024-01-07 21:52:31Device '0x0004740001241a63' is supported, identified as: Legrand DIN smart relay for light control (412170)
Info 2024-01-07 21:52:31MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"DIN smart relay for light control","exposes":[{"features":[{"access":7,"description":"On/off state of the switch","label":"State","name":"state","property":"state","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"},{"access":7,"description":"On/off (works only if device is in \"switch\" mode)","label":"State","name":"state","property":"state","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"access":5,"description":"Instantaneous measured power","label":"Power","name":"power","property":"power","type":"numeric","unit":"W"},{"access":7,"description":"switch: allow on/off, auto will use wired action via C1/C2 on teleruptor with buttons","label":"Device mode","name":"device_mode","property":"device_mode","type":"enum","values":["switch","auto"]},{"access":1,"description":"Link quality (signal strength)","label":"Linkquality","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"412170","options":[{"access":2,"description":"Calibrates the power value (percentual offset), takes into effect on next report of device.","label":"Power calibration","name":"power_calibration","property":"power_calibration","type":"numeric"},{"access":2,"description":"Number of digits after decimal point for power, takes into effect on next report of device. This option can only decrease the precision, not increase it.","label":"Power precision","name":"power_precision","property":"power_precision","type":"numeric","value_max":3,"value_min":0},{"access":2,"description":"State actions will also be published as 'action' when true (default false).","label":"State action","name":"state_action","property":"state_action","type":"binary","value_off":false,"value_on":true}],"supports_ota":true,"vendor":"Legrand"},"friendly_name":"0x0004740001241a63","ieee_address":"0x0004740001241a63","status":"successful","supported":true},"type":"device_interview"}'
Info 2024-01-07 21:52:31Configuring '0x0004740001241a63'
Debug 2024-01-07 21:52:39Received Zigbee message from '0x0004740001241a63', type 'read', cluster 'genBasic', data '[61440]' from endpoint 1 with groupID 0
Error 2024-01-07 21:52:41Failed to configure '0x0004740001241a63', attempt 1 (Error: Bind 0x0004740001241a63/1 genIdentify from '0x00124b002c3ac33e/1' failed (AREQ - ZDO - bindRsp after 10000ms) at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35) at listOnTimeout (node:internal/timers:569:17) at processTimers (node:internal/timers:512:7))
Warning 2024-01-07 21:52:42Device '0x0004740001241a63' left the network
Info 2024-01-07 21:52:42MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0004740001241a63","ieee_address":"0x0004740001241a63"},"type":"device_leave"}'
Info 2024-01-07 21:52:44Device '0x0004740001241a63' joined
Info 2024-01-07 21:52:44MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0004740001241a63","ieee_address":"0x0004740001241a63"},"type":"device_joined"}'
Info 2024-01-07 21:52:44Starting interview of '0x0004740001241a63'
Info 2024-01-07 21:52:44MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0004740001241a63","ieee_address":"0x0004740001241a63","status":"started"},"type":"device_interview"}'
Debug 2024-01-07 21:52:44Device '0x0004740001241a63' announced itself
Info 2024-01-07 21:52:44MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0004740001241a63","ieee_address":"0x0004740001241a63"},"type":"device_announce"}'
Debug 2024-01-07 21:52:44Received Zigbee message from '0x0004740001241a63', type 'readResponse', cluster 'genBasic', data '{"appVersion":0}' from endpoint 1 with groupID 0
Debug 2024-01-07 21:52:44Skipping message, still interviewing
Info 2024-01-07 21:52:44Successfully interviewed '0x0004740001241a63', device has successfully been paired
Info 2024-01-07 21:52:44Device '0x0004740001241a63' is supported, identified as: Legrand DIN smart relay for light control (412170)
Info 2024-01-07 21:52:44MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"DIN smart relay for light control","exposes":[{"features":[{"access":7,"description":"On/off state of the switch","label":"State","name":"state","property":"state","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"},{"access":7,"description":"On/off (works only if device is in \"switch\" mode)","label":"State","name":"state","property":"state","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"access":5,"description":"Instantaneous measured power","label":"Power","name":"power","property":"power","type":"numeric","unit":"W"},{"access":7,"description":"switch: allow on/off, auto will use wired action via C1/C2 on teleruptor with buttons","label":"Device mode","name":"device_mode","property":"device_mode","type":"enum","values":["switch","auto"]},{"access":1,"description":"Link quality (signal strength)","label":"Linkquality","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"412170","options":[{"access":2,"description":"Calibrates the power value (percentual offset), takes into effect on next report of device.","label":"Power calibration","name":"power_calibration","property":"power_calibration","type":"numeric"},{"access":2,"description":"Number of digits after decimal point for power, takes into effect on next report of device. This option c [... cut ...]
Debug 2024-01-07 21:57:06Received Zigbee message from 'Chambre ZG-227Z', type 'attributeReport', cluster 'msRelativeHumidity', data '{"measuredValue":3610}' from endpoint 1 with groupID 0
Debug 2024-01-07 21:57:06No converter available for 'ZG-227Z' with cluster 'msRelativeHumidity' and type 'attributeReport' and data '{"measuredValue":3610}'
Debug 2024-01-07 21:59:54Saving state to file /config/zigbee2mqtt/state.json
Debug 2024-01-07 22:04:54Saving state to file /config/zigbee2mqtt/state.json
Debug 2024-01-07 22:09:54Saving state to file /config/zigbee2mqtt/state.json
```

mircolino, 2024-01-08:
I'm experiencing the exact same issue with [Bticino L4411C/N4411C/NT4411C](https://www.zigbee2mqtt.io/devices/L4411C_N4411C_NT4411C.html)

In my case this problem started with v1.35.0 and was working fine with 1.34.0

Eoras, 2024-01-08:
I don't have the possibility to downgrade to 1.34.0 but some people have the same problem with both version and this divice... 
I think Z2M forgot to send something to finalize the pairring and therefore the divice resets. I don't know, i need help

Gjuju, 2024-01-08:
I have the exact same problem with 4 TS011F_plug_1 and a xiaomi lywsd03mmc.

With 1.35.1 (1.35.0 didn't start) and with my 1.34 backup.

Eoras, 2024-01-09:
Hello, I purchased a LEGRAND 4 121 81 control module in order to update my remote switches. I was on version 37 and they were updated to version 82.
I then reset the toggle switch, and reconnected it to Z2M and it works perfectly well now, the problem was fixed with the update. This requires having a Legrand Modul Control for the first update but it works perfectly well.

Koenkk, 2024-01-09:
@mircolino could you provide the herdsman debug log with both 1.34.0 (working) and 1.35.1 (not working)

See [this](https://www.zigbee2mqtt.io/guide/usage/debug.html) on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

Eoras, 2024-01-09:
I can't do it now but I ordered new switches, when I receive them, I won't update them right away to have the logs.

But with  SONOFF ZigBee E EFR32MG21 and updated switch relay it's working fine.

JiDW, 2024-01-12:
Hey!
I have the same issue.
I do not have the 1.34 logs at this moment as I'm not sure how to safely downgrade the add-on version of z2mqtt.

Here is the 1.35.1 log files tho
[legrandlogfile.log](https://github.com/Koenkk/zigbee2mqtt/files/13917563/legrandlogfile.log)

Thanks for your help!

steve41200, 2024-01-15:
Hey. I have the same pb.
During this week I've update Z2M add-on on HA.
This Sunday I discover 2 on my 3 https://www.zigbee2mqtt.io/devices/412170.html#legrand-412170
who my automatization via NodeRed seems works.
But the command from Z2M WebUi or dashboard d'ont work. But when the device is off, it really launch on (light is on in the house) but. don't see it.
I see some time log like this 

`<small>2024-01-14 18:57:55</small>`Publish 'set' 'state' to 'lumières allée jardin' failed: 'Error: Command 0x000474000085569a/1 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 42668 - 1 - 190 - 6 - 11 after 10000ms)'``

I have try to unpaired it. And when I repair I have the problem of device join well but leave the network quickly, rejoin ...
and light indicator on device stay on red and don't come green.

After some hours of trying it finally will work, but an another similar device do the same thing ....

I have discover to late this problem to downgrade the last version of z2m my backup was gone :(

tangane, 2024-01-16:
Hello. I have also exactly the same problem.  Should I move to ZHA (not possible to downgrade Z2M) or do you think a fix might be available in the coming days ?
Thanks a lot!

nicozocco, 2024-01-16:
I've been using a solution posted in a separate issue for this same thing and has been working, although it's a bit burdensome 

- Reset switch (blink red)
- Put it in pairing mode (blinking green)
- Allow join in Z2M
- Pay attention to the config messaged, once it states it Successfully Setup (and before it leaves the network) stop Z2M instance (docker or service), while the switch is still blinking green.
- Wait till the switch stops blinking
- Restart Z2M
- Reset switch and pair it again, it'll blink green, in Z2M it will setup successfuly and after a couple of blinks it'll blinkg magenta (successful pair)

steve41200, 2024-01-16:
did you have seen an another pb with the last update ?
Sometimes the teleruptor legrand did'nt respond (in HA or in Z2M) but when you click on it in HA he light on in realife. But don't respond in z2m or HA.I see in log Publish 'set' 'state' to 'lumières allée jardin' failed: 'Error: Command 0x000474000085569a/1 genOnOff.off({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 42668 - 1 - 141 - 6 - 11 after 10000ms)'

I cannot light off then from HA or z2M but with my automation from node red it works ...

Sunday it works like this for 2 of my 3 legrand and today 1 of the 3 ...

And in z2m the last seen is 7H, just aftet the warning message and the light is lighthning in reality ....
[image]

tangane, 2024-01-17:
After having pulled the last dev branch of z2m (not sure it helped actually) I tried what nicozocco wrote.
Delete from z2m
Reset the 412170
Put it back in pairing mode and allow pairing on z2m
Killed z2m juste after "pairing success"
And after a couple of pushes on the setup button the led lighted magenta (never saw thos color before)
z2m up again and... miracle, it works !
I did the same on all my 5 412170
This is a little bit blurry, but it worked ;)
Thanks a lot

nicozocco, 2024-01-17:
@steve41200 yes, im having similar issues, sometimes i press a switch and it does nothing at all. I haven't looked at the logs yet, will do it this weekend.

jngdan, 2024-01-20:
Hi, Thanks a lot, specially to nicozocco and tangane to provide this (a little bit special) procedure. I can correctly linked my 412171 module now, without any very short net leaving issues after a successful pairing as I experimented previously.
Congrats !

steve41200, 2024-01-20:
It is really strange. I try to reconnect my last (of 3 Legrand device teleruptor). One time I see the magenta light on dev branch. It will work, but after an another Legrand device who work will have the same comportement (light on but no return).

I'll be crazy with that

bonjour81, 2024-01-28:
Hi, I experience same with [Legrand 412173 ](https://www.zigbee2mqtt.io/devices/412173.html).
it's my 1st legrand device, and the first time I install it in z2m.
1.35.1 => "loop" in pairing / leaving.
1.34 => not sure what I did but after some tries, it seems finally paired and stays (some people talked about reboot z2m once it's paired, before it leaves)
I don't have magenta led, but green (maybe magenta is for other device).

skwat75, 2024-02-01:
On my side, the version 1.34 works fine, 1.35.1 -> "loop" in pairing / leaving. Iv'e got yhis behavior with all my legrand devices
Event if the dowgrade process is easy, it is painfull top switch version just to pair a device

sjorge, 2024-02-06:
Mine has been paired fine for a few months but I had a power outage and it was not responding, after trying to reset I also see the leave/join behavior and the device never leaves the red led state.

Whatever the manual seems seems incorrect as it never starts flashing (It says to hold >5 to reset, >3 (I assume between 3-5) to enter pair mode, the once green, push once to confirm and exit pair mode.

Mine just never turns green and seems to try and read an unknown attribute, then it imediatly leaves after

```
debug 2024-02-06 23:38:18: Received Zigbee message from '0x0004740000a1a444', type 'read', cluster 'genBasic', data '[61440]' from endpoint 1 with groupID 0
```

I'm guessing we need to reply to this one to have it stick, but not sure what.

hervespitz, 2024-02-11:
Hello,
Same issue here under HAOS and Z2M 1.35.3, I have a pairing/leaving loop. It's impossible to add a Legrand device to my fresh Z2M install

ddrag001, 2024-02-15:
I have the same issue with brand new BTicino FC80CC (Legrand 412171). As soon as the device joins, it leaves the network within 5 sec and then rejoins again.. basically a loop.

thk-socal, 2024-02-19:
I have the same issue. I have tried the following devices:

Coordinators: SkyConnect, SONOFF Dongle E, and now PoE TubesZB (ALL EZSP chips)
Devices: Inovelli Switches Blue 2-1 and Blue Fan, Over 12 different models of Hue Lights

At about 30 devices in the network EVERY device starts to join, interview, configure, successfully configure and then about 5-10 seconds later leaves the network. Brand and coordinator do not seem to matter at all.

The following will work to keep them on:
  Inovelli: pull the air gap to cut power to the switch the second it "successfully configured"
  Hue: Cut the power to the light the second it "successfully configured"
  Restart Z2M or change where joining is for a device

My guess is that there is some message that is either sent or expected after "successfully configured" and it is delayed as the network gets larger with lots of routers. Need someone who knows the code better to determine. I can also say it feels that if I join a new switch and use that switch as the place to permit joining that it seems to last a bit longer. This was not a problem in previous versions, as I was well over 200 devices in a single network and did not experience this until recently, not sure latest fully working firmware yet though.

sjorge, 2024-03-16:
This is probably related to https://github.com/Koenkk/zigbee-herdsman/issues/979, until this is fixed you can theoretically manually comment the readResponse coming from inside zh. I've tried this and then the device pairs again.

closed 2024-10-14 by github-actions[bot]

titou4307, 2024-11-04:
Hello
I have the same problem : 

I have pairing in the past 3x Legrand relay 412170 devices...

All was OK but after a time then leave network after pairring. I have reset each one (button passed Red) and after repairing..
All was OK but was lost after a time

Only one stay reliable, with all function, and state ON or OFF mount into Home Assistant
The two others when they return (after reset, restart) not maintain the real state (ON or OFF) and HA was not display the good state...

GofranAy, 2025-03-12:
> I've been using a solution posted in a separate issue for this same thing and has been working, although it's a bit burdensome
>
> * Reset switch (blink red)
> * Put it in pairing mode (blinking green)
> * [...]

first timer here can you please tell me wich one is the reset button is it small one with led around it ?

steve41200, 2025-04-05:
> first timer here can you please tell me wich one is the reset button is it small one with led around it ?

Yes it is the button with light around (not the big you can easily trigger with finger and have physical action on ight)

steve41200, 2025-04-05:
Since few day one of my teleruptor not response again ....
I have try to reset him but the blinking scene are different. always blink red a little, I ear a tick in the telerruptor and it stop.
If I try to pair z2m it works, and no leave network.
Unfortunetly it works few time and after without anything I hear a "tick" in teleruptor as the reset tick and action in z2m don't work

Same as the last pairing, but with no more information to act ...
This device seem a piece of shit
Regulary it don't work in z2m

Eoras, 2025-04-05:
As mentioned in several of my posts, this product is excellent.

However, Legrand hasn't taken the necessary steps to update its basic products. It's essential to update the firmware of each remote switch, but to do this, you have to use their Legrand Home Control module.

Here's the procedure that allowed me to solve all my problems with this device. I have 27 remote switches at home.

1) I purchased Legrand Home Control.
2) I synchronized all the remote switches with Legrand Home Control.
3) I updated the firmware of all my switches (need to be patient like 2/3 days, impossible to force it)
4) I reset each remote switch and connected them to my Home Assistant with Zigbee2Mqtt and my SONOFF key = 0 problem 
5) No more problems, and the OTA update via Z2M was successful.
6) I put my Legrand Home Connect away in the closet. 

That's it, it's very simple. Legrand sells its products out of date and the Zigbee synchronization validation of the commercial version of the firmware is not Z2M compatible

steve41200, 2025-04-12:
With the last z2m version, when I have repair the teleruptor who react strange, I have successfully update him (v65, v25 before).
After few day same things.
Sometimes:
- he loose information whiteout any change 
- or he send the command but noting appair in z2m or Ha 
- the action can come in long long time or not

and if I do nothing, sometimes it works again after x time

It seems pb of network, but i have other zigbee device in the electric board or in the same area, same very away then my coordinator with no pb.
The link quality is 112

steve41200, 2025-08-24:
> Hello i tried to update one legrand but it did not work. With z2m So mine question is does sombody already tried it with this. https://amzn.eu/d/2TmvYzg
> I think its a hub plug that i can also connect the legrand 412170  and update with that ?.

On z2m update is ok now, but it tale a long long long Time

lboue, 2026-11-08:
Adding a data point for anyone hitting this with a Legrand device: in my case (Legrand 067771 "Wired switch without neutral"), this join → configure → leave loop was caused by a very old factory firmware (`App 0.0 build 52` / `Stack 4.2 build 3`, dated 2021-09-03) that never got updated. Same symptom as reported here: joins, interviews, configures successfully, then leaves the network a few seconds later, in a loop.

What worked, without needing Legrand's own "Home + Control" hub:

1. **Get the device to stay joined long enough to be usable**: right after Z2M logs `Successfully configured`, press the device's pairing/reset button a few times (quickly) before it leaves — in my case the LED went green → magenta and it stuck.
2. **Improve RF conditions**: moved the coordinator/device closer to each other. Initial OTA attempts failed outright with `Device didn't respond to OTA request` at ~110 LQI; after improving link quality to 240-255 LQI, the OTA transfer worked.
3. **Run Zigbee2MQTT's native OTA update** (Settings → OTA page → Check, then Update). An update was available through the standard `Koenkk/zigbee-OTA` index (`App build 52 → 78`, `Stack 4.2 → 4.5`) — no Legrand hub/app required.

After the firmware update, the device has stayed joined normally with no further leave events. So for old/never-updated units, this looks less like a coordinator/stack compatibility problem and more like the device's own old firmware being flaky about finalizing the join — worth trying a Z2M OTA update before reaching for channel changes or a Legrand hub.

(Separately, this specific model also had an unrelated `configure()` crash — `lightingBallastCfg` bind failing with `INVALID_EP` — fixed in Koenkk/zigbee-herdsman-converters#13737, in case anyone sees that error too.)