Koenkk/zigbee2mqtt #20668: Cannot bind light groups to Lutron Aurora smart bulb dimmer (Z3-1BRL)

fractalcounty, 2024-01-08:
### What happened?

When attempting to bind the Lutron Aurora smart bulb dimmer to a group containing three Philips Hue bulbs, the dimmer switch stops producing responses and fails to work until it is factory reset and repaired to zigbee2mqtt. This issue has been experienced for a few years as reported in #16248 and #11342, with the latter being marked as closed. 

Additionally, the Lutron Aurora will regularly fail to respond with zigbee2mqtt unless the dimmer switch is "woken up" by being physically pressed or rotated. This may be unrelated to the issues with binding to groups and may very well be expected behavior of the Z3-1BRL.

### What did you expect to happen?

I expected to bind the Lutron Aurora to a light group in the same way I can bind my Philips Hue dimmer switch to light groups. However, this does not seem possible at the moment. Instead, I have to bind it to a dozen or so individual smart bulbs which is, as I understand, is undesirable compared to binding to groups for performance reasons. 

### How to reproduce it (minimal and precise)

1. Factory reset the Lutron Aurora smart bulb dimmer switch (Z3-1BRL) by quickly tapping the dimmer button 3 times, holding on the third tap until the LED blinks, and pressing the dimmer 3 more times to put the device into pairing mode.
2. Once the Aurora has been successfully connected with zigbee2mqtt, attempt to bind the device to a light group.
3. The Lutron Aurora will become unresponsive and fail to communicate with zigbee2mqtt until it is factory reset and paired again.

### Zigbee2MQTT version

1.33.0-dev commit: 9a7aeaf

### Adapter firmware version

7.1.1.0 build 273

### Adapter

Home Assistant SkyConnect (EZSP v9)

### Debug log

N/A.   
  
The logs below are not particularly relevant as they only show expected behavior of the Z3-1BRL when bound to individual bulbs as opposed to groups, as binding to groups results in the Z3-1BRL (or zigbee2mqtt itself?) locking up and not producing any response with or without debug logs enabled from what I can tell.  
  
```
Debug 2024-01-08 01:25:20Received Zigbee message from 'lutron_dimmer_lower_bathroom', type 'commandMoveToLevelWithOnOff', cluster 'genLevelCtrl', data '{"level":95,"transtime":2}' from endpoint 1 with groupID 0
Info 2024-01-08 01:25:20MQTT publish: topic 'zigbee2mqtt/lutron_dimmer_lower_bathroom', payload '{"action":"brightness_move_to_level","action_level":95,"action_transition_time":0.02,"brightness":null,"linkquality":152,"update":{"installed_version":3080,"latest_version":3080,"state":"idle"},"update_available":null}'
Info 2024-01-08 01:25:20MQTT publish: topic 'zigbee2mqtt/lutron_dimmer_lower_bathroom', payload '{"action":"","brightness":null,"linkquality":152,"update":{"installed_version":3080,"latest_version":3080,"state":"idle"},"update_available":null}'
Info 2024-01-08 01:25:20MQTT publish: topic 'zigbee2mqtt/lutron_dimmer_lower_bathroom/action', payload 'brightness_move_to_level'
Debug 2024-01-08 01:25:20Received Zigbee message from 'hue_bulb_100w_white_ambience_lower_bathroom_a', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":22}' from endpoint 11 with groupID 0
Info 2024-01-08 01:25:20MQTT publish: topic 'zigbee2mqtt/lower_bathroom_lights', payload '{"brightness":22,"color":{"h":27,"hue":27,"s":92,"saturation":92,"x":0.5056,"y":0.4152},"color_mode":"color_temp","color_temp":454,"state":"OFF"}'
Info 2024-01-08 01:25:20MQTT publish: topic 'zigbee2mqtt/hue_bulb_100w_white_ambience_lower_bathroom_a', payload '{"brightness":22,"color_mode":"color_temp","color_temp":454,"linkquality":208,"power_on_behavior":null,"state":"OFF","update":{"installed_version":16785162,"latest_version":16785162,"state":"idle"},"update_available":null}'
Debug 2024-01-08 01:25:20Received Zigbee message from 'lutron_dimmer_lower_bathroom', type 'commandMoveToLevelWithOnOff', cluster 'genLevelCtrl', data '{"level":153,"transtime":2}' from endpoint 1 with groupID 0
```

jacobb, 2024-01-08:
Also experiencing this. I was able to (after a lot of manual labor + use of deconz+ downgrade one of the switches to version 3040, which DOES allow the pairing of group, but it's otherwise buggy -- dimming is inconsistent at best and I get regular "failed to poll" errors when taking any action with the dimmer.

jacobb, 2024-05-09:
@fractalcounty was wondering if you ever made headway with this. I just swapped both HA hardware and zigbee adapter and have the same exact issue as before.

fractalcounty, 2024-05-23:
> @fractalcounty was wondering if you ever made headway with this. I just swapped both HA hardware and zigbee adapter and have the same exact issue as before.

Nope, from what I can gather you can still only bind the LvlCtrl (and maybe OnOff?) cluster to each individual light. 

I've been running mine this way for over a year now and haven't had any issues, even if it maybe isn't the best practice. 

If the setup process is being a bit tricky, try factory resetting and remember to twiddle the knob when binding through z2m to ensure it's "awake".

fractalcounty, 2024-12-29:
Any update on this? I ran into separate issues with the Z3-1BRL today and can confirm group binding is broken.

closed 2025-03-09 by github-actions[bot]