Koenkk/zigbee2mqtt #20669: AwoX Modell [33955] - If I enter a value for Transition, no transition

Seger85, 2024-01-08:
### What happened?

If I enter a value for Transition, no transition is executed by the lamp, and the lamp no longer switches off if any value is entered for Transition (see point below: reproduce)

Details of the lamp from Z2M

Manufacturer
[AwoX]
Modell
[33955]
Firmware-Version
2.7.3_1387

`const {light} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['EGLO_ZM_TW'],
    model: 'EGLO_ZM_TW',
    vendor: 'AwoX',
    description: 'Automatically generated definition',
    extend: [light({"colorTemp":{"range":[153,370]}})],

https://www.zigbee2mqtt.io/devices/33955.html

### What did you expect to happen?

The transition is applied when it is set. The request would be to find a technical solution or workaround for this, as the manufacturer has created something incorrectly. I have also written to the manufacturer, but so far no response. I have installed 20 lamps in my house and am very unhappy at the moment and would really appreciate some help. If there is any way to find a solution, please help me find out exactly what needs to be entered.

Many thanks
Seger

### How to reproduce it (minimal and precise)

Transition if you enter a value 0,1,2,3 the lamp no longer switches off. Transition is not supported. The field must be left empty, only then is proper operation possible, but without transition. 

### Zigbee2MQTT version

 1.35.1

### Adapter firmware version

7.3.2.0 build 212

### Adapter

Skyconect

sdotter, 2024-01-08:
Don't use awox devices 😜

sdotter, 2024-01-08:
It stewed up my zigbee network

Seger85, 2024-01-08:
too late, but the bug you are talking about no longer exists, they have released about 7 firmware updates in 1.5 years.
Unfortunately, I installed 20 ceiling lights and thought, well, Zigbee Standard -> great, then nothing can go wrong. I picked out the black sharp, so to speak. But now I'm almost satisfied, just transition.

Please help with this :-)

Seger85, 2024-01-13:
maybe a little push before the bot strikes :-), haha :-)

closed 2024-08-11 by github-actions[bot]