Koenkk/zigbee2mqtt #20670: Re-Open: Interview fails after deleting and re-pairing the same Zigbee device

theimo1221, 2024-01-08:
### What happened?

Interview fails after deleting and re-pairing the same Zigbee device

This is reported by several people in closed issue #14135

### What did you expect to happen?

Normal interview being succesfull

### How to reproduce it (minimal and precise)

1. Force delete device (some Ikea/Sonoff/illuminize/Ubisys device)
2. Try to repair it
--> Device is half paired (showing warning triangle with text `Interview failed`)

### Zigbee2MQTT version

1.35.0

### Adapter firmware version

 20230922

### Adapter

Sonoff Dongle P

Gjuju, 2024-01-08:
Same problem here.

Z2m 1.35.1
Sonoff dongle e rcp 7.3.1.0
Ha multiprotocol 2.3.2

Realy annoying.
I can't pair anymore 3 TS011F_plug_1 and a xiaomi lywsd03mmc I used before.

It pairs then leave network.
It seems the device stay in pairing mode after it connects.

Eoras, 2024-01-08:
> Same problem here.
> 
> Z2m 1.35.1 Sonoff dongle e rcp 7.3.1.0 Ha multiprotocol 2.3.2
> 
> Realy annoying. I can't pair anymore 3 TS011F_plug_1 and a xiaomi lywsd03mmc I used before.
> 
> It pairs then leave network. It seems the device stay in pairing mode after it connects.

This look like same problem with my LEGRAND : https://github.com/Koenkk/zigbee2mqtt/issues/20664

sveinse, 2024-01-09:
I have the same problem for many IKEA devices. This happens when units have already been paired with z2m previously. New devices seems to pair on first attempt.

The fix that seems to work for me is:

1. Force remove the device from z2m
2. Restart zigbee2mqtt
3. Repair the device

TonKro, 2024-01-13:
This was also for me the solution.

Krikand, 2024-01-21:
Same problem here, for Z2M 1.28.4 / zigbee-herdsman 0.14.76
I use the same workaround (force remove, restart Z2M, pairing) when a device for some reason loose connection. Sonoff [SNZB-02] can be a pain.

wormvortex, 2024-01-23:
Doesn’t work for me. Removed device. Restarted. Tried to repair and it still fails to add.

LouBen3010, 2024-02-20:
Same here. Force removed a LEDVANCE bulb and now the inteview process is failing. Restartet everything. Nothing works for me.

LouBen3010, 2024-02-20:
After deleting the un-paired device without force (and many error messages) the state seems to be good again and a pairing was possible.

satrik, 2024-02-25:
I have similar issues with 2 Nous A1Z plugs. I force removed them and now Im not able to pair them again, they are not even seen by z2m anymore….

satrik, 2024-05-07:
@Koenkk can you take a look at the problems here when you get the chance? I would like to re-use my Nous A1Z plugs again -.- 
Im running Zigbee2MQTT 1.37.0 as service on a raspberry pi 4 (4GB) with a Sonoff Dongle P (20230507). 
Neither the logs nor the journal are showing anything if I go into pairing mode, so I don't know what else I could share or where to start debugging

BoneheadFraggle, 2024-05-29:
> @Koenkk can you take a look at the problems here when you get the chance? I would like to re-use my Nous A1Z plugs again -.- Im running Zigbee2MQTT 1.37.0 as service on a raspberry pi 4 (4GB) with a [...]

I have the exact same issue with the same setup, Sonoff Dongle P and Nous A1Z plugs (but with HA and Z2M in Proxmox).
Nothing in the logs, just as if I didn't even try to pair. Adding other (not previously added) devices works fine.

0xPDG, 2024-07-18:
Same issue as @satrik and @BoneheadFraggle.
Nous A1Z with SONOFF ZigBee 3.0 USB Dongle Plus, TI CC2652 and Z2M.

jeroenst, 2024-08-07:
Same issue here, even force removing device, restarting zigbee2mqtt and rejoining device doesn't work.

Dongle: https://www.amazon.nl/dp/B09KXTCMSC
Zigbee Device: https://zigbee.blakadder.com/AwoX_33948.html

```
[2024-08-07 19:22:43] info: 	z2m: Logging to console, file (filename: log.log)
[2024-08-07 19:22:43] info: 	z2m: Starting Zigbee2MQTT version 1.39.0 (commit #0326926)
[2024-08-07 19:22:43] info: 	z2m: Starting zigbee-herdsman (0.50.1)
[2024-08-07 19:22:43] info: 	zh:zstack:znp: Opening SerialPort with {"path":"/dev/ttyACM0","baudRate":115200,"rtscts":false,"autoOpen":false}
[2024-08-07 19:22:43] info: 	zh:zstack:znp: Serialport opened
[2024-08-07 19:22:43] info: 	z2m: zigbee-herdsman started (resumed)
[2024-08-07 19:22:43] info: 	z2m: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20210708,"transportrev":2},"type":"zStack3x0"}'
[2024-08-07 19:22:43] info: 	z2m: Currently 0 devices are joined:
[2024-08-07 19:22:43] warning: 	z2m: `permit_join` set to  `true` in configuration.yaml.
[2024-08-07 19:22:43] warning: 	z2m: Allowing new devices to join.
[2024-08-07 19:22:43] warning: 	z2m: Set `permit_join` to `false` once you joined all devices.
[2024-08-07 19:22:43] info: 	z2m: Zigbee: allowing new devices to join.
[2024-08-07 19:22:44] info: 	z2m: Connecting to MQTT server at mqtts://mqtt.hiddendomainname.nl
[2024-08-07 19:22:44] info: 	z2m: Connected to MQTT server
[2024-08-07 19:22:44] info: 	z2m:mqtt: MQTT publish: topic 'home/zigbee2mqtt/bridge/state', payload '{"state":"online"}'
[2024-08-07 19:22:44] info: 	z2m: Started frontend on port 8080
[2024-08-07 19:22:44] info: 	z2m: Zigbee2MQTT started!
[2024-08-07 19:23:27] info: 	zh:controller: Interview for '0xa4c1381b2535107d' started
[2024-08-07 19:23:27] info: 	z2m: Device '0xa4c1381b2535107d' joined
[2024-08-07 19:23:27] info: 	z2m: Starting interview of '0xa4c1381b2535107d'
[2024-08-07 19:23:27] info: 	z2m:mqtt: MQTT publish: topic 'home/zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1381b2535107d","ieee_address":"0xa4c1381b2535107d"},"type":"device_joined"}'
[2024-08-07 19:23:27] info: 	z2m:mqtt: MQTT publish: topic 'home/zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1381b2535107d","ieee_address":"0xa4c1381b2535107d","status":"started"},"type":"device_interview"}'
[2024-08-07 19:25:45] error: 	zh:controller: Interview failed for '0xa4c1381b2535107d with error 'Error: Interview failed because can not get node descriptor ('0xa4c1381b2535107d')'
[2024-08-07 19:25:45] error: 	z2m: Failed to interview '0xa4c1381b2535107d', device has not successfully been paired
[2024-08-07 19:25:45] info: 	z2m:mqtt: MQTT publish: topic 'home/zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1381b2535107d","ieee_address":"0xa4c1381b2535107d","status":"failed"},"type":"device_interview"}'
```

jeroenst, 2024-08-07:
Updating the dongle firmware to didn't change anything:

```
[2024-08-07 19:49:30] info: 	z2m: Logging to console, file (filename: log.log)
[2024-08-07 19:49:30] info: 	z2m: Starting Zigbee2MQTT version 1.39.0 (commit #0326926)
[2024-08-07 19:49:30] info: 	z2m: Starting zigbee-herdsman (0.50.1)
[2024-08-07 19:49:30] info: 	zh:zstack:znp: Opening SerialPort with {"path":"/dev/ttyACM0","baudRate":115200,"rtscts":false,"autoOpen":false}
[2024-08-07 19:49:30] info: 	zh:zstack:znp: Serialport opened
[2024-08-07 19:49:33] info: 	z2m: zigbee-herdsman started (resumed)
[2024-08-07 19:49:33] info: 	z2m: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20230507,"transportrev":2},"type":"zStack3x0"}'
[2024-08-07 19:49:33] info: 	z2m: Currently 0 devices are joined:
[2024-08-07 19:49:33] warning: 	z2m: `permit_join` set to  `true` in configuration.yaml.
[2024-08-07 19:49:33] warning: 	z2m: Allowing new devices to join.
[2024-08-07 19:49:33] warning: 	z2m: Set `permit_join` to `false` once you joined all devices.
[2024-08-07 19:49:33] info: 	z2m: Zigbee: allowing new devices to join.
[2024-08-07 19:49:33] info: 	z2m: Connecting to MQTT server at mqtts://mqtt.hiddendomainname.nl
[2024-08-07 19:49:33] info: 	z2m: Connected to MQTT server
[2024-08-07 19:49:33] info: 	z2m:mqtt: MQTT publish: topic 'home/zigbee2mqtt/bridge/state', payload '{"state":"online"}'
[2024-08-07 19:49:33] info: 	z2m: Started frontend on port 8080
[2024-08-07 19:49:33] info: 	z2m: Zigbee2MQTT started!
[2024-08-07 19:50:00] info: 	zh:controller: Interview for '0xa4c1381b2535107d' started
[2024-08-07 19:50:00] info: 	z2m: Device '0xa4c1381b2535107d' joined
[2024-08-07 19:50:00] info: 	z2m: Starting interview of '0xa4c1381b2535107d'
[2024-08-07 19:50:00] info: 	z2m:mqtt: MQTT publish: topic 'home/zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1381b2535107d","ieee_address":"0xa4c1381b2535107d"},"type":"device_joined"}'
[2024-08-07 19:50:00] info: 	z2m:mqtt: MQTT publish: topic 'home/zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1381b2535107d","ieee_address":"0xa4c1381b2535107d","status":"started"},"type":"device_interview"}'
[2024-08-07 19:52:18] error: 	zh:controller: Interview failed for '0xa4c1381b2535107d with error 'Error: Interview failed because can not get node descriptor ('0xa4c1381b2535107d')'
[2024-08-07 19:52:18] error: 	z2m: Failed to interview '0xa4c1381b2535107d', device has not successfully been paired
[2024-08-07 19:52:18] info: 	z2m:mqtt: MQTT publish: topic 'home/zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1381b2535107d","ieee_address":"0xa4c1381b2535107d","status":"failed"},"type":"device_interview"}'
```

jeroenst, 2024-08-07:
Debug log while joining:

```
[2024-08-07 20:06:50] debug: 	zh:zstack:unpi:parser: <-- [254,12,69,202,150,77,125,16,53,37,27,56,193,164,0,0,99]
[2024-08-07 20:06:50] debug: 	zh:zstack:unpi:parser: --- parseNext [254,12,69,202,150,77,125,16,53,37,27,56,193,164,0,0,99]
[2024-08-07 20:06:50] debug: 	zh:zstack:unpi:parser: --> parsed 12 - 2 - 5 - 202 - [150,77,125,16,53,37,27,56,193,164,0,0] - 99
[2024-08-07 20:06:50] debug: 	zh:zstack:znp: AREQ: <-- ZDO - tcDeviceInd - {"nwkaddr":19862,"extaddr":"0xa4c1381b2535107d","parentaddr":0}
[2024-08-07 20:06:50] debug: 	zh:controller: Device '0xa4c1381b2535107d' joined
[2024-08-07 20:06:50] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-08-07 20:06:50] debug: 	zh:controller: Device '0xa4c1381b2535107d' accepted by handler
[2024-08-07 20:06:50] debug: 	zh:controller: New device '0xa4c1381b2535107d' joined
[2024-08-07 20:06:50] debug: 	zh:controller: Creating device '0xa4c1381b2535107d'
[2024-08-07 20:06:50] debug: 	zh:controller:database: Writing database to '/app/data/database.db'
[2024-08-07 20:06:50] info: 	zh:controller: Interview for '0xa4c1381b2535107d' started
[2024-08-07 20:06:50] debug: 	zh:controller:device: Interview - start device '0xa4c1381b2535107d'
[2024-08-07 20:06:50] debug: 	zh:zstack:znp: SREQ: --> ZDO - nodeDescReq - {"dstaddr":19862,"nwkaddrofinterest":19862}
[2024-08-07 20:06:50] debug: 	zh:zstack:unpi:writer: --> frame [254,4,37,2,150,77,150,77,35]
[2024-08-07 20:06:50] info: 	z2m: Device '0xa4c1381b2535107d' joined
[2024-08-07 20:06:50] info: 	z2m: Starting interview of '0xa4c1381b2535107d'
[2024-08-07 20:06:50] info: 	z2m:mqtt: MQTT publish: topic 'home/zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1381b2535107d","ieee_address":"0xa4c1381b2535107d"},"type":"device_joined"}'
[2024-08-07 20:06:50] info: 	z2m:mqtt: MQTT publish: topic 'home/zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1381b2535107d","ieee_address":"0xa4c1381b2535107d","status":"started"},"type":"device_interview"}'
[2024-08-07 20:06:50] debug: 	zh:zstack:unpi:parser: <-- [254,1,101,2,0,102]
[2024-08-07 20:06:50] debug: 	zh:zstack:unpi:parser: --- parseNext [254,1,101,2,0,102]
[2024-08-07 20:06:50] debug: 	zh:zstack:unpi:parser: --> parsed 1 - 3 - 5 - 2 - [0] - 102
[2024-08-07 20:06:50] debug: 	zh:zstack:znp: SRSP: <-- ZDO - nodeDescReq - {"status":0}
[2024-08-07 20:06:50] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-08-07 20:07:00] debug: 	zh:zstack: Node descriptor request for '19862' failed (Error: AREQ - ZDO - nodeDescRsp after 10000ms), retry
[2024-08-07 20:07:00] debug: 	zh:zstack: Discovering route to 19862
[2024-08-07 20:07:00] debug: 	zh:zstack:znp: SREQ: --> ZDO - extRouteDisc - {"dstAddr":19862,"options":0,"radius":30}
[2024-08-07 20:07:00] debug: 	zh:zstack:unpi:writer: --> frame [254,4,37,69,150,77,0,30,161]
[2024-08-07 20:07:00] debug: 	zh:zstack:unpi:parser: <-- [254,1,101,69,0,33]
[2024-08-07 20:07:00] debug: 	zh:zstack:unpi:parser: --- parseNext [254,1,101,69,0,33]
[2024-08-07 20:07:00] debug: 	zh:zstack:unpi:parser: --> parsed 1 - 3 - 5 - 69 - [0] - 33
[2024-08-07 20:07:00] debug: 	zh:zstack:znp: SRSP: <-- ZDO - extRouteDisc - {"status":0}
[2024-08-07 20:07:00] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-08-07 20:07:03] debug: 	zh:zstack:znp: SREQ: --> ZDO - nodeDescReq - {"dstaddr":19862,"nwkaddrofinterest":19862}
[2024-08-07 20:07:03] debug: 	zh:zstack:unpi:writer: --> frame [254,4,37,2,150,77,150,77,35]
[2024-08-07 20:07:03] debug: 	zh:zstack:unpi:parser: <-- [254,1,101,2,0,102]
[2024-08-07 20:07:03] debug: 	zh:zstack:unpi:parser: --- parseNext [254,1,101,2,0,102]
[2024-08-07 20:07:03] debug: 	zh:zstack:unpi:parser: --> parsed 1 - 3 - 5 - 2 - [0] - 102
[2024-08-07 20:07:03] debug: 	zh:zstack:znp: SRSP: <-- ZDO - nodeDescReq - {"status":0}
[2024-08-07 20:07:03] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-08-07 20:07:13] debug: 	zh:controller:device: Interview - quirks check for 'undefined'-'undefined'-'Unknown'
[2024-08-07 20:07:13] debug: 	zh:controller:device: Interview - quirks did not match
[2024-08-07 20:07:13] debug: 	zh:controller:device: Interview - node descriptor request failed for '0xa4c1381b2535107d', attempt 1
[2024-08-07 20:07:13] debug: 	zh:zstack:znp: SREQ: --> ZDO - nodeDescReq - {"dstaddr":19862,"nwkaddrofinterest":19862}
[2024-08-07 20:07:13] debug: 	zh:zstack:unpi:writer: --> frame [254,4,37,2,150,77,150,77,35]
[2024-08-07 20:07:13] debug: 	zh:zstack:unpi:parser: <-- [254,1,101,2,0,102]
[2024-08-07 20:07:13] debug: 	zh:zstack:unpi:parser: --- parseNext [254,1,101,2,0,102]
[2024-08-07 20:07:13] debug: 	zh:zstack:unpi:parser: --> parsed 1 - 3 - 5 - 2 - [0] - 102
[2024-08-07 20:07:13] debug: 	zh:zstack:znp: SRSP: <-- ZDO - nodeDescReq - {"status":0}
[2024-08-07 20:07:13] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-08-07 20:07:23] debug: 	zh:zstack: Node descriptor request for '19862' failed (Error: AREQ - ZDO - nodeDescRsp after 10000ms), retry
[2024-08-07 20:07:23] debug: 	zh:zstack: Discovering route to 19862
[2024-08-07 20:07:23] debug: 	zh:zstack:znp: SREQ: --> ZDO - extRouteDisc - {"dstAddr":19862,"options":0,"radius":30}
[2024-08-07 20:07:23] debug: 	zh:zstack:unpi:writer: --> frame [254,4,37,69,150,77,0,30,161]
[2024-08-07 20:07:23] debug: 	zh:zstack:unpi:parser: <-- [254,1,101,69,0,33]
[2024-08-07 20:07:23] debug: 	zh:zstack:unpi:parser: --- parseNext [254,1,101,69,0,33]
[... cut ...]
[2024-08-07 20:09:08] error: 	zh:controller: Interview failed for '0xa4c1381b2535107d with error 'Error: Interview failed because can not get node descriptor ('0xa4c1381b2535107d')'
[2024-08-07 20:09:08] error: 	z2m: Failed to interview '0xa4c1381b2535107d', device has not successfully been paired
[2024-08-07 20:09:08] info: 	z2m:mqtt: MQTT publish: topic 'home/zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1381b2535107d","ieee_address":"0xa4c1381b2535107d","status":"failed"},"type":"device_interview"}'
```

Reverting to Tasmota-Zigbee for now.

jeroenst, 2024-08-07:
The device was not fully reset why it failed to pair. After resetting it a few time with the vendors instruction video the lamp started flashing confirming the reset. Joining worked flawless after that.

I also updated the usb dongle in my attempts to succeed: https://github.com/git-developer/ti-cc-tool

https://awox.support.awox.group/kb/article/1406-how-to-reset-a-bulb-or-a-luminaire/

satrik, 2025-01-04:
Push 

Issue is still unsolved

satrik, 2025-03-10:
Push

I just tested it again with the newest Zigbee2MQTT-Version 2.1.3 and this issue is still unsolved...

satrik, 2025-05-16:
still not working, but as I move my complete setup to home assistant, I think I will just move my zigbee stuff to ZHA. maybe I can then finally use all of my devices again 🤞🏻

jannikac, 2025-07-15:
Hi i had the same issue after migrating form zha to z2m. What i had to do was update the firmware on the SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P itself. after that it started working perfectly. I used this [tutorial](https://www.youtube.com/watch?v=-o_jG7Ll1PE). it's german but you can enable english subtitles. Basically it uses the [cc2538-bsl ](https://github.com/JelmerT/cc2538-bsl) script with the firmware from the [z2m page](https://www.zigbee2mqtt.io/guide/adapters/zstack.html) to update the stick.

closed 2025-09-21 by github-actions[bot]