Koenkk/zigbee2mqtt #20676: Philips Dimmer V2 (929002398602) OnOff sync

markchalloner, 2024-01-08:
### What happened?

When using the single power button on a Philips Hue Dimmer V2 bound either to a group (with Hue bulb) or single Hue bulb, the on/off button gets out of sync requiring a second press to turn the light on or off.

Debug logs show that the Dimmer variously sends a commandOn and commandOffWithEffect, whereas it probably *should* send a commandToggle?

```
debug 2024-01-08 15:37:52Received Zigbee message from 'Office Light Dimmer', type 'commandOffWithEffect', cluster 'genOnOff', data '{"effectid":0,"effectvariant":0}' from endpoint 1 with groupID 51433
debug 2024-01-08 15:37:52Received Zigbee message from 'Office Light Dimmer', type 'commandOffWithEffect', cluster 'genOnOff', data '{"effectid":0,"effectvariant":0}' from endpoint 1 with groupID 38372
debug 2024-01-08 15:37:52Received Zigbee message from 'Office Light Bulb', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 11 with groupID 0
debug 2024-01-08 15:37:52Received Zigbee message from 'Office Light Dimmer', type 'commandOffWithEffect', cluster 'genOnOff', data '{"effectid":0,"effectvariant":0}' from endpoint 1 with groupID 0
debug 2024-01-08 15:37:53Received Zigbee message from 'Office Light Bulb', type 'readResponse', cluster 'genOnOff', data '{"onOff":0}' from endpoint 11 with groupID 0
debug 2024-01-08 15:37:54Received Zigbee message from 'Office Light Dimmer', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 51433
debug 2024-01-08 15:37:54Received Zigbee message from 'Office Light Dimmer', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
debug 2024-01-08 15:37:54Received Zigbee message from 'Office Light Dimmer', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 38372
debug 2024-01-08 15:37:54Received Zigbee message from 'Office Light Bulb', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 11 with groupID 0
debug 2024-01-08 15:37:55Received Zigbee message from 'Office Light Bulb', type 'readResponse', cluster 'genOnOff', data '{"onOff":1}' from endpoint 11 with groupID 0
```

### What did you expect to happen?

The On/Off button will always toggle the light bulb to the other state.

### How to reproduce it (minimal and precise)

* Bind the dimmer to a hue bulb or group (as in https://www.zigbee2mqtt.io/devices/929002398602.html)
* Turn on the bulb with the dimmer on/off button, note that the bulb turns off
* Turn off the bulb via Z2M or Home Assistant
* Attempt to turn on the bulb via the on/off button, note that the bulb remains off
* Turn on the bulb via the on/off button (second attempt), note that the bulb turns on

### Zigbee2MQTT version

1.35.1 commit: unknown

### Adapter firmware version

6.10.3.0 build 297

### Adapter

SONOFF ZBDONGLE-E

lasharor, 2024-01-30:
Is this a problem with Z2M or these Dimmers? it is really annoying. Happens not only with Hue bulbs but with other zigbee bulbs as well.

splump, 2024-05-19:
Agreed this is really annoying, if it would be possible to change this behaviour to toggle instead of alternating on/off than that would be ideal, but unsure if this would have to be adressed in firmware?

shizzlemo, 2024-12-28:
Hi, I have the same issue here. Will there be a fix or something I can do?

shizzlemo, 2025-03-01:
Has anyone a solutions yet?

closed 2025-05-09 by github-actions[bot]

Fixoon, 2025-05-31:
I'm also wondering about this issue. Where is the limitation coming from? I would really like to use the binding for on/off directly to the lamp, as I don't have to rely on my server working. If I recall correctly, it worked fine when I used the Hue hub, meaning the on/off was working when the hub wasn't connected and it was synced properly when the hub was on. Maybe the Hue hub does some magic to get it working, I have no idea myself.

jojeju9428, 2025-08-25:
Hello, I am also having a problem with the switches 929002398602. The ON function works fine, but the OFF does not.

zigbee2mqtt: 2.6.0
SLZB-06: 20250321; v2.9.8

Thank you for your help.

califlower, 2025-12-15:
https://vegard.blog.engen.priv.no/?p=562