Koenkk/zigbee2mqtt #20678: EcoDim Eco-Dim.07 - Images currently unavailable for device

Veldkornet, 2024-01-08:
### What happened?

All of a sudden today, my EcoDim [Eco-Dim.07](https://www.zigbee2mqtt.io/devices/Eco-Dim.07_Eco-Dim.10.html) is giving an OTA error that the image has disappeared (it installed the update moths ago though)

```
Zigbee2MQTT:warn  2024-01-08 17:53:32: OTA: Images currently unavailable for device 'EcoDim-Zigbee 3.0', hardwareVersion '1', manufacturerName EcoDim BV, {"fieldControl":0,"manufacturerCode":4714,"imageType":60,"fileVersion":16869200}'
```

[image]

If I check the [OTA index](https://github.com/Koenkk/zigbee-OTA/blob/master/index.json), but it looks like the imageType is different, but I don’t really see any updates recently (for this device anyway).

```
    {
        "fileVersion": 4,
        "fileSize": 286646,
        "manufacturerCode": 4714,
        "imageType": 1000,
        "sha512": "9b7835e32ef5af35385710332838099c5dce99a398ae04a1a769d67935fb0752ca4da58892f13bc842cbf8f9ae20516142fe9b66eff3e8ebef6577a71a1d9b0b",
        "url": "https://github.com/Koenkk/zigbee-OTA/raw/master/images/EcoDim/ZB21S3_HZC_Dimmer1_EcoDim-Zigbee%203.0_1.01_20230908.ota",
        "path": "images/EcoDim/ZB21S3_HZC_Dimmer1_EcoDim-Zigbee 3.0_1.01_20230908.ota"
    }
```

### Zigbee2MQTT version

1.35.1-1

### Adapter firmware version

20230507

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P

hydro-b, 2024-01-09:
I have the same problem for those dimmers. Running latest HA on a HA yellow. I use a "slae.sh_cc2652rb_stick".

Veldkornet, 2024-01-09:
I guess the problem with imageType 60 is the same as https://github.com/Koenkk/zigbee2mqtt/issues/16672

Which means that it needs this specific "sub-model" firmware. 
I contacted EcoDim before, but apparently there isn't an update and this is the factory image. Where as the "other" version had its firmware updated to the same as this model.

Veldkornet, 2024-01-09:
Closed this issue as there is already another issue logged for the "-1" problem.