Koenkk/zigbee2mqtt #20683: Devices do not respond to commands from z2m

kizio1000, 2024-01-08:
### What happened?

After some time z2m is working, random devices are not receiving commands (state change) from HA (or even directly from z2m panel. Restart Z2M solves issue for some hours.

### What did you expect to happen?

Devices should react to on/off commands

### Zigbee2MQTT version

version: 1.35.1-1

### Adapter firmware version

Coordinator revision 7.3.1.0 build 176

### Adapter

ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2

### Debug log

When one of devices is becoming unresponsive below data is appearing in log (debug level) of z2m:

`Debug 2024-01-08 20:54:50Received MQTT message on 'zigbee2mqtt/Lampa wiata/set' with data 'ON'
Debug 2024-01-08 20:54:50Publishing 'set' 'state' to 'Lampa wiata'
` 

After restarting z2m ( settings -> tools -> resrtart zigbee2mqtt ) switch is responding again and below data is being logged:

`Debug 2024-01-08 21:04:35Received MQTT message on 'zigbee2mqtt/Lampa wiata/set' with data 'ON'
Debug 2024-01-08 21:04:35Publishing 'set' 'state' to 'Lampa wiata'
Info 2024-01-08 21:04:35MQTT publish: topic 'zigbee2mqtt/Lampa wiata', payload '{"current":0,"energy":1.7,"linkquality":220,"power":0,"power_outage_memory":"restore","state":"ON","switch_type":"toggle","voltage":237}'
Debug 2024-01-08 21:04:35Received Zigbee message from 'Lampa wiata', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
Info 2024-01-08 21:04:35MQTT publish: topic 'zigbee2mqtt/Lampa wiata', payload '{"current":0,"energy":1.7,"linkquality":216,"power":0,"power_outage_memory":"restore","state":"ON","switch_type":"toggle","voltage":237}'
Debug 2024-01-08 21:04:38Received Zigbee message from 'Lampa wiata', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsCurrent":260}' from endpoint 1 with groupID 0
Info 2024-01-08 21:04:38MQTT publish: topic 'zigbee2mqtt/Lampa wiata', payload '{"current":0.3,"energy":1.7,"linkquality":220,"power":0,"power_outage_memory":"restore","state":"ON","switch_type":"toggle","voltage":237}'
Debug 2024-01-08 21:04:39Received Zigbee message from 'Lampa wiata', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":21}' from endpoint 1 with groupID 0
Info 2024-01-08 21:04:39MQTT publish: topic 'zigbee2mqtt/Lampa wiata', payload '{"current":0.3,"energy":1.7,"linkquality":220,"power":21,"power_outage_memory":"restore","state":"ON","switch_type":"toggle","voltage":237}'
Debug 2024-01-08 21:04:43Received Zigbee message from 'Lampa wiata', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsCurrent":170}' from endpoint 1 with groupID 0
Info 2024-01-08 21:04:43MQTT publish: topic 'zigbee2mqtt/Lampa wiata', payload '{"current":0.2,"energy":1.7,"linkquality":220,"power":21,"power_outage_memory":"restore","state":"ON","switch_type":"toggle","voltage":237}'`

Worth of noting is fact that device which is becoming unresponsive is still successfully sending reports about its state

sdotter, 2024-01-08:
Seems a lot of people "suffering" from issues after 1.33... what stick do u use>? dongle zbdongle e?

kizio1000, 2024-01-08:
> Seems a lot of people "suffering" from issues after 1.33... what stick do u use>? dongle zbdongle e?

Yes sonoff E. I seen that there is many reports about similar problems but so far it seams that problem remains. Strange part is that log is not giving any errors just silence.

kizio1000, 2024-01-08:
Situation from right now. One of switches did not reacted to off command, log:

`Debug 2024-01-08 22:11:00Received MQTT message on 'zigbee2mqtt/lampki nad lada/set' with data 'OFF'
Debug 2024-01-08 22:11:00Publishing 'set' 'state' to 'lampki nad lada'
Debug 2024-01-08 22:11:11Received Zigbee message from 'lampki nad lada', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":6,"rmsCurrent":55,"rmsVoltage":239}' from endpoint 1 with groupID 0
Info 2024-01-08 22:11:11MQTT publish: topic 'zigbee2mqtt/lampki nad lada', payload '{"current":0.06,"energy":2.04,"linkquality":92,"power":6,"power_outage_memory":"off","state":"ON","switch_type":"toggle","voltage":239}'
Debug 2024-01-08 22:11:11Received Zigbee message from 'lampki nad lada', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":6,"rmsCurrent":55,"rmsVoltage":239}' from endpoint 1 with groupID 0
Debug 2024-01-08 22:11:11Received Zigbee message from 'lampki nad lada', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":6,"rmsCurrent":55,"rmsVoltage":239}' from endpoint 1 with groupID 0
Debug 2024-01-08 22:11:11Received Zigbee message from 'lampki nad lada', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":6,"rmsCurrent":55,"rmsVoltage":239}' from endpoint 1 with groupID 0
Debug 2024-01-08 22:11:12Received Zigbee message from 'lampki nad lada', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":6,"rmsCurrent":55,"rmsVoltage":239}' from endpoint 1 with groupID 0
Debug 2024-01-08 22:11:12Received Zigbee message from 'lampki nad lada', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":6,"rmsCurrent":55,"rmsVoltage":239}' from endpoint 1 with groupID 0
Debug 2024-01-08 22:11:12Received Zigbee message from 'lampki nad lada', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":6,"rmsCurrent":55,"rmsVoltage":239}' from endpoint 1 with groupID 0
Debug 2024-01-08 22:11:12Received Zigbee message from 'lampki nad lada', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":6,"rmsCurrent":55,"rmsVoltage":239}' from endpoint 1 with groupID 0
Debug 2024-01-08 22:11:12Received Zigbee message from 'lampki nad lada', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":6,"rmsCurrent":55,"rmsVoltage":239}' from endpoint 1 with groupID 0
Debug 2024-01-08 22:11:13Received Zigbee message from 'lampki nad lada', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":6,"rmsCurrent":55,"rmsVoltage":239}' from endpoint 1 with groupID 0
Debug 2024-01-08 22:11:13Received Zigbee message from 'lampki nad lada', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":6,"rmsCurrent":55,"rmsVoltage":239}' from endpoint 1 with groupID 0
Debug 2024-01-08 22:11:13Received Zigbee message from 'lampki nad lada', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":6,"rmsCurrent":55,"rmsVoltage":239}' from endpoint 1 with groupID 0
Debug 2024-01-08 22:11:13Received Zigbee message from 'lampki nad lada', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":6,"rmsCurrent":55,"rmsVoltage":239}' from endpoint 1 with groupID 0
Debug 2024-01-08 22:11:13Received Zigbee message from 'lampki nad lada', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":6,"rmsCurrent":55,"rmsVoltage":239}' from endpoint 1 with groupID 0
Debug 2024-01-08 22:11:14Received Zigbee message from 'lampki nad lada', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":6,"rmsCurrent":55,"rmsVoltage":239}' from endpoint 1 with groupID 0
Debug 2024-01-08 22:11:14Received Zigbee message from 'lampki nad lada', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,204]}' from endpoint 1 with groupID 0
Info 2024-01-08 22:11:14MQTT publish: topic 'zigbee2mqtt/lampki nad lada', payload '{"current":0.06,"energy":2.04,"linkquality":96,"power":6,"power_outage_memory":"off","state":"ON","switch_type":"toggle","voltage":239}'`

After restart of z2m from settings->tools:

`Debug 2024-01-08 22:14:50Received MQTT message on 'zigbee2mqtt/lampki nad lada/set' with data 'OFF'
Debug 2024-01-08 22:14:50Publishing 'set' 'state' to 'lampki nad lada'
Info 2024-01-08 22:14:50MQTT publish: topic 'zigbee2mqtt/lampki nad lada', payload '{"current":0.07,"energy":2.04,"linkquality":196,"power":6,"power_outage_memory":"off","state":"OFF","switch_type":"toggle","voltage":239}'
Debug 2024-01-08 22:14:52Received Zigbee message from 'lampki nad lada', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsCurrent":0}' from endpoint 1 with groupID 0
Info 2024-01-08 22:14:52MQTT publish: topic 'zigbee2mqtt/lampki nad lada', payload '{"current":0,"energy":2.04,"linkquality":196,"power":6,"power_outage_memory":"off","state":"OFF","switch_type":"toggle","voltage":239}'`

kizio1000, 2024-01-08:
Only visible issue comes at start of z2m:

`Assertion failed: Command (setConfigurationValue) returned unexpected state: [object Object]`

sdotter, 2024-01-08:
> Only visible issue comes at start of z2m:
> 
> `Assertion failed: Command (setConfigurationValue) returned unexpected state: [object Object]`

Yeah I've had this one also on 7.3.10 ... Downgraded fw to 7.2.2.0 build 190 and that issue was gone although I didn't know exactly what it was causing it or causing issues

HelpITru, 2024-01-10:
I have the same issues, please help. 

Zigbee2MQTT version
    [1.35.1](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.35.1) commit: [unknown](https://github.com/Koenkk/zigbee2mqtt/commit/unknown)
Coordinator type
    EZSP v12
Coordinator revision
    7.3.2.0 build 212
Coordinator IEEE Address
    0x385b44fffe00322f
Frontend version
    0.6.151
Zigbee-herdsman-converters version
    18.9.0
Zigbee-herdsman version
    0.30.0
Stats
        Total 46
        By device type
        End devices: 25
        Router: 21

[LOG](https://pastebin.com/tH1UNDHc)

pjcoo, 2024-02-08:
I also have the same adapter and same issues. I have it stable by reversing to an older version but I no longer dare to update because the problem sometimes only occurs after a couple of days/hours. 
Anyone tried the latest update yet?

closed 2024-09-06 by github-actions[bot]