Koenkk/zigbee2mqtt #20686: OTA error for Ikea VALLHORN Wireless Motion Sensor

usma0118, 2024-01-09:
### What happened?

I have tried to check for OTA update on Ikea device VALLHORN Wireless Motion Sensor.
It's resulting into error.

### What did you expect to happen?

System should inform about new firmware available.
installed firmware: [1.0.57](https://ww8.ikea.com/ikeahomesmart/releasenotes/releasenotes.html)

latest: 1.26

### Zigbee2MQTT version

1.35.1 

### Adapter firmware version

20220219

### Adapter

zStack3x0

### Debug log

Debug 2024-01-09 09:50:33Received MQTT message on 'zigbee2mqtt/bridge/request/device/ota_update/check' with data '{"id":"Entrance sensor","transaction":"q5a6l-2"}'
Info 2024-01-09 09:50:33Checking if update available for 'Entrance sensor'
Debug 2024-01-09 09:50:33OTA: Checking if update available for '0x***************' (VALLHORN Wireless Motion Sensor)
Debug 2024-01-09 09:50:33OTA: Using endpoint '1'
Info 2024-01-09 09:50:57MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{"id":"Entrance sensor"},"error":"Failed to check if update available for 'Entrance sensor' (OTA: Device didn't respond to OTA request)","status":"error","transaction":"q5a6l-2"}'
Error 2024-01-09 09:50:57Failed to check if update available for 'Entrance sensor' (OTA: Device didn't respond to OTA request)
Debug 2024-01-09 09:50:57Error: OTA: Device didn't respond to OTA request at requestOTA (/app/node_modules/zigbee-herdsman-converters/src/lib/ota/common.ts:304:15) at Object.isUpdateAvailable (/app/node_modules/zigbee-herdsman-converters/src/lib/ota/common.ts:382:25) at OTAUpdate.onMQTTMessage (/app/lib/extension/otaUpdate.ts:205:45) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:167:17)

danieledwardgeorgehitchcock, 2024-01-09:
It doesn't look as-if there are any OTA images for this device currently published in the IKEA environments:

[Production](http://[fw.ota.homesmart.ikea.net/feed/version_info.json](http://fw.ota.homesmart.ikea.net/feed/version_info.json))

[Test](http://[fw.test.ota.homesmart.ikea.net/feed/version_info.json](http://fw.test.ota.homesmart.ikea.net/feed/version_info.json))

usma0118, 2024-01-09:
[Link](https://ww8.ikea.com/ikeahomesmart/releasenotes/releasenotes.html) provided by Z2M shows:

`Release Version: 1.26.0
04th Dec 2023
New features and changes in Gateway: 1.21.31
◆ Support for new devices.

New features and changes in Accessories:
JETSTRÖM white spectrum (2.4.5)
Product ID: L2207, L2208
◆ New product.

**VALLHORN motion sensor (1.0.57)**
Product ID: E2134
◆ New product.

SOMRIG shortcut button (1.0.20)
Product ID: E2213
◆ New product.`

usma0118, 2024-01-09:
Shouldn't sensor still display under OTA?

danieledwardgeorgehitchcock, 2024-01-09:
Yes, so the device is now supported via the changes that you mention above, but as there is no OTA image on the sites I reference, the OTA functionality has not been included.

This is standard practice as who is to say that Ikea wont use a different site for uploading the OTA images for your device?

Until we can confirm an OTA images existence, it is not recommended to assume that it follows the logic of others.

usma0118, 2024-01-09:
Makes sense.

closed 2024-01-09 by usma0118