Koenkk/zigbee2mqtt #20687: On the first reboot of 1.35 and 1.35-1 I got the following error message

Jeepmg, 2024-01-09:
### What happened?

On the first reboot of 1.35 and 1.35-1 I got the following error message which disappears when restarting Homebridge.

```
error 07-01-2024 15:33:23: Failed to configure 'Volet Droit', attempt 1 (Error: ConfigureReporting 0x20918a00000e0189/1 closuresWindowCovering([{"attribute":"currentPositionLiftPercentage","minimumReportInterval":1,"maximumReportInterval":65000,"reportableChange":1}], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:321:28)
    at Endpoint.configureReporting (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:665:22)
    at Object.currentPositionLiftPercentage (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:49:5)
    at Object.configure (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/devices/profalux.ts:91:13)
    at Configure.configure (/opt/zigbee2mqtt/lib/extension/configure.ts:120:13)
    at Immediate.<anonymous> (/opt/zigbee2mqtt/lib/extension/configure.ts:72:17))
error 07-01-2024 15:33:44: Failed to configure 'Volet Gauche', attempt 1 (Error: ConfigureReporting 0x20918a00000e0187/1 closuresWindowCovering([{"attribute":"currentPositionLiftPercentage","minimumReportInterval":1,"maximumReportInterval":65000,"reportableChange":1}], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:321:28)
    at Endpoint.configureReporting (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:665:22)
    at Object.currentPositionLiftPercentage (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:49:5)
    at Object.configure (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/devices/profalux.ts:91:13)
    at Configure.configure (/opt/zigbee2mqtt/lib/extension/configure.ts:120:13)
    at Immediate.<anonymous> (/opt/zigbee2mqtt/lib/extension/configure.ts:72:17))
error 07-01-2024 15:34:21: Failed to configure 'Volet Droit', attempt 2 (Error: ConfigureReporting 0x20918a00000e0189/1 closuresWindowCovering([{"attribute":"currentPositionLiftPercentage","minimumReportInterval":1,"maximumReportInterval":65000,"reportableChange":1}], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:321:28)
    at Endpoint.configureReporting (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:665:22)
    at Object.currentPositionLiftPercentage (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:49:5)
    at Object.configure (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/devices/profalux.ts:91:13)
    at Configure.configure (/opt/zigbee2mqtt/lib/extension/configure.ts:120:13)
    at EventEmitter.wrappedCallback (/opt/zigbee2mqtt/lib/eventBus.ts:167:17))
error 07-01-2024 15:34:22: Failed to configure 'Volet Gauche', attempt 2 (Error: ConfigureReporting 0x20918a00000e0187/1 closuresWindowCovering([{"attribute":"currentPositionLiftPercentage","minimumReportInterval":1,"maximumReportInterval":65000,"reportableChange":1}], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:321:28)
    at Endpoint.configureReporting (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:665:22)
    at Object.currentPositionLiftPercentage (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:49:5)
    at Object.configure (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/devices/profalux.ts:91:13)
    at Configure.configure (/opt/zigbee2mqtt/lib/extension/configure.ts:120:13)
    at EventEmitter.wrappedCallback (/opt/zigbee2mqtt/lib/eventBus.ts:167:17))
error 07-01-2024 15:34:24: Failed to check if update available for 'Philips_Remote' (OTA: Device didn't respond to OTA request)
error 07-01-2024 16:18:29: Failed to configure 'Volet Droit', attempt 3 (Error: ConfigureReporting 0x20918a00000e0189/1 closuresWindowCovering([{"attribute":"currentPositionLiftPercentage","minimumReportInterval":1,"maximumReportInterval":65000,"reportableChange":1}], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:321:28)
    at Endpoint.configureReporting (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:665:22)
    at Object.currentPositionLiftPercentage (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:49:5)
    at Object.configure (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/devices/profalux.ts:91:13)
    at Configure.configure (/opt/zigbee2mqtt/lib/extension/configure.ts:120:13)
    at EventEmitter.wrappedCallback (/opt/zigbee2mqtt/lib/eventBus.ts:167:17))
error 07-01-2024 16:18:37: Failed to configure 'Volet Gauche', attempt 3 (Error: ConfigureReporting 0x20918a00000e0187/1 closuresWindowCovering([{"attribute":"currentPositionLiftPercentage","minimumReportInterval":1,"maximumReportInterval":65000,"reportableChange":1}], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:321:28)
    at Endpoint.configureReporting (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:665:22)
    at Object.currentPositionLiftPercentage (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:49:5)
    at Object.configure (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/devices/profalux.ts:91:13)
    at Configure.configure (/opt/zigbee2mqtt/lib/extension/configure.ts:120:13)
    at EventEmitter.wrappedCallback (/opt/zigbee2mqtt/lib/eventBus.ts:167:17))
error 07-01-2024 16:23:07: Failed to check if update available for 'Philips_Remote' (OTA: Device didn't respond to OTA request)
```

After update 1.35-1 my Profalux MAI-ZTP20 remote control became MAI-ZTM20C remote.

These 2 changes apparently have no impact on the operation of Z2M.

### How to reproduce it (minimal and precise)

.

### Zigbee2MQTT version

1.35-1

### Adapter firmware version

20230507

### Adapter

    zStack3x0

Koenkk, 2024-01-09:
What devices are `Volet Droit` and `Volet Gauche`?

NYZack, 2024-01-09:
I am having this problem too.  All Sonoff S31ZB:

Zigbee2MQTT:error 2024-01-09 16:57:18: Failed to configure 'Front hall', attempt 3 (Error: ConfigureReporting 0x00124b0022ee27e2/1 genOnOff([{"minimumReportInterval":0,"maximumReportInterval":65000,"reportableChange":1,"attribute":"onOff"}], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 49561 - 1 - 14 - 6 - 7 after 10000ms)
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (node:internal/timers:569:17)
    at processTimers (node:internal/timers:512:7))
Zigbee2MQTT:error 2024-01-09 16:57:20: Failed to configure 'Master computer switch', attempt 2 (Error: ConfigureReporting 0x00124b00250c4a61/1 genOnOff([{"minimumReportInterval":0,"maximumReportInterval":65000,"reportableChange":1,"attribute":"onOff"}], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 29213 - 1 - 15 - 6 - 7 after 10000ms)
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (node:internal/timers:569:17)
    at processTimers (node:internal/timers:512:7))
Zigbee2MQTT:error 2024-01-09 16:57:20: Failed to configure 'Master floor lamp', attempt 3 (Error: ConfigureReporting 0x00124b00250c2b53/1 genOnOff([{"minimumReportInterval":0,"maximumReportInterval":65000,"reportableChange":1,"attribute":"onOff"}], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 26450 - 1 - 16 - 6 - 7 after 10000ms)
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (node:internal/timers:569:17)
    at processTimers (node:internal/timers:512:7))

NYZack, 2024-01-09:
I downgraded to 1.34.0-1, and no more log errors.

evelant, 2024-01-10:
I'm seeing a similar issue after updating. All of my Sonoff SNZB-06P sensors are offline and the log gets spammed with "Failed to configure" errors.

Jeepmg, 2024-01-10:
> What devices are `Volet Droit` and `Volet Gauche`?

Sorry, it is Profalux NSAV061 Cover.  https://www.zigbee2mqtt.io/devices/NSAV061.html

Koenkk, 2024-01-10:
@NYZack SZ31ZB configure errors are already fixed in the dev branch.

All others: please let me know for what device the configure fails + the error log 

@Jeepmg when changing the cover position physically (not via z2m), does the position update?

Jeepmg, 2024-01-10:
> @Jeepmg when changing the cover position physically (not via z2m), does the position update?

This time, with version 1.35.1, the position is updated in Z2M and Homebridge. This was not the case with version 1.35.0 where the position was not updated.

I remind you that the error message "Failed to configure 'Right Pane'..........." disappears when rebooting Homebridge.

LordXaosa, 2024-01-10:
Got same thing with smart sockets, but realized that update was not the reason. Few hours before I added pci-e-sata card with 3 SSDs. Seems like they are interfering with zigbee. So the solution was simple - just move usb dongle far away from server.

jangshik, 2024-01-11:
same error and unexpected behavior

Device name : SBM300Z3

Failed to configure 'baby_pswitch', attempt 3 (Error: ConfigureReporting 0x00158d000736dabc/4 genOnOff([{"minimumReportInterval":0,"maximumReportInterval":65000,"reportableChange":1,"attribute":"onOff"}], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 18686 - 4 - 16 - 6 - 7 after 10000ms) at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35) at listOnTimeout (node:internal/timers:569:17) at processTimers (node:internal/timers:512:7))

crunchip77, 2024-01-19:
> Alle anderen: Bitte teilen Sie mir mit, für welches Gerät die Konfiguration fehlschlägt + das Fehlerprotokoll

sonoff SNZB-06P
Zigbee2MQTT-Version
    [1.35.1](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.35.1) commit: [9eaaa0f]

`Failed to configure 'radar Kueche', attempt 3 (Error: Read 0x3410f4fffee962a3/1 64529([8193], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:321:28) at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:463:22) at processTicksAndRejections (node:internal/process/task_queues:95:5) at setupAttributes (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:62:9) at configure (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:72:9) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:140:25) at Configure.configure (/app/lib/extension/configure.ts:120:13) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:167:17))`
`Failed to configure 'radar Kueche', attempt 2 (Error: Read 0x3410f4fffee962a3/1 64529([8193], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:321:28) at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:463:22) at processTicksAndRejections (node:internal/process/task_queues:95:5) at setupAttributes (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:62:9) at configure (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:72:9) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:140:25) at Configure.configure (/app/lib/extension/configure.ts:120:13) at Immediate.<anonymous> (/app/lib/extension/configure.ts:72:17))`
`Failed to configure 'radar Kueche', attempt 1 (Error: Read 0x3410f4fffee962a3/1 64529([8193], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:321:28) at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:463:22) at processTicksAndRejections (node:internal/process/task_queues:95:5) at setupAttributes (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:62:9) at configure (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:72:9) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:140:25) at Configure.configure (/app/lib/extension/configure.ts:120:13) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:167:17))`

Koenkk, 2024-01-20:
@jangshik fixed the issue

@crunchip77 SNZB-06P is also fixed in the dev branch.

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

Jeepmg, 2024-01-20:
is there a link with https://github.com/Koenkk/zigbee2mqtt/issues/20833 ?

crunchip77, 2024-01-21:
> SNZB-06P is also fixed in the dev branch.
> 
> Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now

thanks, but now with the dev

Xiaomi RTCGQ14LM

`Failed to configure 'BWM Wohnzimmer', attempt 3 (Error: Read 0x54ef44100043eb88/1 genPowerCfg(["batteryVoltage"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 10592 - 1 - 202 - 1 - 1 after 10000ms) at Object.start (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:63:23) at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:499:47) at processTicksAndRejections (node:internal/process/task_queues:95:5) at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20) at Request.send (/app/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:79:20) at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:478:28) at configure (/app/node_modules/zigbee-herdsman-converters/src/devices/xiaomi.ts:1764:13) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:140:25) at Configure.configure (/app/lib/extension/configure.ts:120:13) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))`

Koenkk, 2024-01-22:
Try wakeing up the device right before doing the configure (press a button on it). You can do the configure via the z2m frontend -> yellow refresh button.

Jeepmg, 2024-01-22:
> Try wakeing up the device right before doing the configure (press a button on it). You can do the configure via the z2m frontend -> yellow refresh button.

error message after yellow refresh button :+1: 

``error 2024-01-22 20:38:01Failed to configure 'Volet Gauche', attempt 5 (Error: ConfigureReporting 0x20918a00000e0187/1 closuresWindowCovering([{"attribute":"currentPositionLiftPercentage","minimumReportInterval":1,"maximumReportInterval":65000,"reportableChange":1}], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:321:28) at Endpoint.configureReporting (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:665:22) at Object.currentPositionLiftPercentage (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:49:5) at Object.configure (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/devices/profalux.ts:91:13) at Configure.configure (/opt/zigbee2mqtt/lib/extension/configure.ts:120:13) at Configure.onMQTTMessage (/opt/zigbee2mqtt/lib/extension/configure.ts:55:21) at EventEmitter.wrappedCallback (/opt/zigbee2mqtt/lib/eventBus.ts:167:17))

Koenkk, 2024-01-23:
@Jeepmg what device is `Volet Gauche`?

Jeepmg, 2024-01-23:
> @Jeepmg what device is `Volet Gauche`?

Always the same :

it is Profalux NSAV061 Cover. https://www.zigbee2mqtt.io/devices/NSAV061.html

Koenkk, 2024-01-24:
@Jeepmg does z2m detect it as `MOT-C1ZxxC/F`?

Jeepmg, 2024-01-25:
> @Jeepmg does z2m detect it as `MOT-C1ZxxC/F`?

No there are correctly seen as NSAV061 but after update 1.35-1 my Profalux MAI-ZTP20 remote control became MAI-ZTM20C remote.

is there a link with issue :  https://github.com/Koenkk/zigbee2mqtt/issues/20833 ?

Koenkk, 2024-01-26:
Can you provide a screenshot of the z2m frontend -> device -> about page?

Jeepmg, 2024-01-27:
> Can you provide a screenshot of the z2m frontend -> device -> about page?

[image]

Like this ?

Koenkk, 2024-01-28:
The issue will be solved in the next z2m release (1 feb) and is already fixed in the  [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

lukbe, 2024-02-02:
I upgraded to 1.35.2-1 and I have this issue.

Zigbee2MQTT:error 2024-02-02 17:15:25: Failed to configure 'Senzor chodba', attempt 1 (Error: Read 0xa4c138e4cfeb4201/1 hvacUserInterfaceCfg(["programmingVisibility"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:319:28)
    at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:469:22)
    at setupAttributes (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:63:13)
    at configure (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:84:17)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:140:25)
    at Configure.configure (/app/lib/extension/configure.ts:120:13)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))

closed 2024-09-01 by github-actions[bot]