Koenkk/zigbee2mqtt #20688: OTA support for PARASOLL Door/Window Sensor

usma0118, 2024-01-09:
### What happened?

Support for Ikea E2013 is implemented in #20007.
but no OTA update is reflected.

### Zigbee2MQTT version

1.35

### Adapter firmware version

20221226

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus,TI CC2652P + CP2102(N)

danieledwardgeorgehitchcock, 2024-01-09:
It doesn't look as-if there are any OTA images for this device currently published in the IKEA environments:

[Production](http://[fw.ota.homesmart.ikea.net/feed/version_info.json](http://fw.ota.homesmart.ikea.net/feed/version_info.json))

[Test](http://[fw.test.ota.homesmart.ikea.net/feed/version_info.json](http://fw.test.ota.homesmart.ikea.net/feed/version_info.json))

usma0118, 2024-01-09:
Carrying discussion from #20686.
OTA updates are not supported for this device as this is not displayed under OTA tab in Z2M

pvangorp, 2024-01-13:
It was working correctly in 1.35.0.1. Since 1.35.1.1 I suddenly got issues as well. See attachments.

[image]

[image]

After reverting to 1.35.0.1 everything works fine.

[image]

mklooss, 2024-01-15:
@pvangorp use contact not tamper.

Temper has been removed
https://github.com/Koenkk/zigbee-herdsman-converters/commit/5403c8aa704fa1a8db7f2fcc01623dd26215b7a9

LarsStegman, 2024-01-18:
It does have a tamper function? How do you know?

pvangorp, 2024-01-18:
When you open the batteryslot, it triggers tamper e.g. sabotage. Because I have them

LarsStegman, 2024-01-18:
Interesting. How can you see what the device does? I have them as well but I'm relatively new to zigbee.

pvangorp, 2024-01-18:
[image]

When I open the batteryslot. It triggers tamper

LarsStegman, 2024-01-18:
Strange, mine don't seem to trigger when I open the battery slot

pvangorp, 2024-01-19:
Your right. Only one gifs sabotage, the others don't. I bought one earlier than the others. Might be a batch difference?!

LarsStegman, 2024-01-19:
What is the firmware version of each sensor?

pvangorp, 2024-01-19:
All of them have firmware 1.0.19

mklooss, 2024-01-19:
this Topic is about OTA and not the Tamper, could u create an ticket here:
https://github.com/Koenkk/zigbee-herdsman-converters

x2005254, 2024-01-25:
Hello, 

from z2m 1.35.1, the plugin crash if I tried to add a new parasoll with this error :
TypeError: Cannot read properties of undefined (reading 'resolveDefinition')
 at Controller.<anonymous> (/var/www/html/plugins/z2m/resources/zigbee2mqtt/lib/zigbee.ts:89:26)
    at Controller.emit (node:events:517:28)
    at Controller.selfAndDeviceEmit (/var/www/html/plugins/z2m/resources/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/controller.ts:527:14)
    at Controller.onDeviceJoined (/var/www/html/plugins/z2m/resources/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/controller.ts:586:22)

should I update the zigbee.ts with the one available there 
https://github.com/Koenkk/zigbee-herdsman-converters

but it's not an external converter right?

usma0118, 2024-07-24:
Still valid.

MrAdam, 2025-01-03:
Still relevant

LarsStegman, 2025-03-06:
Not stale

sjorge, 2025-03-06:
The device does have OTA enabled, see https://github.com/Koenkk/zigbee-herdsman-converters/blob/4b7c15c91a5f8582c18ab7a4d46123a1db45d062/src/devices/ikea.ts#L943

It probably means there are no ota images available yet, or if there are they are not added to https://github.com/Koenkk/zigbee-OTA

LarsStegman, 2025-03-06:
Okay, then it can be closed right?

sjorge, 2025-03-06:
> Okay, then it can be closed right?

I think so, and if there is an ota image available that is not in zigbee-OTA it probably worth trying to add it your self or open an issue with the URL to the image so someone else can add it.

closed 2025-03-06 by Koenkk