Koenkk/zigbee2mqtt #20689: TS0601 Dimmer knob does not support indicator_mode change

sbolten, 2024-01-09:
### What happened?

I have purchased the following dimmer know from Aliexpress: https://de.aliexpress.com/item/1005006054787238.html?spm=a2g0o.order_list.order_list_main.28.31695c5fo1YwL4&gatewayAdapt=glo2deu

It Z2M it is discovered as a MOES dimmer knob (according to the picture).
Manufacturer: Tuya
Model: TS0601_dimmer_knob

Everything is working except when trying to set the "indicator mode" (none/relay/pos) - z2m responds with the following error message - when for example trying to switch to "none":

No converter available for 'indicator_mode' ("none")

What can I do about this? I would really like to be able to switch of the led.

Regards, Stephan

### What did you expect to happen?

Changing the indicator mode should change the behaviour of the led on the device

### How to reproduce it (minimal and precise)

- Pair the device - successful
- use z2m to change change the indicator mode (devices>details)
- error message appears

### Zigbee2MQTT version

1.35.1

### Adapter firmware version

20230507

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus

### Debug log

Debug 2024-01-09 10:18:55Received MQTT message on 'zigbee2mqtt/0x60a423fffeef4a5c/set' with data '{"indicator_mode":"relay"}'
Error 2024-01-09 10:18:55No converter available for 'indicator_mode' ("relay")

sbolten, 2024-01-28:
Any chance that this will work at some point or is it just not supported by the device? How can I check this? Willing to contribute here but don't know how.

ruilima, 2024-02-06:
@sbolten did you solve it?
I am facing the same problem.
Thanks

sbolten, 2024-02-06:
Unfortunately not solved yet

ruilima, 2024-02-07:
I was trying to solve it but with no luck. i dont have a tuya gateway to get de datapoints to test.
None of **Light type**  and **Indicator mode** is working. 

I just found out that the developers know about the issue. 
https://github.com/Koenkk/zigbee-herdsman-converters/pull/5204

hdmvide, 2024-05-15:
I have the same problem.
there are news?

closed 2025-02-05 by github-actions[bot]

yurtalan92, 2026-02-20:
Any progress? I'm experiencing the same problem.

NorDroN, 2026-09-03:
I have the same problem.
there are news?