Koenkk/zigbee2mqtt #20690: MQTT error: connect EHOSTUNREACH

carlosnh96, 2024-01-09:
### What happened?

Good morning,
I have my Docker container running perfectly for many versions,  I do it from Container Station on a QNAP TS-453D NAS. Since I updated to version 1.35 or 1.35.1, it doesn't connect to the MQTT server, both are connected to the same bridge. 
To make the system operational again, I have to revert to version 1.34.0. 
Has anyone else experienced this and found a solution?

### How to reproduce it (minimal and precise)

When I update to version 1.35 or 1.35.1

### Zigbee2MQTT version

1.35.1

### Adapter firmware version

20230507

### Adapter

SONOFF TI CC2652P

### Debug log

info  08-01-2024 10:02:45: Zigbee: disabling joining new devices.
info  08-01-2024 10:02:46: Connecting to MQTT server at mqtt://192.168.1.55:1883
debug 08-01-2024 10:02:46: Using MQTT keepalive: 60
debug 08-01-2024 10:02:46: Using MQTT login with username: mqtt
error 08-01-2024 10:02:49: MQTT error: connect EHOSTUNREACH 192.168.1.55:1883
error 08-01-2024 10:02:49: MQTT failed to connect, exiting...
info  08-01-2024 10:02:49: Stopping zigbee-herdsman...
error 08-01-2024 10:02:52: MQTT error: connect EHOSTUNREACH 192.168.1.55:1883
error 08-01-2024 10:02:55: MQTT error: connect EHOSTUNREACH 192.168.1.55:1883
error 08-01-2024 10:02:58: MQTT error: connect EHOSTUNREACH 192.168.1.55:1883

aflegler, 2024-01-10:
I have the same issue with version 1.35.1-1

DubhAd, 2024-01-10:
Your broker isn't reachable, which sounds like a networking problem.

```
connect EHOSTUNREACH 192.168.1.55:1883
```

aflegler, 2024-01-10:
@carlosnh96
I restored version 1.34.0 but had the same problem. I then discovered that Mosquitto Broker wasn't running because I had a problem with the SSL certificate. After I solved that, everything worked again - even with version 1.35.1-1.

carlosnh96, 2024-01-11:
> @carlosnh96
> 
> I restored version 1.34.0 but had the same problem. I then discovered that Mosquitto Broker wasn't running because I had a problem with the SSL certificate. After I solved that, everything worked again - even with version 1.35.1-1.

In my case, I don't use SSL/TLS authentication, I only use a username and password to access my MQTT broker.

Could you please specify the exact change you made to make it work?

Thank you.

groggy99, 2024-02-04:
@carlosnh96 
I have the same problem. So far z2m v1.34 ran in the Docker container on QNAP TS-453B with QTS 5.1.5.2645 without any problems. To get z2m v1.35.x working, I had to change the container network mode from “Bridge” to “Host”. After that, z2m v1.35.2 connects to Mosquito without any problems. Haven't changed anything else.

freddieleeman, 2024-02-29:
Same issue, unable to upgrade to anything higher than 1.34. Did anyone find the cause or solution?

groggy99, 2024-05-04:
I have the same issue, unable to upgrade to anything higher than 1.34. Has been running on Docker with SONOFF TI CC2652P for years without any problems.

johnrobertcobbold, 2024-05-08:
I was facing the same thing until I pulled what must be a new version of mqtt:

`docker-compose pull mqtt`

groggy99, 2024-05-18:
> I was facing the same thing until I pulled the `mqtt` and `portainer` containers before restarting (not sure which one was was the culprit).
> 
> * `docker-compose pull mqtt`
> * `docker-compose pull portainer`
> * `docker-compose up -d`

This doesn't work for me.

But it's working again for me. I now have 1.37.1 running.

It worked again when I reconfigured the virtual switch on the qnap nas. I created a second virtual switch with the z2m-conatiner as the only participant. Then I added the z2m container back to the original switch and now it works. 
I can't explain it any further.

closed 2024-12-30 by github-actions[bot]