Koenkk/zigbee2mqtt #20692: Aqara QBKG24LM Energy not avalaible

Johnymalina, 2024-01-09:
### What happened?

Switch doesnt send Energy value. I tried to wait a few hours hoping it would start if there is something to report but now its almost 12 hours and nothing is being reported.

### Zigbee2MQTT version

1.35.1-1

### Adapter firmware version

 20230507

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus-P

### Debug log

{
    "linkquality": 0,
    "operation_mode_left": "decoupled",
    "operation_mode_right": "decoupled",
    "power": 3.46,
    "state_left": "OFF",
    "state_right": "ON",
    "update": {
        "installed_version": 66,
        "latest_version": 66,
        "state": "idle"
    },
    "action": "",
    "energy": null,
    "update_available": null
}

avihome63, 2024-02-09:
Reproduced the same issue on

**Zigbee2MQTT version**
1.35.2-1

**Adapter**
SONOFF Zigbee 3.0 USB Dongle Plus-**E** (EZSP)

yhjohn, 2024-03-03:
Just to provide additional context, I Have all 1, 2, 3 gang switches and only the 2 Gang version is not captured.

1 Gang (QBKG23LM) - Reporting
2 Gang (QBKG24LM) - NOT Reporting
3 Gang (QBKG26LM) - Reporting

Koenkk, 2024-03-04:
Should be fixed now.
Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

yhjohn, 2024-03-08:
I just switch to the edge version and can confirmed that this issue has indeed been fixed.

closed 2024-03-08 by Koenkk