Koenkk/zigbee2mqtt #20694: Aqara SRTS-A01 converter unknown keys

shrx, 2024-01-09:
### What happened?

I upgraded the Aqara SRTS-A01 firmware using the OTA functionality, to the latest firmware version (0.0.0_01030). When the update finished, everything is OK except there are some warnings in the logs about "Unknown key"s:

```
warning 2024-01-09 12:22:47zigbee-herdsman-converters:aqara_trv: Unknown key 641 = 
warning 2024-01-09 12:22:47zigbee-herdsman-converters:aqara_trv: Unknown key 642 = `
warning 2024-01-09 12:22:48zigbee-herdsman-converters:aqara_trv: Unknown key 643 = 0
warning 2024-01-09 12:22:48zigbee-herdsman-converters:aqara_trv: Unknown key 644 = 0
```

Including a screenshot due to unprintable characters:
[image]

Does this mean there is additional data published by the device not recognized by the converter?

### What did you expect to happen?

No warnings in the logs.

### How to reproduce it (minimal and precise)

Connect an Aqara SRTS-A01. Run the OTA firmware upgrade.

### Zigbee2MQTT version

1.35.0 commit: 3d8f0fc3

### Adapter firmware version

20211115

### Adapter

CC2531

ankaboutobruni, 2024-02-16:
I have the same problem.

Conbee II stick., Z2M, here are my errors, and they just pop up randomly, not during upgrades or anything.
`Warning 2024-02-16 11:11:41zigbee-herdsman-converters:lumi_trv: Unknown key 641 = 4��84�4
Warning 2024-02-16 11:11:42zigbee-herdsman-converters:lumi_trv: Unknown key 642 = `4��84�4
Warning 2024-02-16 11:11:42zigbee-herdsman-converters:lumi_trv: Unknown key 643 = 0
Warning 2024-02-16 11:11:42zigbee-herdsman-converters:lumi_trv: Unknown key 644 = 0`

Mikaka27, 2024-03-18:
I'm on ZHA and I also see some problems with this TRV since this firmware version.
For me time synchronization doesn't work, does it work for you in z2m?

Also I think the schedule syntax has changed with this firmware update, at least according to people on the internet, I don't have the aqara hub, so I cannot check myself.
https://www.reddit.com/r/Aqara/comments/11pnc44/e1_radiator_thermostat_schedules_after_1030_update/
https://www.reddit.com/r/Aqara/comments/11haxwy/e1_trvs_latest_firmware_update/

closed 2024-10-15 by github-actions[bot]