Koenkk/zigbee2mqtt #20702: BSEED IN WALL SOCKET recognized as TS011F_plug_1

franele, 2024-01-09:
### What happened?

Hello,
In July 2023 I purchased a BSEED wall socket.
When connected to Zigbee2mqtt it was recognized as "TS011F_plug_1" and exposed consumption and power. I use it to control the consumption of a refrigerator in Home Assistant.
Recently and because of the good results it has given me, I have bought two more plugs from the same brand, but this time when I linked it they appear as "TS011F_wall_outlet" but it can only be controlled like a switch.
My intention was to use these plugs to control the consumption of a washing machine and a dryer.

### What did you expect to happen?

I hope to use this device to control electrical consumption

### Zigbee2MQTT version

1.35.1-1

### Adapter firmware version

20210708

### Adapter

sonoff ZBDongle-P

Koenkk, 2024-01-10:
Could you provide the data/database.db entry of your device detecteed as TS011F_wall_outlet?

franele, 2024-01-12:
{"id":13,"type":"Router","ieeeAddr":"0xa4c138ec2fdf815b","nwkAddr":13161,"manufId":4417,"manufName":"_TZ3000_b28wrpvx","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[3,4,5,6,1794,2820,57344,57345,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":":zR,\u0003","65506":54,"65508":0,"65534":0,"modelId":"TS011F","manufacturerName":"_TZ3000_b28wrpvx","powerSource":1,"zclVersion":3,"appVersion":192,"stackVersion":0,"hwVersion":1,"dateCode":""}},"haElectricalMeasurement":{"attributes":{"acCurrentDivisor":1000,"acCurrentMultiplier":1,"rmsVoltage":218,"rmsCurrent":0,"activePower":0}},"seMetering":{"attributes":{"divisor":100,"multiplier":1,"currentSummDelivered":[0,27612]}},"genOnOff":{"attributes":{"32768":0,"onOff":1,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":2,"tuyaBacklightSwitch":1}},"manuSpecificTuya_3":{"attributes":{"53248":1,"53249":0,"53250":2802818,"53251":122,"53252":11118,"53253":1284,"powerOnBehavior":2,"switchType":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0024cbdc43","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b0024cbdc43","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b0024cbdc43","endpointID":1}],"configuredReportings":[{"cluster":2820,"attrId":1285,"minRepIntval":5,"maxRepIntval":3600,"repChange":5,"manufacturerCode":null},{"cluster":2820,"attrId":1288,"minRepIntval":5,"maxRepIntval":3600,"repChange":50,"manufacturerCode":null},{"cluster":2820,"attrId":1291,"minRepIntval":5,"maxRepIntval":3600,"repChange":10,"manufacturerCode":null},{"cluster":1794,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":[1,1],"manufacturerCode":null}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":192,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1310285083},"lastSeen":1705066106835,"defaultSendRequestWhen":"immediate"}

Koenkk, 2024-01-13:
`_TZ3000_b28wrpvx` should be detected as `TS011F_plug_1`, I need the data/database.db entries of the ones detected as `TS011F_wall_outlet

franele, 2024-01-14:
Sorry, I read wrong. Here you are the first one:

{"id":27,"type":"Router","ieeeAddr":"0xa4c138829a350251","nwkAddr":41820,"manufId":4417,"manufName":"_TZ3000_o1jzcxou","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[3,4,5,6,57344,57345,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"dateCode":""}},"genOnOff":{"attributes":{"32768":0,"onOff":1,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":1,"tuyaBacklightSwitch":1}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":1,"switchType":0}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1705254198729,"defaultSendRequestWhen":"immediate"}

franele, 2024-01-14:
And the other one:

{"id":28,"type":"Router","ieeeAddr":"0xa4c138223600c41f","nwkAddr":30302,"manufId":4417,"manufName":"_TZ3000_o1jzcxou","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[3,4,5,6,57344,57345,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS011F","manufacturerName":"_TZ3000_o1jzcxou","powerSource":1,"zclVersion":3,"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{"32768":0,"onOff":1,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":1,"tuyaBacklightSwitch":1}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":1,"switchType":0}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1705254199107,"defaultSendRequestWhen":"immediate"}

Koenkk, 2024-01-15:
Fixed!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-01-15 by Koenkk

Anylitium, 2024-03-05:
zigbee_herdsman_converters_version18.42.0 - 
Zigbee Model
TS011F
Zigbee Manufacturer
_TZ3000_o1jzcxou

After adding the device status is permanent offline
But turning it on and off works stably. When connecting any electrical device, power is sold correctly, but all sensors on the outlet are in Null or not available

rosicenko, 2024-03-18:
Same here. Everything works well except I don't any metering data out of this socket. 

Zigbee Model
TS011F
Zigbee Manufacturer
_TZ3000_o1jzcxou