Koenkk/zigbee2mqtt #20703: z2mqtt stops working randomly

nicolapejo, 2024-01-09:
### What happened?

I have a Z2mqtt installation on a docker in an Unraid server, it occasionally stops receiving data from zigbee devices and if I try to give any input it returns the error: _sendZclFrameToEndpointInternal_. Restarting the docker everything starts working properly again.

### What did you expect to happen?

Zigbee devices should send their status and buttons should send input.

### Zigbee2MQTT version

 1.35.1 commit: 9eaaa0f

### Adapter firmware version

 6.10.3.0 build 297

### Adapter

ITEAD Sonoff ZBDongle-E

usma0118, 2024-01-10:
Related: #19269

usma0118, 2024-01-10:
Related #20715

ottoketney, 2024-01-12:
I have same behavior

apoltanov, 2024-01-12:
Same for me. Z2M is running as an add-on in HA and stops getting states from all devices at once. Log contains "Failed to ping..." entries due to availability feature.
Z2M keeps running for about 1 day long, and then all devices stop responding until z2m is reloaded. No any sign of an error in logs, except availability warnings.

syrusfem, 2024-01-13:
I have the same problem, once a day (sometimes 2 or 3 times a day) all Z2M controlled devices stop responding until i reload the integration. i have Z2M as an add-on in HA. Reverting to 1.3.4 solved my problems so it seems correlatied with 1.3.5.1 (and 1.3.5.0) version.

DarkVillager, 2024-01-13:
I have the same problem/

nicjansma, 2024-01-16:
Having the same issue here once I went to 1.35.1

AndreasFlach, 2024-01-16:
Have the same behavoir on Odroid M1 with Home Assistant OS and Zigbee2MQTT 1.35.1-1 Addin
HA Core 2024.1.3, Supervisor 2023.12.1, Operating System 11.4, Frontend 20240104.0.
Zigbee device stream dies, if you check with the overview card the error "lqi for 'coordinator' failed
Watchdog does not identify dead of Zigbee2MQTT.
Restart of Container solves it for a few hours, in this style not stable.
Stick: Aeotec Z-Stick 7

lmarcelocc, 2024-01-16:
Same here running on a Raspberry Pi.

fixed77, 2024-01-17:
Having the same issue, rolled back to 1.34.0 from backups.

Koenkk, 2024-01-17:
Please provide the herdman debug log from starting z2m until the crash with 1.35.1

See [this](https://www.zigbee2mqtt.io/guide/usage/debug.html) on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

fixed77, 2024-01-22:
[4.txt](https://github.com/Koenkk/zigbee2mqtt/files/14014224/4.txt)
first failed timestamp: 2024-01-22T18:13:16.944Z

Koenkk, 2024-01-23:
@fixed77 the coordinator crashes in your case, what fw are you using? If using `20230507` try [`20221226`](https://github.com/Koenkk/Z-Stack-firmware/releases/tag/Z-Stack_3.x.0_coordinator_20221226)

fixed77, 2024-01-23:
> @fixed77 the coordinator crashes in your case, what fw are you using? If using `20230507` try [`20221226`](https://github.com/Koenkk/Z-Stack-firmware/releases/tag/Z-Stack_3.x.0_coordinator_20221226)

20221226

fixed77, 2024-01-29:
> @fixed77 the coordinator crashes in your case, what fw are you using? If using `20230507` try [`20221226`](https://github.com/Koenkk/Z-Stack-firmware/releases/tag/Z-Stack_3.x.0_coordinator_20221226)

update to 20230507. my problem is resolved.

nicolapejo, 2024-01-30:
> Please provide the herdman debug log from starting z2m until the crash with 1.35.1
> 
> See [this](https://www.zigbee2mqtt.io/guide/usage/debug.html) on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

[log_zigbee2mqtt3.txt](https://github.com/Koenkk/zigbee2mqtt/files/14102673/log_zigbee2mqtt3.txt)
Here is my log, it has been a while since it froze

nicolapejo, 2024-07-29:
problem solved with latest release

closed 2024-07-29 by nicolapejo