Koenkk/zigbee2mqtt #20707: Bosch TRV BTH-RA wrong mqtt discovery, modes: ´auto´ missing, leading to hvac_mode ´unknown´ in ha

dierochade, 2024-01-09:
### What happened?

After upgrade to latest (from 1.34.1) my Bosch devices (https://www.zigbee2mqtt.io/devices/BTH-RA.html#bosch-bth-ra) report wrong hvac modes to home assistant leading to an incomplete hvac_state selection and therefore unknown state.

This is the mqtt discovery of one of the devices, modes ´auto´ missing:

Transmitted messages:
Thermostat Bosch 4 Bad (climate.thermostat_bosch_4_bad)
MQTT discovery data:
Topic: homeassistant/climate/0x18fc260000042a42/climate/config
Payload
action_template: >-
  {% set values =
  {None:None,'idle':'idle','heat':'heating','cool':'cooling','fan_only':'fan'}
  %}{{ values[value_json.running_state] }}
action_topic: zigbee2mqtt/Thermostat Bosch 4 Bad
availability:
  - topic: zigbee2mqtt/bridge/state
  - topic: zigbee2mqtt/Thermostat Bosch 4 Bad/availability
availability_mode: all
current_temperature_template: '{{ value_json.local_temperature }}'
current_temperature_topic: zigbee2mqtt/Thermostat Bosch 4 Bad
device:
  identifiers:
    - zigbee2mqtt_0x18fc260000042a42
  manufacturer: Bosch
  model: Radiator thermostat II (BTH-RA)
  name: Thermostat Bosch 4 Bad
  sw_version: 3.05.09
  via_device: zigbee2mqtt_bridge_0xe0798dfffe0f683d
json_attributes_topic: zigbee2mqtt/Thermostat Bosch 4 Bad
max_temp: '30'
min_temp: '5'
mode_command_topic: zigbee2mqtt/Thermostat Bosch 4 Bad/set/system_mode
mode_state_template: '{{ value_json.system_mode }}'
mode_state_topic: zigbee2mqtt/Thermostat Bosch 4 Bad
modes:
  - 'off'
  - heat
name: null
object_id: thermostat_bosch_4_bad
origin:
  name: Zigbee2MQTT
  sw_version: 1.35.1
  support_url: https://www.zigbee2mqtt.io
temp_step: 0.5
temperature_command_topic: zigbee2mqtt/Thermostat Bosch 4 Bad/set/occupied_heating_setpoint
temperature_state_template: '{{ value_json.occupied_heating_setpoint }}'
temperature_state_topic: zigbee2mqtt/Thermostat Bosch 4 Bad
temperature_unit: C
unique_id: 0x18fc260000042a42_climate_zigbee2mqtt
platform: mqtt

This is how it looks in developer tools on ha

climate.thermostat_bosch_1_wohnzimmerThermostat Bosch 1 Wohnzimmer | **unknown** | **hvac_modes: off, heat** min_temp: 5 max_temp: 30 target_temp_step: 0.5 current_temperature: 22.2 temperature: 21.5 hvac_action: idle battery: 68 boost: OFF child_lock: UNLOCK display_brightness: 3 display_onTime: 20 display_ontime: 20 display_orientation: normal displayed_temperature: measured last_seen: 2024-01-09T21:35:03+01:00 linkquality: 255 local_temperature: 22.2 local_temperature_calibration: 0 occupied_heating_setpoint: 21.5 pi_heating_demand: 0 remote_temperature: 22.2 running_state: idle setpoint_change_source: externally **system_mode: auto** update:  installed_version: 889787668 latest_version: 889787668 state: idle
-- | -- | --

</body>
</html>

Might be related to: 

https://github.com/home-assistant/core/issues/105683
https://github.com/home-assistant/core/issues/44255 (did not find follow up on this repository)

### What did you expect to happen?

Still all supported modes available

### Zigbee2MQTT version

1.35.1

### Adapter firmware version

7.3.1.0 build 0 (multipan)

### Adapter

skyconnect EZSP v12

dierochade, 2024-01-28:
Anyone else with these kind of problem? Any ideas how to fix? Assistance is much appreciated, as the state of the climate entity is wrong in ha and all automations that depend on this do not work any more.

GatoPharaoh, 2024-01-31:
auto mode has been removed in [d91e2f3](https://github.com/Koenkk/zigbee-herdsman-converters/commit/d91e2f3b5f602487357fc9414740365c496866dc) as setting the schedule doesn't seem to be possible.

on a side note - does your home assistant still show the "window_open" switches for the TRVs? these have gone missing for me after the last dev-update. and it doesn't look like they should be missing.

dierochade, 2024-02-01:
1. 
This change is rather surprising to me, cause the device does offer the mode, moreover it can be selected directly with the hardware button/dial. If it is useful or not may be dependent on point of view? 
I just use(d) both modes, so for my setup this is a hard breaking change

I tried to set up an external converter, and revert this single line that was changed. 
(exposes: [ 
....
        e.climate()
            .withSystemMode(['off', 'heat', 'auto'])

I did not succeed though. 

- The "somename.js" must be placed into the programm folder, for ha with z2m addon, thus: /homeassistant/zigbee2mqtt ?
- It then can be included using the gui frontend (→settings → eternal converters)
- I tried to figure out the appropriate content, but was unsuccessful (because I got no clue). The files just seem to define a lot of variables in the start and a definition for the single device? 
- I was unable to determine which part I should include. Everything did not work. just const definition = {original json} and module.exports = definition;
I just do not know what to include/what I am doing wrong.

2. 
@GatoPharaoh 
For me, the switch still is present and can be used to toggle, see screenshot
[image]

GatoPharaoh, 2024-02-01:
> This change is rather surprising to me, cause the device does offer the mode, moreover it can be selected directly with the hardware button/dial. 

maybe the author of the pull request can chime in on this - personally, I never considered auto mode to be a "standard" as for example the aqaras don't seem to offer one either

> I tried to set up an external converter, and revert this single line that was changed. (exposes: [ .... e.climate() .withSystemMode(['off', 'heat', 'auto'])

sadly, I can't help you with this one as I don't use the addon and have no experience with ha-addons

> @GatoPharaoh For me, the switch still is present and can be used to toggle

thank you - will have to check what went wrong on my side

ylieder, 2024-02-03:
> auto mode has been removed in [d91e2f3](https://github.com/Koenkk/zigbee-herdsman-converters/commit/d91e2f3b5f602487357fc9414740365c496866dc) as setting the schedule doesn't seem to be possible.
>
> on [...]

I faced the same issue tonight - "window_open" toggle is missing for all my Bosch Thermostat II in HA. In zigbee2mqtt it is still visible.

GatoPharaoh, 2024-02-04:
interesting - yes, it disappeared after updating z2m - specifically, I updated to version 1.35.1-dev

haven't really found the cause of this issue yet

ylieder, 2024-02-04:
> interesting - yes, it disappeared after updating z2m - specifically, I updated to version 1.35.1-dev
> 
> haven't really found the cause of this issue yet

Since this topic is unrelated to this issue, I created a new issue: https://github.com/Koenkk/zigbee2mqtt/issues/21208

dierochade, 2024-02-04:
I did create a new issue on herdsman-converters regarding the auto mode problem: Bosch [#6995 BTH-RA: Proposed reversion of removal of "auto mode" (as done in #6761 - 1.35.1) ](https://github.com/Koenkk/zigbee-herdsman-converters/issues/6995). 

To clarify my statement on the open window entity missing issue: After upgrade to 1.35.2 I face this too...

As this thread seems forked into two new discussions, I will close it right now.

closed 2024-02-04 by dierochade