Koenkk/zigbee2mqtt #20712: Ikea VINDSTYRKA Joins when no permitted & creates race condition

Boztekau, 2024-01-10:
### What happened?

After upgrading to the MQTT V1.35.1-1 my six Ikea VINDSTYRKA stations when into a race condition all sending data several times a seconds crashing MQTT getting the recent error; 
State attributes for sensor.zigbee2mqtt_bridge_permit_join_timeout exceed maximum size of 16384 bytes. This can cause database performance issues; Attributes will not be stored

I completed a new install of HA and MQTT adding dozens of other devices but when I turned on the Ikea VINDSTYRKA it joined MQTT without Permit Join along  with a unable to config error and went back into a race condition. Force delete and then Permit Join then turning the on the Ikea VINDSTYRKA with the same race conditions filling the MQTT log several times per second again. 

All six Ikea VINDSTYRKA station have current firmware ver: 1.0.11 build 20230202. I also tried a new unopened Ikea VINDSTYRKA with firmware ver 1.0.010 build 20220719 While it needs permit join is also had unable to config errors and goes into a race condition send data to the MQTT log several times per second. 

### Zigbee2MQTT version

1.35.1-1

### Adapter firmware version

EZSP v8  6.10.3.0 build 297

### Adapter

SONOFF  Dongle Plus

Blandness5107, 2024-01-19:
Also same issue - running both Z2M and ZHA - moved Ikea VINDSTYRKA over to ZHA to workaround the issue

Blandness5107, 2024-02-03:
Issue seems to be resolved with 1.35.2-1

closed 2024-02-16 by Boztekau