Koenkk/zigbee2mqtt #20714: Ikea SOMRIG not reporting actions correctly. Wrong action names in Converter maybe?

ielliott75, 2024-01-10:
### What happened?

Zigbee2MQTT correctly identifies the SOMRIG Ikea device, but all actions are ignored by homebridge.

Is it possible that there isn't a match, because homebridge maps dots_1_initial_press as the action but it is actually 1_initial_press in the MQTT action message?

### What did you expect to happen?

Action to flow through to HomeBridge

### How to reproduce it (minimal and precise)

Set up SOMRIG switch in Z2M, push button and nothing happens in Home.app, however action with incorrect name is seen in the homebridge logs.

### Zigbee2MQTT version

1.35.1

### Adapter firmware version

20210708

### Adapter

SONOFF

### Setup

Docker on Synology

### Debug log

HOMEBRIDGE LOG:
[1/9/2024, 7:08:28 PM] [zigbee2mqtt] Restoring accessory: NewSwitch (0x5cc7c1fffe87e21c)
[1/9/2024, 7:08:28 PM] [zigbee2mqtt] Config for accessory NewSwitch : {}
[1/9/2024, 7:08:28 PM] [zigbee2mqtt] Mapping of property 'action' of device 'NewSwitch':
Button 10 (dots_1_initial):
- SINGLE: dots_1_initial_press
Button 11 (dots_1_long):
- SINGLE: dots_1_long_press
Button 20 (dots_2_initial):
- SINGLE: dots_2_initial_press
Button 21 (dots_2_long):
- SINGLE: dots_2_long_press
[1/9/2024, 7:08:28 PM] [zigbee2mqtt] Configuring Battery Service for NewSwitch
[1/9/2024, 7:08:28 PM] [zigbee2mqtt] Updating service names for NewSwitch (from NewSwitch)
[1/9/2024, 7:08:28 PM] [zigbee2mqtt] Pending get: battery
[1/9/2024, 7:08:36 PM] [zigbee2mqtt] Publish to 'zigbee2mqtt/0x5cc7c1fffe87e21c/get': '{"battery":0}'
[1/9/2024, 7:08:37 PM] [zigbee2mqtt] Availability feature for NewSwitch is now disabled
[1/9/2024, 7:08:37 PM] [zigbee2mqtt] availability (DIS): NewSwitch: change status of characteristics to 0

With Debug turned on i can also see the following when i press the button on the switch:

[1/9/2024, 7:14:12 PM] [zigbee2mqtt] Handled device update for NewSwitch: {"action":"1_initial_press","battery":100,"linkquality":30,"update":{"installed_version":16777248,"latest_version":16777248,"state":"idle"},"voltage":1500}
[1/9/2024, 7:14:13 PM] [zigbee2mqtt] Handled device update for NewSwitch: {"action":"1_short_release","battery":100,"linkquality":66,"update":{"installed_version":16777248,"latest_version":16777248,"state":"idle"},"voltage":1500}

Koenkk, 2024-01-10:
These actions have already been corrected in [1.35.1](https://www.zigbee2mqtt.io/devices/E2213.html#action-enum) (`dots_` prefix has been removed)

ielliott75, 2024-01-11:
Yep, all working now. Thanks :)
Just needed to clear out the Homebridge cached accessories. 
Many thanks.

closed 2024-01-11 by ielliott75